stlink-org/stlink #810: Verification of flash failed - STM32WB55RG

section14, 2019-06-16:
- [X] Programmer/board type:Stlink/v2.1 onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [X] Operating system: Ubuntu 18.04
- [X] Stlink tools version and/or git commit hash: v1.5.1-30-g84f63d2
- [X] Stlink commandline tool name: `st-flash`
- [X] Target chip (and optional board): STM32WB55RGV6

If I compile a program that exceeds 8192 bytes (2 pages) on my STM32WB55RG board, I receive the error below. If I keep it below 8192 bytes, it flashes just fine. So, apparently, something is happening at the end of page 2 or beginning of page 3. 

I've tried running `st-flash erase` prior to flashing, but it makes no difference. I've also tried multiple resets and power cycles.

The programs (both working and failing) are compiled via a make file that was generated with cubemx. The program that fails to flash is the default setup generated by cubemx, without any additional code.

I have a L476RG board that I've flashed programs in the 500KiB range using this version of st-flash. No errors were produced in that case.

Output:

```
st-flash write WB55_02.bin 0x8000000
st-flash 1.5.1-30-g84f63d2
2019-06-16T13:58:49 INFO common.c: Loading device parameters....
2019-06-16T13:58:49 INFO common.c: Device connected is: WB55 device, id 0x20016495
2019-06-16T13:58:49 INFO common.c: SRAM size: 0x40000 bytes (256 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 4096 bytes
2019-06-16T13:58:49 INFO common.c: Attempting to write 11612 (0x2d5c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08002000 erased
2019-06-16T13:58:49 INFO common.c: Finished erasing 3 pages of 4096 (0x1000) bytes
Writing
Starting   2 page write
2019-06-16T13:58:50 INFO common.c: Starting verification of write complete
2019-06-16T13:58:50 ERROR common.c: Verification of flash failed at offset: 8192
stlink_fwrite_flash() == -1

```

Expected/description:
successful flash

eupn, 2019-07-04:
@section14 Hey, any luck with that? I also summon the @WRansohoff since they are the author of STM32WB integration.

xor-gate, 2019-07-14:
I think this issue can now be closed @section14 as I have merged the fix from @eupn. Could you verify?

WRansohoff, 2019-07-25:
Oh, thank you for fixing this - it looks like it might have been a similar issue as #797? Sorry about that, and sorry that I missed the ping.

DL6AKU, 2019-08-15:
Unfortunately this fix does not work for me, I keep getting :(

```
ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```

Nightwalker-87, 2020-02-20:
Obviously fixed by #816.

closed 2020-02-20 by Nightwalker-87