stlink-org/stlink #813: Cannot find or flash  STM32F103C8T6 on Linux using ST-LINK V.2

TravJav, 2019-06-27:
I have bought the blue pill STM32F103C8T6 using ST-LINK V.2. I have downloaded the required software to run ST-LINK V.2. So the issue I am having it that it appears the chip is not recognised when running st-util or any other st-*  command. 

When using the St-Link GUI on Arch Linux I can connect it seems but it appears the chip is a unknown device and Core = 0x0. 

Using Atollic Studio and attempting to connect the debugger I am getting this output.
```

STMicroelectronics ST-LINK GDB server. Version 5.1.0
Copyright (c) 2018, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 1
        Listen Port Number         : 61234
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled

Target no device found

Error in initializing ST-LINK device.
Reason: No device found on target.
```

**RUNNING $ st-util**
```

^C[travjav@rey]: ~/Development/Invictus-board/STM32F103C8T6>$ st-util
st-util 1.5.1
2019-06-26T23:41:46 INFO usb.c: -- exit_dfu_mode
2019-06-26T23:41:46 INFO common.c: Loading device parameters....
2019-06-26T23:41:46 WARN common.c: Invalid flash type, please check device declaration
2019-06-26T23:41:46 INFO gdb-server.c: Chip ID is 00000000, Core ID is  00000000.
2019-06-26T23:41:46 INFO gdb-server.c: Listening at *:4242...

```

**Running st-chipid**
`0x0000`

in conclusion I am essentially not able to do anything and this is preventing me from setting up my environment. Should be noted, when plugging other ARM devices in and connecting them properly according to the datasheet I am getting the same error so something is missing in terms of libs or there must be some file that needs to be edited?

xor-gate, 2019-06-27:
When the original tooling of ST doesn't work there must be some kind of hardware problem. And st-util does a readout and reads only zeroes. Which could be something wrong with SWD.

TravJav, 2019-06-27:
@xor-gate how do I proceed with my set up then?

TravJav, 2019-06-27:
and SWD? .. not familiar with these terms yet :)

TravJav, 2019-06-27:
I believe you're referring to one of the pins actually that are used by the St Link v2 programmer. Okay so if this is the case is there an alternative pin I can use that accomplishes the same function?

TravJav, 2019-06-27:
[image]
[image]

Jochen0x90h, 2019-06-27:
I have the very same setup. st-link --probe gives
```Found 1 stlink programmers
 serial: 543f7106656754533850093f
openocd: "\x54\x3f\x71\x06\x65\x67\x54\x53\x38\x50\x09\x3f"
  flash: 65536 (pagesize: 1024)
   sram: 20480
 chipid: 0x0410
  descr: F1 Medium-density device
```

I can flash it using st-flash, but with st-util I get this error:

```
2019-06-27T18:26:58 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2019-06-27T18:26:58 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2019-06-27T18:26:58 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2019-06-27T18:26:58 INFO flash_loader.c: Successfully loaded flash loader in sram

2019-06-27T18:26:58 INFO common.c: Starting verification of write complete
2019-06-27T18:26:58 INFO common.c: Flash written and verified! jolly good!
2019-06-27T18:26:58 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134218752 (0x8000400)
Flash page at addr: 0x08000400 erased
2019-06-27T18:26:58 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2019-06-27T18:26:58 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2019-06-27T18:26:58 INFO flash_loader.c: Successfully loaded flash loader in sram
2019-06-27T18:26:58 ERROR flash_loader.c: write error, count == 511
2019-06-27T18:26:58 ERROR common.c: stlink_flash_loader_run(0x8000400) failed! == -1
2019-06-27T18:26:58 ERROR gdb-server.c: cannot recv: -2
```

The cdcacm.c usb virtual com port example (https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f1/stm32-h103/usb_cdcacm where LED changed to PC13) with loopback does not work though, I get this error in dmesg
```[30770.719644] usb 1-10: new full-speed USB device number 30 using xhci_hcd
[30770.719807] usb 1-10: Device not responding to setup address.
[30770.927734] usb 1-10: Device not responding to setup address.
[30771.135665] usb 1-10: device not accepting address 30, error -71
```

My makefile is
```DEVICE = STM32F103C8
OPENCM3_DIR = libopencm3
OBJS += cdcacm.o

CFLAGS += -Os -ggdb3
CPPFLAGS += -MD
LDFLAGS += -static -nostartfiles
LDLIBS += -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group

include $(OPENCM3_DIR)/mk/genlink-config.mk
include $(OPENCM3_DIR)/mk/gcc-config.mk

.PHONY: clean all

all: binary.elf binary.bin

clean:
	$(Q)$(RM) -rf binary.* *.o

include $(OPENCM3_DIR)/mk/genlink-rules.mk
include $(OPENCM3_DIR)/mk/gcc-rules.mk
```

TravJav, 2019-06-28:
I should also note when checking kernel messages with 
```
$ dmesg | tail 
$[42220.561649] usb 1-2: Product: STM32 STLink
[42220.561651] usb 1-2: Manufacturer: STMicroelectronics
[42220.561653] usb 1-2: SerialNumber: UÿorfHH g
[42277.382320] usb 1-2: USB disconnect, device number 5
[42459.313228] usb 1-1: new full-speed USB device number 6 using xhci_hcd
[42459.454959] usb 1-1: New USB device found, idVendor=0483, idProduct=3748, bcdDevice= 1.00
[42459.454964] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[42459.454966] usb 1-1: Product: STM32 STLink
[42459.454969] usb 1-1: Manufacturer: STMicroelectronics
[42459.454971] usb 1-1: SerialNumber: UÿorfHH g

```

TravJav, 2019-06-28:
Results of running the below command 
```
$ st-util -v99
st-util 1.5.1
2019-06-28T00:37:40 DEBUG common.c: stlink current mode: dfu
2019-06-28T00:37:40 INFO usb.c: -- exit_dfu_mode
2019-06-28T00:37:40 DEBUG common.c: *** stlink_exit_dfu_mode ***
2019-06-28T00:37:40 DEBUG common.c: stlink current mode: mass
2019-06-28T00:37:40 DEBUG common.c: *** stlink_enter_swd_mode ***
2019-06-28T00:37:40 DEBUG common.c: *** looking up stlink version
2019-06-28T00:37:40 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2019-06-28T00:37:40 DEBUG common.c: stlink pid     = 0x3748
2019-06-28T00:37:40 DEBUG common.c: stlink version = 0x2
2019-06-28T00:37:40 DEBUG common.c: jtag version   = 0x21
2019-06-28T00:37:40 DEBUG common.c: swim version   = 0x7
2019-06-28T00:37:40 DEBUG common.c: *** stlink_jtag_reset ***
2019-06-28T00:37:40 DEBUG common.c: *** stlink_reset ***
2019-06-28T00:37:40 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2019-06-28T00:37:40 INFO common.c: Loading device parameters....
2019-06-28T00:37:40 DEBUG common.c: *** stlink_core_id ***
2019-06-28T00:37:40 DEBUG common.c: core_id = 0x00000000
2019-06-28T00:37:40 DEBUG common.c: *** stlink_read_debug32 0 is 0xe0042000
2019-06-28T00:37:40 DEBUG common.c: *** stlink_read_debug32 0 is 0x40015800
2019-06-28T00:37:40 WARN common.c: Invalid flash type, please check device declaration
2019-06-28T00:37:40 DEBUG common.c: *** set_swdclk ***
2019-06-28T00:37:40 DEBUG common.c: *** stlink_reset ***
2019-06-28T00:37:40 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2019-06-28T00:37:40 INFO gdb-server.c: Chip ID is 00000000, Core ID is  00000000.
2019-06-28T00:37:40 INFO gdb-server.c: Listening at *:4242..

```

TravJav, 2019-06-28:
@xor-gate I think this board has r/w permissions do you have a command that can wipe the board so I may flash and debug it? the more I read the more I am convinced it's showing symptoms of this, i'm just not familiar with the args that go in.

Jochen0x90h, 2019-06-28:
Have you tried to remove the cover of the st-link dongle that may touch the components inside? So remove it and then try st-info --probe agian. Also check if the soldering of the swclk and swdio pins is ok

Jochen0x90h, 2019-06-28:
I think I found the solution to your problem: I flashed an arduino bootloader (https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/bootloader_only_binaries/generic_boot20_pc13.bin) and then I got the same output like you for st-info --probe. I then set the upper jumper to 1 to enable the intermal bootloader and thereby disable the arduino bootloader. Then st-info --probe looked normal again. I was able to flash the bootloader again, then changed the jumper back to 0. The arduino bootloader seems to work, no errors in dmesg and lsusb shows device 1eaf:0004, even after recompiling it with my installed gcc (6.3.1). I still have to find out why cdcacm.c does not work.

TravJav, 2019-06-28:
[image]

@Jochen0x90h 
Like this correct? I did this last night actually and noticed the green blinking light was not present anymore instead; it just had the solid red light. I ran the above commands again and it still can't find the chip.  Odd thing is I installed the needed software on my Mac as well and I end up with the same result on both machines.

TravJav, 2019-06-28:
also, plugging in my rasp pi to the correct connectors and another board yields the same error i described

TravJav, 2019-06-28:
when trying to flash using the procedure with st-flash I get this with the jumpers in the same position as the photo [travjav@rey]: ~/Development/Invictus-```
board/STM32F103C8T6>$ st-flash write main.bin 0x08000000
st-flash 1.5.1
2019-06-28T11:23:24 INFO common.c: Loading device parameters....
2019-06-28T11:23:24 WARN common.c: Invalid flash type, please check device declaration
  core status: unknown
Unknown memory region
I sent above 

```

Jochen0x90h, 2019-06-28:
The blinking light disappeared because now the internal serial bootloader (factory programmed) is now active. But if it didn't do the trick then I unfortunately can't help you. Did you remove the cover of the st-link dongle?

TravJav, 2019-06-28:
I did try removing the cover yes.. I guess I could buy another one since they're so cheap and hopefully it's a defect.

TravJav, 2019-06-28:
FOUND IT! this is the most insidious thing ever but the pin out was WRONG on the cover! I will include the link here from the amazon review I bought the product from, but in case it gets removed or something.. take the cover off and actually look at the pin out!! the cover was not correct the swc had to be swapped with the swio.

TravJav, 2019-06-28:
> The blinking light disappeared because now the internal serial bootloader (factory programmed) is now active. But if it didn't do the trick then I unfortunately can't help you. Did you remove the cover of the st-link dongle?

Can you please clarify this: so moving the pins to the position shown in the last image I included, these need to remain in  this position when debugging always? then can be put back to 0 position when the program is executing/ live?

closed 2019-06-29 by TravJav

Jochen0x90h, 2019-07-01:
The jumper can also be set back to 0 when debugging, it only has to be set to 1 if there is some program flashed that somehow prevents flasing over swd, for example by using these pins as output.
But great that you found the problem

RockyF, 2019-10-16:
I looked up a lot of information, and finally got the answer here. It was the reverse of the text of the boot jumper!
Thank you very much.
@TravJav @Jochen0x90h