stlink-org/stlink #814: "Verification of flash failed" for STM32L4S5

etxmsol, 2019-06-27:
- [Stlink/v2-onboard] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [V2.J32.M22 STM32 Debug+Mass storage] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [Linux Raspbian] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ v1.5.1-30-g84f63d2] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [st-flash ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [STM32L4S5VIT6, custom board] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

I have two of the same PCB. On one, the flashing works and the flow is as following:

pi@raspberrypi:~ $ st-flash write BaseSTM32L4S5.bin 0x8000000
st-flash 1.5.1-30-g84f63d2
2019-06-27T22:38:15 INFO common.c: Loading device parameters....
2019-06-27T22:38:15 INFO common.c: Device connected is: L4Rx device, id 0x100f6470
2019-06-27T22:38:15 INFO common.c: SRAM size: 0xa0000 bytes (640 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 4096 bytes
2019-06-27T22:38:15 INFO common.c: Attempting to write 652720 (0x9f5b0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0809f000 erasedEraseFlash - Sector:0x8 Size:0x1000
2019-06-27T22:38:20 INFO common.c: Finished erasing 160 pages of 4096 (0x1000) bytes
2019-06-27T22:38:20 INFO common.c: Starting Flash write for F2/F4/L4
2019-06-27T22:38:20 INFO flash_loader.c: Successfully loaded flash loader in sram
Target voltage (9 mV) too low for 32-bit flash, using 8-bit flash writes
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 30128
2019-06-27T22:38:36 INFO common.c: Starting verification of write complete
2019-06-27T22:38:42 INFO common.c: Flash written and verified! jolly good!

On the other one it fails. The printout is below:

pi@raspberrypi:~ $ st-flash write BaseSTM32L4S5.bin 0x8000000
st-flash 1.5.1-30-g84f63d2
2019-06-27T22:40:37 INFO common.c: Loading device parameters....
2019-06-27T22:40:37 INFO common.c: Device connected is: L4Rx device, id 0x100f6470
2019-06-27T22:40:37 INFO common.c: SRAM size: 0xa0000 bytes (640 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 4096 bytes
2019-06-27T22:40:37 INFO common.c: Attempting to write 652720 (0x9f5b0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0809f000 erasedEraseFlash - Sector:0x8 Size:0x1000
2019-06-27T22:40:43 INFO common.c: Finished erasing 160 pages of 4096 (0x1000) bytes
2019-06-27T22:40:43 INFO common.c: Starting Flash write for F2/F4/L4
2019-06-27T22:40:43 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 30128
2019-06-27T22:40:58 INFO common.c: Starting verification of write complete
2019-06-27T22:40:58 ERROR common.c: Verification of flash failed at offset: 36864
stlink_fwrite_flash() == -1

```
 ERROR common.c: Verification of flash failed at offset: 36864
stlink_fwrite_flash() == -1
```

Expected/description:
should flash the code without errors

xor-gate, 2019-06-27:
There are two differences:

* One which works seems to detect incorrect target voltage and uses 8bit writes
  * `Target voltage (9 mV) too low for 32-bit flash, using 8-bit flash writes`
* One which doesn't work seem to detect correct target voltage and uses 32bit writes

I think there is something wrong in the flash loader code when writing in 32bit mode, and the verify is done after it. Or the verify is broken in the 32bit mode. Long story short, there is something wrong with the st-flash tool.

etxmsol, 2019-06-27:
oh, thanks for quick answer. That was exactly my thought too, but proved wrong. This 32-bit mode is not the reason. I've discovered that both 8-bit and 32-bit modes may or may not fail. What i see consistently is when I flash using windows (stlink drag-drop) makes the future flashes from my Linux go fine, both 8-bit and 32-bit. I agree with you it is the st-flash,. Probably not yet made compatible with the STM32L4S5.

etxmsol, 2019-06-29:
further analysis reveals that it is the writing to the flash that  does not work. The reading is fine. So when I previously successfully flashed with Windows st-Link utility, "flashing" the same file with st-flash results in success. Actually it does not do any flashing (nor erasing?), just reading and comparing. As soon as the file to flash changes, the verification does not work. It is a newer binary, but still the old flash content, created by st-link utility on Windows.

etxmsol, 2019-06-29:
I seem to have sorted this out. All boiled down to adding the (sibling) device STLINK_CHIPID_STM32_L4RX to wherever other L4s were in the if statements in the common.c. I do not know if it fixes the issue completely, but it certainly does for my needs.

One thing I had to comment out was the voltage measurement in common.c:
        } else {
//            /* L4 does not have a byte-write mode */
//            int voltage = stlink_target_voltage(sl);
//            if (voltage == -1) {
//                printf("Failed to read Target voltage\n");
//                return voltage;
//            } else if (voltage < 1710) {
//                printf("Target voltage (%d mV) too low for flash writes!\n", voltage);
//                return -1;
//            }
        }

my voltage reading is always 0, and I currently do not have time to investigate why.

cheers

brad-natelborg, 2019-10-11:
This is fixed with merge #839

Nightwalker-87, 2020-02-21:
Closed by #839

Nightwalker-87, 2020-02-21:
Closed by #839

closed 2020-02-21 by Nightwalker-87