stlink-org/stlink #815: can i use stlink as a  swo viewer under linux?

chopin1998, 2019-07-02:


xor-gate, 2019-07-14:
Currently the texane/stlink project doesn't support SWO. But it could be implemented of course.

xor-gate, 2019-07-14:
You could use OpenOCD for that with the following settings:

```
itm port 0 on
tpiu config internal <logfile> uart off <cpu_freq_herz>
```

For some other readers: https://mcuoneclipse.com/2016/10/17/tutorial-using-single-wire-output-swo-with-arm-cortex-m-and-eclipse/

chopin1998, 2019-07-15:
> You could use OpenOCD for that with the following settings:
> 
> ```
> itm port 0 on
> tpiu config internal <logfile> uart off <cpu_freq_herz>
> ```
> 
> For some other readers: https://mcuoneclipse.com/2016/10/17/tutorial-using-single-wire-output-swo-with-arm-cortex-m-and-eclipse/

thank you very much!

closed 2020-12-09 by Nightwalker-87