stlink-org/stlink #818: Verification of flash failed - STM32F051RBTx

Goldenstarc, 2019-07-23:
same as others here is the problem: 

`2019-07-23T16:56:36 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2019-07-23T16:56:36 INFO flash_loader.c: Successfully loaded flash loader in sram
 18/18 pages written
2019-07-23T16:56:37 INFO common.c: Starting verification of write complete
2019-07-23T16:56:37 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
`

There is no problem with programming or erasing.

AntoinedeChassey, 2019-08-21:
Same for STM32L051 series
Verification fails

```2019-08-21T11:50:03 INFO common.c: Starting verification of write complete
2019-08-21T11:50:03 DEBUG common.c: *** stlink_read_mem32 ***
data_len = 128 0x80
 80 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

2019-08-21T11:50:03 ERROR common.c: Verification of flash failed at offset: 0
2019-08-21T11:50:03 DEBUG common.c: *** stlink_read_debug32 20002900 is 0x8000000
2019-08-21T11:50:03 DEBUG common.c: *** stlink_write_reg
data_len = 2 0x2
 81 00

2019-08-21T11:50:03 DEBUG common.c: *** stlink_read_debug32 801b715 is 0x8000004
2019-08-21T11:50:03 DEBUG common.c: *** stlink_write_reg
data_len = 2 0x2
 81 00

2019-08-21T11:50:03 DEBUG common.c: *** stlink_run ***
stlink_fwrite_flash() == -1
2019-08-21T11:50:03 DEBUG common.c: *** stlink_exit_debug_mode ***
2019-08-21T11:50:03 DEBUG common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2019-08-21T11:50:03 DEBUG common.c: *** stlink_close ***

Nightwalker-87, 2020-03-25:
@Goldenstarc @AntoinedeChassey: What version of the stlink-tools did you use? Please provide more information on used hardware and build environment.

bootsector, 2020-06-24:
I have this very same issue with this STM32F1 clone: https://robotdyn.com/black-pill-apm32f103cb-128kb-flash-20kb-sram-stm32-compatible-arm-cortexr-m3-mcu-mini-board.html

Nightwalker-87, 2020-06-24:
Which version of the tools do you use? Can you give us some information on your environment?
You can have a look what info would be relevant to us at our issue template (--> new issue --> Bug Report), without submitting a new issue of course. Simply copy the filled out form into a new posting here. This would be helpful as this old ticket does not provide any detailed info yet, apart from the error output. We may then address this topic in an appropriate way.

bootsector, 2020-06-24:
I'm happy to provide as much info as you need. Let me know if you need any further tests!

- Programmer/board type: Cheap ST-LINV V2 clone from AliExpress
- Programmer firmware version: How can I check this?
- Operating system and version: Ubuntu 18.04.4 LTS (64-bit)
- **Stlink tools version** and/or git commit hash: origin/develop commit hash eeb3190444f1bedc2b63de536793b4e2e8c59384
- Stlink commandline tool name: st-flash
- Target chip (and board if applicable): APM32F103CB - https://robotdyn.com/black-pill-apm32f103cb-128kb-flash-20kb-sram-stm32-compatible-arm-cortexr-m3-mcu-mini-board.html

Commandline-Output:

```
st-flash --debug write HIDBOOTLOADER.bin 0x8000000
st-flash 1.6.0-390-geeb3190
2020-06-24T07:29:44 DEBUG common.c: *** looking up stlink version
2020-06-24T07:29:44 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-06-24T07:29:44 DEBUG common.c: stlink pid     = 0x3748
2020-06-24T07:29:44 DEBUG common.c: stlink version = 0x2
2020-06-24T07:29:44 DEBUG common.c: jtag version   = 0x11
2020-06-24T07:29:44 DEBUG common.c: swim version   = 0x4
2020-06-24T07:29:44 DEBUG common.c: stlink current mode: dfu
2020-06-24T07:29:44 DEBUG usb.c: -- exit_dfu_mode
2020-06-24T07:29:44 DEBUG common.c: *** stlink_exit_dfu_mode ***
2020-06-24T07:29:44 DEBUG usb.c: JTAG/SWD freq set to 0
2020-06-24T07:29:44 DEBUG common.c: *** set_swdclk ***
2020-06-24T07:29:44 DEBUG common.c: stlink current mode: mass
2020-06-24T07:29:44 DEBUG common.c: *** stlink_enter_swd_mode ***
2020-06-24T07:29:44 DEBUG common.c: *** stlink_jtag_reset ***
2020-06-24T07:29:44 DEBUG common.c: *** stlink_reset ***
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-06-24T07:29:44 DEBUG common.c: Loading device parameters....
2020-06-24T07:29:44 DEBUG common.c: *** stlink_core_id ***
2020-06-24T07:29:44 DEBUG common.c: core_id = 0x2ba01477
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 20036410 is 0xe0042000
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 ffff0080 is 0x1ffff7e0
2020-06-24T07:29:44 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
2020-06-24T07:29:44 DEBUG common.c: stlink current mode: debug (jtag or swd)
2020-06-24T07:29:44 DEBUG common.c: stlink current mode: debug (jtag or swd)
2020-06-24T07:29:44 DEBUG common.c: *** stlink_force_debug_mode ***
2020-06-24T07:29:44 DEBUG common.c: *** stlink_status ***
2020-06-24T07:29:44 DEBUG usb.c: core status: 00030003
2020-06-24T07:29:44 DEBUG common.c:   core status: halted
file HIDBOOTLOADER.bin md5 checksum: 882d355c48f5e4c65cd0e4f15ecd86e7, stlink checksum: 0x0003ddb0
2020-06-24T07:29:44 INFO common.c: Attempting to write 3060 (0xbf4) bytes to stm32 address: 134217728 (0x8000000)
2020-06-24T07:29:44 DEBUG common.c: *** stlink_core_id ***
2020-06-24T07:29:44 DEBUG common.c: core_id = 0x2ba01477
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 0 is 0x4002200c
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 80 is 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 45670123 to 0x40022004
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 cdef89ab to 0x40022004
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 0 is 0x40022010
2020-06-24T07:29:44 DEBUG common.c: Successfully unlocked flash
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 0 is 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 0 to 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 0 is 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 2 to 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 8000000 to 0x40022014
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 2 is 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 42 to 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 1 is 0x4002200c
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 1 is 0x4002200c
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 1 is 0x4002200c
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 1 is 0x4002200c
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 20 is 0x4002200c
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 2 is 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 82 to 0x40022010
2020-06-24T07:29:44 INFO common.c: Flash page at addr: 0x08000000 erased
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 20 is 0x4002200c
2020-06-24T07:29:44 DEBUG common.c: *** stlink_read_debug32 82 is 0x40022010
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 45670123 to 0x40022004
[... 172 lines cut ...]
2020-06-24T07:29:44 DEBUG common.c: data_len = 1024 0x400
2020-06-24T07:29:44 DEBUG common.c:  002020-06-24T07:29:44 DEBUG common.c:  502020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  c12020-06-24T07:29:44 DEBUG common.c:  092020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  ff2020-06-24T07:29:44 DEBUG common.c:  c12020-06-24T07:29:44 DEBUG comm [... cut ...]
2020-06-24T07:29:44 DEBUG common.c: data_len = 2 0x2
2020-06-24T07:29:44 DEBUG common.c:  802020-06-24T07:29:44 DEBUG common.c:  002020-06-24T07:29:44 DEBUG common.c: 

2020-06-24T07:29:44 DEBUG common.c: *** stlink_run ***
stlink_fwrite_flash() == -1
2020-06-24T07:29:44 DEBUG common.c: *** stlink_exit_debug_mode ***
2020-06-24T07:29:44 DEBUG common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2020-06-24T07:29:44 DEBUG common.c: *** stlink_close ***

```

Expected/description:

Flash verification should have no issues.

Nightwalker-87, 2020-06-24:
>     * Programmer firmware version: How can I check this?

As you are under linux, you may use the ST-Link firmware upgrade tool based on Java to read out the current version:
https://www.st.com/en/development-tools/stsw-link007.html

As I could convince myself previously this also works with non-genuine ST programmers and boards.
Also these can safely be updated to the latest available version.
It even works with CKS32F1xxx clone ICs as can be found on some Bluepill-Boards and a few cheap STM32 programmers.

bootsector, 2020-06-24:
Thanks!

Here's the programmer info:

https://i.imgur.com/H3S5QRL.png

bootsector, 2020-06-24:
I've also upgraded to the suggested version and got the same issue. Let me know if you need additional info. Thanks for looking at this!

Nightwalker-87, 2020-06-24:
That's ok, I'd would have done the same. Please give us some time.
I hope that there will be some help and input from the community on this as well.

Ant-ON, 2021-03-25:
@bootsector Can you test st-flash from the `develop` branch?
@Nightwalker-87 Most likely closes by #1113

bootsector, 2021-03-25:
@Ant-ON @Nightwalker-87 I've confirmed that IT WORKS NOW! :D

Thank you guys for fixing this! Amazing job!

Nightwalker-87, 2021-03-25:
@Ant-ON is indeed giving valuable input to the project. :1st_place_medal:

Nightwalker-87, 2021-03-25:
Closed by #1113.

closed 2021-03-25 by Nightwalker-87