stlink-org/stlink #823: st-link can't work in STM32H743 ?

programdog, 2019-07-29:
hello
i use a board named NUCLEO-H743ZI , the board from STMicroelectronics company , the mcu is STM32H743ZIT6U, but when i ust st-link tools in ubuntu18, the tools can not work, the info below:

```
$ lsusb
Bus 001 Device 059: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)
```
```
$ st-info --probe
Found 1 stlink programmers
 serial: 303637324646353034393535383537353637313934373530
openocd: "\x30\x36\x37\x32\x46\x46\x35\x30\x34\x39\x35\x35\x38\x35\x37\x35\x36\x37\x31\x39\x34\x37\x35\x30"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device
```
```
$ st-info --chipid
0x0000
```
```
$ st-info --flash
0x0
```
maybe your st-link tools not support the STM32H743 ?
all of the st-link cmd can work in my stm32f4discovery ,but STM32H743 can not.
I don't know
could you solve this question? 

THX
GOOD DAY

ted0xff, 2019-08-28:
Hello,

The Nucleo STM32H743  board embed a STLINK-V3E, seems to not be already supported yet, see #820

programdog, 2019-09-27:
> Hello,
> 
> The Nucleo STM32H743 board embed a STLINK-V3E, seems to not be already supported yet, see #820
okay~ , i think so too, thx

closed 2020-10-21 by Nightwalker-87