stlink-org/stlink #826: MAINTAINER WANTED

xor-gate, 2019-08-17:
Dear users,

Currently the project has no active maintainer as @xor-gate has stepped up from active maintaining the project and @texane has also other life priorities. This issue is used for discussion of a new possible maintainers. Feel free to ask questions.

Thanks for all the trust, time and effort using the texane/stlink software!

Kind regards,
Jerry Jacobs

slyshykO, 2019-08-20:
if no one else promotes yourself, I can try.
My disadvantages:
- weak English;
- primary OS - win10.

texane, 2019-08-21:
Hi Slyshyk,

thanks for proposing your help.

As a first start, I propose that when you see issues reported, you try fix
them and push a merge request. I can review that. After some time, if the
process goes well then we will see to grant you the 'maintainer' role.

Also, you can answer to the user questions, to see how well you interact
with them.

Is it good for you ?

Cheers,

slyshykO, 2019-08-21:
Yes, it is Ok for me. 
I'll try

bmarvo, 2019-10-07:
Any progress on finding a maintainer?

EmBitz, 2019-10-26:
I used Texane a lot in the past and have forked it several times for building different flavours. But one of the annoying things was always the maintenance of this tool. New devices are coming in a very fast rate and every time  all kind of patches have to be applied and rebuilds are needed even for the smallest changes. Very time consuming. But I liked the idea of a dedicated cortex-m tool instead of OpenOCD which is often to bloated to use. Texane is much easier to understand.

But every time that you started the tool and the device ID was "unknown" even if the family was already supported, that was frustrating.

That's the reason I started all over again with the idea of a dedicated cortex-m tool which can be maintained by the end-user in an easy way (no rebuild) to keep track of new devices.  So I build EBlink (C++), an independent vendor tool easy maintainable. And with [EBlink](https://github.com/EmBitz/EBlink) you can use your stlink with any vendor (e.g. Silabs).   Currently only the windows binary and scripts are on github (as transport) but I could push that whole project if people are interested. 

Texane was very helpful in the beginning and I'm very thankful for that.

Nightwalker-87, 2020-02-17:
@xor-gate @texane : I could imagine to opt in for documentation updates and/or fixes in a role as a co-maintainer. This project definitely needs to stay alive and see active maintenance.

bmarvo, 2020-02-17:
@Nightwalker-87 : Agreed. I think we (people who have an active interest in this project) can keep it afloat during the dark times. I'm willing to contribute as well. The PRs are stacking up!

Nightwalker-87, 2020-02-17:
AFAIK there are also a few issues that seem to have been solved in the meanwhile which still remain open. I'd prefer if this project continued to be maintained in place instead of a clone, because it is already widely referenced. @texane: Would it be possible to grand some extended user rights for this project without a transfer of ownership? This could be an approach to get in contributions and active users as well as interaction and collaboration with the former maintainers.

texane, 2020-02-19:
Hi everyone, sorry for the delay ... I added both of you to the project as maintainers. Be careful as you have full power ... but it is clear that this project needs help from the community to keep going. We will see how it goes, thanks for helping !

Nightwalker-87, 2020-02-22:
> The PRs are stacking up!

@bmarvo: no longer ;-) Also we had enough solved issues and fixes for a new release, which seemed somehow overdue. By now I have further reviewed most of the remaining issues and sorted them to milestones and spent some efforts on correct tagging and tracking. Looking forward to various author, submitter, and contributor-feedback now. Maybe we will see some light at the end of the tunnel soon...

bmarvo, 2020-02-22:
Excellent work! Once the dust settles maybe we can take a look at some areas that need some attention and come up with ideas for improvements. I'm sure there are some areas of the repo that can use some love. Also, I'd like to hear your thoughts on managing future releases if that sort of thing interests you.

Nightwalker-87, 2020-02-22:
@bmarvo: sure, but let's refer to that in detail at a later point in time. For now there is a v1.6.1 release scheduled for June and we'll see how much input we can get for that until then.
@texane: Can you ltake a look at the latest release and sign it? Thx.

I suggest to leave this issue here open for another few days and let texane decide when he desires to finally close it. Talking about me, i'm absolutely fine with continued project maintenance for a longer period together with bmarvo.

Nightwalker-87, 2020-02-27:
@bmarvo: Status update on reviewing: As of today I am down to 250 unreviewed issues incl. PRs compared to 722 in total. Need to take a break and continue with the remaining tickets in March. Can we keep new issues and PRs without a milestone until then? That would be a great help.

bmarvo, 2020-02-27:
Sure thing. Have you even slept in the past week? :)

texane, 2020-02-27:
Thanks for your contribution !

Nightwalker-87, 2020-03-11:
I have opened a new ticket (#874) which is dedicated for upcoming project news and updates.
This "special issue" should remain open for an indefinite period of time and avoid any extension of this ticket beyond the original topic.

@texane: What are your general thoughts on the current progress of the project and it's development in the recent weeks? Unless there are any specific things to address, I think it may be about time to close this topic.

Nightwalker-87, 2020-04-08:
I am closing this topic now, as new maintainers have been nominated.

Further this project has now been transferred to a git-organisation today, as agreed between @texane, @xor-gate and @Nightwalker-87. This will ease collaboration and possible transitions between maintainers in the future, as this project is now hosted within a shared account.

**However we should ensure for the future that there are always at least two active maintainers with full access.**

closed 2020-04-08 by Nightwalker-87