stlink-org/stlink #827: STLINK not detected in QtCreator, MacOS

randoom15, 2019-08-29:
Hello everybody,

I have installed stlink using `brew` and checked it's installation with `pkg-config`.
I had the confirmation that everything seemed fine.
The problem is when I try to compile something i get the error :

**stlink development package not found**

I cannot find anything related about this issue.
Thanks in advance.

> MacOS X Mojave 10.14.6

Nightwalker-87, 2020-02-22:
Please re-verify if the problem still exists.

randoom15, 2020-02-24:
Sorry i cannot give feedback since i am not working on this project anymore

Nightwalker-87, 2020-02-24:
ok, no problem.
Thx for your quick response though, which allows to set the status of the ticket correctly. :+1:

Nightwalker-87, 2020-04-07:
The installation procedure for the `stlink` tools via `homebrew` appears to be working as described in our project documentation. Since the provided information in this issue is not sufficient and there is no possibility to reproduce it, we can not investigate this any further. Thus I am closing this ticket now. Should there be any similar problem in future, please feel free to submit a new ticket.

closed 2020-04-07 by Nightwalker-87