stlink-org/stlink #831: [gdb] target state locked in "running" after step instruction with st-util and gdb

alexonea, 2019-09-14:
Dear developers and maintainers of ```stlink```,

I am trying to debug a simple program for the STM32F769I-DISCO board with Stlink/v2. I am using an Ubuntu machine, st-util for the gdb server and gdb-multiarch for the client. The connection to the target and loading of the program are successful, as well as the detection of the target in st-util (at least it seems so). However, when I try to step through the first instruction (```stepi``` in gdb), it hangs, while the target seems to enter into continuous execution mode. Verbose output of ```st-util``` reveals that indeed, the core status is ```running```.

Attached version information and output logs.

**Ubuntu**:
```
$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="19.04 (Disco Dingo)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 19.04"
VERSION_ID="19.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco

$ uname -a
Linux 5.0.0-27-generic #28-Ubuntu SMP Tue Aug 20 19:53:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```
**st-util**:
```
$ st-util --version
v1.5.1
```
**gdb-multiarch**:
```
$ gdb-multiarch --version
GNU gdb (Ubuntu 8.2.91.20190405-0ubuntu3) 8.2.91.20190405-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

**Target identification**:
```
$ st-info --probe
Found 1 stlink programmers
 serial: 303636394646353735323531373137
openocd: "\x30\x36\x36\x39\x46\x46\x35\x37\x35\x32\x35\x31\x37\x31\x37"
  flash: 2097152 (pagesize: 2048)
   sram: 524288
 chipid: 0x0451
  descr: F76xxx device
```

**STLink Firmware**:
```
(taken from STLinkUpgrade 3.2.12 GUI on Ubuntu)
ST-Link ID: 0669FF575251717867092414
Firmware version: V2J32M22
```

- [x] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [x] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [x] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [x] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [x] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [x] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

Reproduction steps:
1. Run ```st-util -v99```
2. Run ```gdb-multiarch program.elf``` and connect to target, plus load the program into memory

[... 350 lines cut ...]
2019-09-14T11:53:12 DEBUG gdb-server.c: send: 92635738
2019-09-14T11:53:12 DEBUG gdb-server.c: recv: m0,4
2019-09-14T11:53:12 DEBUG common.c: *** stlink_read_mem32 ***
2019-09-14T11:53:12 DEBUG gdb-server.c: send: 92635738
2019-09-14T11:53:12 DEBUG gdb-server.c: recv: m0,4
2019-09-14T11:53:12 DEBUG common.c: *** stlink_read_mem32 ***
2019-09-14T11:53:12 DEBUG gdb-server.c: send: 92635738
2019-09-14T11:53:13 DEBUG gdb-server.c: recv: m0,4
2019-09-14T11:53:13 DEBUG common.c: *** stlink_read_mem32 ***
2019-09-14T11:53:13 DEBUG gdb-server.c: send: 92635738
2019-09-14T11:53:13 DEBUG gdb-server.c: recv: m0,4
2019-09-14T11:53:13 DEBUG common.c: *** stlink_read_mem32 ***
2019-09-14T11:53:13 DEBUG gdb-server.c: send: 92635738
2019-09-14T11:53:13 DEBUG gdb-server.c: recv: p19
2019-09-14T11:53:13 DEBUG common.c: *** stlink_read_reg
2019-09-14T11:53:13 DEBUG common.c:  (16) ***
2019-09-14T11:53:13 DEBUG usb.c: r_idx (16) = 0x00000003
2019-09-14T11:53:13 DEBUG gdb-server.c: send: 03000000
2019-09-14T11:53:13 DEBUG gdb-server.c: recv: qL1200000000000000000
2019-09-14T11:53:13 DEBUG gdb-server.c: send: 
```

Please note that at line ```2019-09-14T11:53:12 DEBUG common.c: *** stlink_force_debug_mode ***``` of output, I pressed ^C to stop execution in GDB.

This seems to be in connection to issue #627 however, premises differ since now we are using a Stlinkv2 with a STM32F7x. I also managed to reproduce the same issue on a STM32VLDISCOVERY. This started happening after updating the firmware to the specified version above. Before the update, the step instruction command worked fine.

Nightwalker-87, 2020-03-19:
@alexonea: Please verify if the problem still exists in Release v1.6.0.

Nightwalker-87, 2020-06-28:
ping @alexonea

alexonea, 2020-06-29:
I don't have the (hardware) resources to test at the moment. I will try in a few days time. If waiting is no longer an option, the issue can be closed and re-opened in case the problem is rediscovered during testing.

Nightwalker-87, 2020-06-29:
That should be fine, thx.

alexonea, 2020-06-30:
I checked with release `v1.6.0` of st-util and the same firmware version of the STLinkv2. No issues detected. Seems fine.

Nightwalker-87, 2020-07-01:
Great, so this problem seems to be resolved. Thanks for reporting and for the feedback! :+1:

Nightwalker-87, 2020-07-01:
I'll close this issue as soon as it is correctly referenced in our bug-tracking system.

closed 2020-07-04 by Nightwalker-87