stlink-org/stlink #835: ST Link Locks up when not connected to a device over the JTAG pins

nzsmith1, 2019-09-19:
**Problem**
If I run any command st-info, st-flash, without a device connected to the JTAG pins, every command after that fails until I remove the ST-Link V2 from USB and plug it back in.

**Expected/description:**
I expect that when a device is not connected due to power loss, connection loss or plain absentmindedness, when I do actually connect a device I am still able to communicate with it when I finally connect it and run the command again

**OS**
```
pi@raspberrypi:~/stlink-repo/build/Release $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

**Programmer:** 
STLinkV2
F/W V2.J32.S7

**commit hash:**
c3577b578ff24c2f5a9c2cb7430986a931c9afcd however it is happening on previous commits as well atleast as far back as d040db5585d33dc36bd8ae093c024583ef1b4f74

**Target:**
STM32L071KZU6
```
pi@raspberrypi:~/stlink-repo/build/Release $ ./st-info --probe
Found 1 stlink programmers
 serial: 563f7306513f52501639133f
openocd: "\x56\x3f\x73\x06\x51\x3f\x52\x50\x16\x39\x13\x3f"
  flash: 196608 (pagesize: 128)
   sram: 20480
 chipid: 0x0447
  descr: L0x Category 5 device
```

- [ x ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ x ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ X ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ X ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ X ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ X ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

**Output:**

```
// STM Device connected to JTAG
pi@raspberrypi:~/stlink-repo/build/Release $ ./st-flash --reset --format ihex write ~/Puck-1.4.1.hex
st-flash 1.5.1-38-gc3577b5
2019-09-19T14:19:39 INFO usb.c: -- exit_dfu_mode
2019-09-19T14:19:39 INFO common.c: Loading device parameters....
2019-09-19T14:19:39 INFO common.c: Device connected is: L0x Category 5 device, id 0x20086447
2019-09-19T14:19:39 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x30000 bytes (192 KiB) in pages of 128 bytes
2019-09-19T14:19:39 INFO common.c: Ignoring 20 bytes of 0x00 at end of file
2019-09-19T14:19:39 INFO common.c: Attempting to write 34996 (0x88b4) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08008880 erased
2019-09-19T14:19:44 INFO common.c: Finished erasing 274 pages of 128 (0x80) bytes
2019-09-19T14:19:44 INFO common.c: Starting Half page flash write for STM32L core id
2019-09-19T14:19:44 INFO flash_loader.c: Successfully loaded flash loader in sram
546/546 halfpages written
2019-09-19T14:19:52 INFO common.c: Starting verification of write complete
2019-09-19T14:19:52 INFO common.c: Flash written and verified! jolly good!
pi@raspberrypi:~/stlink-repo/build/Release $

-----------------------------------------------------------------------------------------
[... 164 lines cut ...]
Sep 19 14:20:18 raspberrypi kernel: [17555.177666] usb 1-1.3: SerialNumber: VÿsQRP9
Sep 19 14:20:18 raspberrypi mtp-probe: checking bus 1, device 19: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
Sep 19 14:20:18 raspberrypi mtp-probe: bus: 1, device: 19 was not an MTP device
Sep 19 14:20:18 raspberrypi mtp-probe: checking bus 1, device 19: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
Sep 19 14:20:18 raspberrypi mtp-probe: bus: 1, device: 19 was not an MTP device
```

kernal logs:
```
Sep 19 14:19:35 raspberrypi kernel: [17512.033914] usb 1-1.3: new full-speed USB device number 18 using dwc_otg
Sep 19 14:19:35 raspberrypi kernel: [17512.167748] usb 1-1.3: New USB device found, idVendor=0483, idProduct=3748, bcdDevice= 1.00
Sep 19 14:19:35 raspberrypi kernel: [17512.167765] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 19 14:19:35 raspberrypi kernel: [17512.167774] usb 1-1.3: Product: STM32 STLink
Sep 19 14:19:35 raspberrypi kernel: [17512.167784] usb 1-1.3: Manufacturer: STMicroelectronics
Sep 19 14:19:35 raspberrypi kernel: [17512.167793] usb 1-1.3: SerialNumber: VÿsQRP9
Sep 19 14:20:17 raspberrypi kernel: [17553.721662] usb 1-1.3: USB disconnect, device number 18
Sep 19 14:20:18 raspberrypi kernel: [17555.044287] usb 1-1.3: new full-speed USB device number 19 using dwc_otg
Sep 19 14:20:18 raspberrypi kernel: [17555.177621] usb 1-1.3: New USB device found, idVendor=0483, idProduct=3748, bcdDevice= 1.00
Sep 19 14:20:18 raspberrypi kernel: [17555.177637] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 19 14:20:18 raspberrypi kernel: [17555.177647] usb 1-1.3: Product: STM32 STLink
Sep 19 14:20:18 raspberrypi kernel: [17555.177656] usb 1-1.3: Manufacturer: STMicroelectronics
Sep 19 14:20:18 raspberrypi kernel: [17555.177666] usb 1-1.3: SerialNumber: VÿsQRP9
```

Thanks

zx81a, 2019-11-11:
With openocd such scenario works fine.

Nightwalker-87, 2020-06-24:
Please test this with Release v1.6.0 and v1.6.1.

nzsmith1, 2020-06-25:
I am going to do my level best, but due to whats going on w/ COVID I am away from the office until September and all of my tools are there.  Feel free to close and I can reopen in September if its still an issue.

Nightwalker-87, 2020-06-25:
Hi @nzsmith1. Thx for your reply. That's ok, we'll leave this open anyway until a solution has been found.
Possibly someone else may also show interest in the meanwhile.

nzsmith1, 2020-09-08:
Hi @Nightwalker-87 - Got back in the office today, 1.6.0 does not work, same as above, however 1.6.1 seems to have resolved the issue!  Do you need any logs or are you good?

Thanks for the work on this project, it is fantastic!

Nightwalker-87, 2020-09-08:
Hi @nzsmith1. Thx for the feedback. Indeed some Logs would be helpful for verification before closing this issue as resolved.

stappersg, 2020-09-09:
On Tue, Sep 08, 2020 at 01:46:38PM -0700, nightwalker-87 wrote:
> Indeed some Logs would be helpful for verification

How / where to supply those log files?

Nightwalker-87, 2020-09-10:
Well just such like the ones he posted above. Nothing more nothing less. We should be absolutely fine with that.

Nightwalker-87, 2020-09-15:
@nzsmith1: Can you provide us some final logs so that we can close this issue with some verification for the public?

nzsmith1, 2020-09-15:
Yupp absolutely, it's on my list of things to do, hoping to get to it in the next day or so.

nzsmith1, 2020-09-16:
Here are the logs from 1.6.1 which shows the bug gone, let me know if there is anything else I can provide.

**OS**
pi@mfg-018:~/stlinknew/stlink/build/Release/bin$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Programmer:
STLinkV2
F/W V2.J32.S7

commit hash tested that is fixed:
6aa08a659a701feb9c08a4cb1628b92645f41309

Target:
STM32L071KZU6

`pi@mfg-018:~/stlinknew/stlink/build/Release/bin$ ./st-info --probe
Found 1 stlink programmers
 serial:     313f6b064d4e353814310143
 hla-serial: "\x31\x3f\x6b\x06\x4d\x4e\x35\x38\x14\x31\x01\x43"
 flash:      196608 (pagesize: 128)
 sram:       20480
 chipid:     0x0447
 descr:      L0xx Category 5`

`
// STM Device connected to JTAG

pi@mfg-018:~/stlinknew/stlink/build/Release/bin$ ./st-flash --reset --format ihex write ~/Puck-1.4.1.hex
st-flash 1.6.1
2020-09-16T12:53:13 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
2020-09-16T12:53:13 INFO common.c: Attempting to write 35016 (0x88c8) bytes to stm32 address: 134217728 (0x8000000)
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000000 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000080 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000100 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000180 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000200 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000280 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000300 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000380 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000400 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000480 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000500 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000580 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000600 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000680 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000700 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000780 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000800 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000880 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000900 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000980 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000a00 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000a80 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000b00 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000b80 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000c00 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000c80 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000d00 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000d80 erased
2020-09-16T12:53:13 INFO common.c: Flash page at addr: 0x08000e00 erased
[... 533 lines cut ...]
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008180 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008200 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008280 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008300 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008380 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008400 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008480 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008500 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008580 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008600 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008680 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008700 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008780 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008800 erased
2020-09-16T12:56:02 INFO common.c: Flash page at addr: 0x08008880 erased
2020-09-16T12:56:02 INFO common.c: Finished erasing 274 pages of 128 (0x80) bytes
2020-09-16T12:56:02 INFO common.c: Starting Half page flash write for STM32L core id
2020-09-16T12:56:02 INFO flash_loader.c: Successfully loaded flash loader in sram
2020-09-16T12:56:02 ERROR flash_loader.c: flash loader run error
2020-09-16T12:56:02 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2020-09-16T12:56:02 WARN common.c:
write_half_pages failed == -1
272/273 pages written
2020-09-16T12:56:38 INFO common.c: Starting verification of write complete
2020-09-16T12:56:38 INFO common.c: Flash written and verified! jolly good!`

Nightwalker-87, 2020-09-17:
Thx @nzsmith1. :+1:
However there is still a flash loader run error, but there is an issue (#607) for that already.
Can you encircle more precisely where the fix appeared between v1.6.0 and the v1.6.1?
That would help for documentation purposes (PR linking and fixing the v1.6.1 changelog).

Nightwalker-87, 2020-10-10:
Ping @nzsmith1

Nightwalker-87, 2020-10-24:
Finally coming back to this, I am still trying to allocate the exact commit that fixed this.
It lies between 393310fc9925907c239f0ad7fac1e8d0565d65c0 and 6aa08a659a701feb9c08a4cb1628b92645f41309.
Can somebody help investigating?

stappersg, 2020-10-25:
On Sat, Oct 24, 2020 at 08:20:14AM -0700, nightwalker-87 wrote:
> Finally coming back to this, I am still trying to allocate the exact commit that fixed this.
> It lies between 393310fc9925907c239f0ad7fac1e8d0565d65c0 and 6aa08a659a701feb9c08a4cb1628b92645f41309.
> Can somebody help investigating?

Here a message from my armchair  :smile:

`git bisect` needs to known which commit is good and which bad.

Nightwalker-87, 2020-10-25:
@strappersg: There were too many changes in-between for a simple bisect.
One would need to select various commits to narrow down the amount of commits which are of interest.

Nightwalker-87, 2020-12-26:
@nzsmith1 Can you test the following commits with your environments?
It would be fine just to know if a commit is working or not by comparing with the output above.
I assume this can help us with a view to reasonable bisecting.

v1.6.1 (working)
38a5c14877964e7ea272565b372576d61710790f
9b19f9225460472af9d98959b7217d0a840ee972
cadfa941621cef85205622d6af0f87648d2a31c7
a09ef54cf53f0e8c194ef33c49dfe96622e5c127
2f9748e52f4739354668bc77bd208ca471cc528b
v1.6.0 (not working)

Nightwalker-87, 2021-01-06:
@Ant-ON You were involved in similar issues recently. Can you help?
I am not really sure if I can reproduce the issue with my hardware...

Nightwalker-87, 2021-03-07:
Fixed in #943.

closed 2021-03-07 by Nightwalker-87