stlink-org/stlink #841: Doesn't detect STM32F103T6U

FFY00, 2019-10-20:
- [X] Programmer/board type: Stlink/v2 (clone)
- [ ] Programmer firmware version: How to find?
- [X] Operating system: Linux
- [X] Stlink tools version and/or git commit hash: 1.5.1
- [X] Stlink commandline tool name: `st-util`
- [X] Target chip (and optional board): STM32F103T6U6A

Output:

```
st-util 1.5.1
2019-10-20T02:12:39 INFO common.c: Loading device parameters....
2019-10-20T02:12:39 WARN common.c: Invalid flash type, please check device declaration
2019-10-20T02:12:39 INFO gdb-server.c: Chip ID is 00000000, Core ID is  00000000.
2019-10-20T02:12:39 INFO gdb-server.c: Listening at *:4242...
```

Expected/description:
I would expect the chip to be detected.

Nightwalker-87, 2020-03-19:
@FFY00: Please verify if the problem still exists in Release v1.6.0.

gimbas, 2020-03-19:
I tested just now with v1.6.0
this is the output:
```
st-util 1.6.0
2020-03-19T19:33:03 DEBUG common.c: stlink current mode: debug (jtag or swd)
2020-03-19T19:33:03 DEBUG common.c: stlink current mode: debug (jtag or swd)
2020-03-19T19:33:03 DEBUG common.c: *** looking up stlink version
2020-03-19T19:33:03 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-03-19T19:33:03 DEBUG common.c: stlink pid     = 0x3748
2020-03-19T19:33:03 DEBUG common.c: stlink version = 0x2
2020-03-19T19:33:03 DEBUG common.c: jtag version   = 0x22
2020-03-19T19:33:03 DEBUG common.c: swim version   = 0x7
2020-03-19T19:33:03 DEBUG common.c: *** set_swdclk ***
2020-03-19T19:33:03 DEBUG common.c: *** stlink_jtag_reset ***
2020-03-19T19:33:03 DEBUG common.c: *** stlink_reset ***
2020-03-19T19:33:03 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-03-19T19:33:03 INFO common.c: Loading device parameters....
2020-03-19T19:33:03 DEBUG common.c: *** stlink_core_id ***
2020-03-19T19:33:03 DEBUG common.c: core_id = 0x00000000
2020-03-19T19:33:03 DEBUG common.c: *** stlink_read_debug32 0 is 0xe0042000
2020-03-19T19:33:03 DEBUG common.c: *** stlink_read_debug32 0 is 0x40015800
2020-03-19T19:33:03 WARN common.c: Invalid flash type, please check device declaration
2020-03-19T19:33:03 DEBUG common.c: *** stlink_reset ***
2020-03-19T19:33:03 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-03-19T19:33:03 INFO gdb-server.c: Chip ID is 00000000, Core ID is  00000000.
2020-03-19T19:33:03 INFO gdb-server.c: Listening at *:4242...
```

Nightwalker-87, 2020-03-19:
Thx for testing. This clearly needs to be addressed.

Nightwalker-87, 2020-03-20:
Have you ruled out any hardware issues?

gimbas, 2020-03-20:
guaranteed not to be an hardware issue.

Nightwalker-87, 2020-04-01:
As of #115 this should actually work...

FFY00, 2020-04-01:
No, that issue does not really provide much information. The only thing that doesn't work is the gdb server, everything else works properly from my experience.

Nightwalker-87, 2020-04-01:
Then ticket name is not correct. Also the error was not really described.

Nightwalker-87, 2020-07-18:
> The only thing that doesn't work is the gdb server, everything else works properly from my experience.

@FFY00: Something is not correct here. Can you also run `st-info --probe` and post the output here?

Nightwalker-87, 2021-03-28:
@FFY00: Have you tested Release v1.6.1 as well?

Nightwalker-87, 2021-04-03:
@FFY00: Sorry for the late reply, but we've not forgotten about this issue.
>     * Programmer firmware version: How to find?
The firmware version is displayed when executing `st-info --probe`
Can you check? Ensure to work with the latest version which should be **V2.J37.S7** for your programmer.

Nightwalker-87, 2021-08-15:
Can someone re-check the gdb-server & st-util with an original F1 series chip using **v1.6.0**, **v1.6.1** and **v1.7.0**?

perigoso, 2022-01-05:
No, sorry, no longer have the hardware handy to test this.

Nightwalker-87, 2022-10-23:
This issue is now closed due to inactivity.
Please also note that any version prior to `v1.7.0` is unsupported according to our [#Security Policy](https://github.com/stlink-org/stlink/security/policy#security-policy).
Should the problem persist, please retry with the latest version in the `develop` branch.
If this is the case, one should then open a new ticket.

closed 2022-10-23 by Nightwalker-87