stlink-org/stlink #845: make package for debian based distros

zx81a, 2019-10-31:
Hello.

Debian package created with 'make package' does not contain **postinst** script.
I do not know how to add it, but installation should run  **depmod** after sucess.

Nightwalker-87, 2020-03-01:
@slyshykO: Would it make sense to fix this as well, as we‘re looking at compiling and build issues at the moment?

slyshykO, 2020-03-01:
@Nightwalker-87 I know nothing about packaging, sorry.
We should attract people who know.
Try to search in discussion #701

Nightwalker-87, 2020-03-01:
Ok, no problem. Haven‘t reviewed that PR yet, but thx for the hint.

Nightwalker-87, 2020-03-16:
@Vascom or @bluca maybe?

Vascom, 2020-03-16:
I am only for Fedora.

bluca, 2020-03-16:
It looks like the makefiles are using CMake's embedded package generator. I am not familiar with it, and have no time to look into it - sorry.

Nightwalker-87, 2020-03-25:
Related to #877.

commit 68fdd392 (2020-04-20): Reconfigured cmake packaging

- Refactoring for package config & build
- deb-pkg: postinst script with depmod -a

(Fixes #845)
closed 2020-04-18 by Nightwalker-87