stlink-org/stlink #846: Inappropriate driver is taken on using v1.3.0

dhineshkumarmcci, 2019-11-04:
- [X] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version: 
- [X] Operating system: Windows 7
- [X] Stlink tools version and/or git commit hash: v1.3.0
- [X] Stlink commandline tool name: `st-util`
- [X] Target chip (and optional board): STM32L082

On using **"st-util"** of v1.3.0, we face issue on connecting with GDB tool. We see an inappropriate driver is considered instead of ST-Link driver. The driver information has been viewed using **"st-info --serial"**.

Output:
**GDB error**
```
warning: while parsing target memory map (at line 1): Can't convert length="0xzx" to an integer
```
**STLink driver issue**
```
C:\stlink\bin>st-info.exe --serial
libusb: info [windows_get_device_list] The following device has no driver: '\\.\
USB#VID_147E&PID_2016#6&32FEB3AB&0&3'
libusb: info [windows_get_device_list] libusb will not be able to access it.
493f72063f7853533956193f
```

Attached the screenshot of driver highlighted with red rectangle.

Later we cloned the source of v1.5.1 and build them, the st-util here works as expected and proper driver is seen.

Thank you,
Dhinesh

Nightwalker-87, 2020-03-21:
@dhineshkumarmcci: Would it be possible for you to find out which is the first version to work as expected? (v1.3.1, v1.4.0, v1.5.0) That would help to allocate, archive and link this issue correctly within the project.

slyshykO, 2020-03-22:
@Nightwalker-87, Think this is the same issue as #438

Nightwalker-87, 2020-03-23:
Well, they may be related, but the error message is different.

Nightwalker-87, 2020-03-25:
@dhineshkumarmcci: Can you please test the versions mentioned before, each with a clean build?

Nightwalker-87, 2020-04-09:
Let me finally address this issue with a comprehensive solution:

1) _GDB error_: This is a bug in the `stlink` toolset when using the MinGW development toolset. It has been fixed in PR #578.
2) _Driver issue_: A short web search reveals that this seems to be a common driver issue on windows where `\Driver\WUDFRd` fails to load. As the cited error also appears with several other usb devices, it is obviously not related to the `stlink` toolset. Should it be related to `libusb`, this bug may have been fixed in a later version of this external library:
* v1.0.21 is used in Release v1.4.0
* v1.0.22 is used in Releases v1.5.0 and v1.5.1

Hence this topic is now fully resolved, I am closing this issue now.

closed 2020-04-09 by Nightwalker-87