stlink-org/stlink #850: Incomplete support for STM32G0xx MCUs

mrfirmware, 2019-12-23:
I have a fix.

- [X] Programmer/board type: Stlink/v2-onboard
- [X] Programmer firmware version: e.g STSW-LINK007 3.3.2 (FW V2J35M26)
- [X] Operating system: e.g Linux, Ubuntu 19.10
- [X] Stlink tools version and/or git commit hash: 3690de9
- [X] Stlink commandline tool name: `st-util`
- [X] Target chip (and optional board): e.g STM32G031K8 (STM32G0 Nucleo-32 board [MB1455])

A as-detailed description possible of the problem with debug output when available.
When trying to flash and debug my Nucleo-32 with the STM32G031K8 st-util failed to recognize the smaller 'G0 part with IDCODE[DEV_ID] = 0x466.

Output:

$ ./src/gdbserver/st-util
st-util 1.5.1-50-g3690de9
2019-12-23T14:03:25 INFO common.c: Loading device parameters....
2019-12-23T14:03:25 WARN common.c: unknown chip id! 0x10016466

Expected/description:
I got it to work with the following changes:

1. First I renamed STLINK_CHIPID_STM32_G0X1 --> STLINK_CHIPID_STM32_G07X and added STLINK_CHIPID_STM32_G03X to better reflect the large vs. small chips. The larger G07X includes the 'G070, 'G071, and 'G080 while the smaller G03X includes the 'G030, 'G031, and 'G041.

2. Then I updated the .description fields in stlink_chipid_params devices[].

3. Then I added an entry to the devices[] array to match the G3X parameters.

4. Then I added an OR to the sl->chip_id checks in stlink_write_option_bytes() to allow the new G03X part.

Now I get the following output:
$ src/gdbserver/st-util
st-util 1.5.1-50-g3690de9
2019-12-23T14:14:20 INFO common.c: Loading device parameters....
2019-12-23T14:14:20 INFO common.c: Device connected is: G030/G031/G041 device, id 0x10016466
2019-12-23T14:14:20 INFO common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 2048 bytes
2019-12-23T14:14:20 INFO gdb-server.c: Chip ID is 00000466, Core ID is  0bc11477.
2019-12-23T14:14:20 INFO gdb-server.c: Listening at *:4242...

and I can connect with arm-none-eabi-gdb --version
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git

My ELF file gets flashed properly and I can set breakpoints, single-step, inspect variables, etc.

mrfirmware, 2019-12-23:
It looks like I cannot push a new branch so I cannot create a pull request.

terrillmoore, 2019-12-23:
@mrfirmware Fork the repo, push the branch to your fork, then create a pull request from that branch against the upstream repo.

mrfirmware, 2019-12-23:
We don't use the fork-and-branch model at work; just branches so I'm not having any luck pushing to my forked, upstream rep.

gdamjan, 2020-01-26:
@mrfirmware you can also push to your repos master, doesn't need to be a branch. 

(OTOH, that timeout might be your corporate firewall?)

mrfirmware, 2020-01-27:
I colleague at work showed me how to get past my authentication issue so I was able to create a PR for my small change set. Hopefully it is useful.

benmartinek, 2020-01-28:
Any progress so far? I'm really hoping to be able to use the tools with G0 chips.

mrfirmware, 2020-01-28:
The PR is up.

Nightwalker-87, 2020-02-21:
Resolved and closed with commit 7c0ebda0b514fb4c707205259ab221bd90f4cf78.

Nightwalker-87, 2020-03-15:
See also issues #825, #856, #857.

Nightwalker-87, 2020-03-16:
Closed by #825 and #857.

closed 2020-03-16 by Nightwalker-87