stlink-org/stlink #852: Mac OS X Catalina cannot find STM32 via USB port

ihassin, 2020-01-05:
Hello,

I connected my STM32 board and it's blinking.
However, I get "Found 0 stlink programmers" when I run
```
st-info --probe
```
I am running Catalina 10.15.2. No USB device was added to /dev, so maybe I am missing a driver?

The hardware is from Amazon:
[initeq] STM32 ARM STM32F103C8T6 Blue Pill Minimum System Development Board with ST-Link V2 USB Programmer (2-Pack + ST-Link V2)

Would anyone know how to resolve this? I looked for info/drivers but did not find anything.
st-info was built manually from git.

Thanks so much for your help!
Itamar

Nightwalker-87, 2020-02-22:
@ihassin: The current macOS driver package dedicated for stlink_v1 has been sucessfully tested on OS X 10.11 El Capitan. As to my current state of knowledge no later versions have been tested so far. However, if anyone did, there is no documentation on this. Also it remains unclear if the driver supports stlink_v2 as well. I am really sorry that I can't give you a more useful feedback by now. I feel that the current state of support for macOS is not where I'd like it to be. I am putting this up on the agenda for one of the upcoming releases as I feel we deserve better support for this platform.

Can anyone who is able to program and/or test driver-related issues on macOS please help out on this? We should ensure that stlink is working properly on macOS 10.14 & 10.15. Older versions may be dropped from support as very rarely seen by now. 10.14 is the latest system version to support x86 apps and binaries and therefore should see good support for several years.

ihassin, 2020-03-01:
Thanks so much for your reply. I could not get it to work on 10.15.3 Catalina, and have shelved my efforts :(

Nightwalker-87, 2020-03-01:
@ihassin: Thx for your feedback. I’m sorry to hear that. Will pull this issue forward to the upcoming release, as I feel it is too significant to push it out any further.

ihassin, 2020-03-01:
Greatly appreciated! I had plans to write an RTOS for it.

Nightwalker-87, 2020-03-23:
The first thing here is: We need at least one contributor on a macOS 10.14 (Mojave) and 10.15. (Catalina) environment. And it would also be a nice to have, if a STlink-v1 programmer was present besides a STLink-v2, as this would also help to fix/verify the other two issues already linked. Volunteers step forward!

Nightwalker-87, 2020-03-23:
@ihassin: Can you describe more precisely what you did? You wrote that you connected the Bluepill and that it is blinking, this is just fine as the Blinky program is linkely to be preflashed prior to shipping. Here there is not driver to expect, as the USB port only serves to power the board in this state. Is your STLink-v2-clone Programmer recognised on USB?

ihassin, 2020-03-23:
Hi, and thanks for following up!

Correct - Blinky works, so I used that to validate that I have a good board.
Inserting the USB cable to my Mac Catalina 10.15.3 did not make a USB device appear, hence could not communicate with it. Am I doing something wrong? I’d love to be able to help somehow. I’ve never programmed MacOS at the device level, but am willing to learn, if you can take on an apprentice :) Definitely willing to help in testing etc.

Itamar

Nightwalker-87, 2020-03-23:
What was connected via the USB you attached, the Board or the STLink-v2-Programmer?

Nightwalker-87, 2020-03-24:
Related to #672.

ihassin, 2020-03-24:
Hi there - connected to the programmer.

Itamar

Nightwalker-87, 2020-03-24:
ok, and what does "System Profiler" say? (`/Programs/Utilities/System Profiler.app`)
Take a look at "Hardware --> USB" in the sidebar within the program window.
I expect to see something there like a device and manufacturer ID.
If it powers up it should actually tell us something...

jcw, 2020-03-26:
Reading through this thread, I think the OP simply connected the Blue Pill via its USB connector. No ST-Link, i.e. no way it can be programmed that way. An ST-Link of some kind needs to be connected to the 4 SWD/power pins on the other side of the Blue Pill to reprogram it.

IOW, this is not macOS related, this is simply a misunderstanding as to how Blue Pills can be re-flashed.

Nightwalker-87, 2020-03-26:
@jcw: I did think that before, but he explained later...

Just to make it very clear:
1) USB-Port <---> STlink-v2 (Programmer) <--SWD+PWR-Wires--> SWD-Pins on Bluepill-Board
=> programming works
2) USB-Port <---> Bluepill-Board
=> not detected by mac AND programming not possible; only PWR supply via USB-Port, USB will not work - this is normal and not an error

@ihassin: if 1) is your setup, then we can continue to look for a solution, as a problem appears to be present.

Nightwalker-87, 2020-03-30:
ping @ihassin ?

ihassin, 2020-03-30:
Yes getting the messages but need to find time to test again. I’ll try to do it today.

Itamar

ihassin, 2020-04-05:
> @jcw: I did think that before, but he explained later...
>
> Just to make it very clear:
>
> 1. USB-Port <---> STlink-v2 (Programmer) <--SWD+PWR-Wires--> SWD-Pins on Bluepill-Board
>    => programming works [...]
[image]
n we can continue to look for a solution, as a problem appears to be present.

@Nightwalker-87 @jcw : I think I am on (1). Attaching photo. I also tried with a Mac that has native USB ports, so I don't think it's the converter that is the problem.

Nightwalker-87, 2020-04-05:
As far as I can see your SWDCLK and SWDIO lines are swapped.

Nightwalker-87, 2020-04-05:
I found this on the web: https://wiki.cuvoodoo.info/doku.php?id=jtag
However, your programmer looks similar, but has a different case marking.

Can you open it by _carefully_ pulling back the metal case over the usb connector. It would be helpful if you could take a closeup picture from the board inside (front side and back side). The resolution of the last picture you sent was quite good BTW. :+1: 
This would allow a look at the µC, the board marking and if the soldering is nicely done. 
I have read reports that this caused some issues on some devices. Keep in mind that they are cheap china clones (which does not necessarily mean that they are not useful, but they ship with issues sometimes as many are soldered very quickly by hand and are not always tested well).

I would just like to try to rule out any hardware issue.

jcw, 2020-04-05:
FWIW, I have a couple of ST-Link clones, which seem to fall into two variants and different header pins:

[image]

[image]

The chips on both are labeled STM32F103CBT6, i.e. 128k flash F103's.

I can't say with 100% certainty, but I don't think I've ever run into issues with one or the other.

(didn't mean to speak for @ihassin of course, just adding some more info, in case this helps ...)

Nightwalker-87, 2020-04-05:
@jcw: I am aware of that, but have two here with different pinouts as well. As one could read the pinout in the picture, I was able to identify.
The upper one of yours has a foam pad, which addresses a common issue as well, where the case can short-circuit the pin row of the connector - that's good. Thx for providing this example though, as it shows how the quality of the assembly can vary (but there are even worse - I can tell).

OK, so let's turn back to the topic...

ihassin, 2020-04-06:
This is the one I got from Amazon:

https://www.amazon.com/gp/product/B079B95L9Y/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Itamar

ihassin, 2020-04-06:
> I found this on the web: https://wiki.cuvoodoo.info/doku.php?id=jtag
> However, your programmer looks similar, but has a different case marking.
> 
> Can you open it by _carefully_ pulling back the metal case over the usb connector. It would be helpful if you could take a closeup picture from the board inside (front side and back side). The resolution of the last picture you sent was quite good BTW. 👍
> This would allow a look at the µC, the board marking and if the soldering is nicely done.
> I have read reports that this caused some issues on some devices. Keep in mind that they are cheap china clones (which does not necessarily mean that they are not useful, but they ship with issues sometimes as many are soldered very quickly by hand and are not always tested well).
> 
> I would just like to try to rule out any hardware issue.

The case is super-glued to the connector. At least they did a good job on that...

Nightwalker-87, 2020-04-06:
Unfortunate - Can't get an impression from here then, whether it is defective.

Maybe it it a solution to buy a new (cheap) programmer, but make sure you get one of these "known ones" with a "ST" printing on the case (these can also be opened BTW) as shown in the link I've sent before. You may consider to order from a local (known) online-distributor or shop, maybe on ebay, but that's up to you. Can you try to get a refund from amazon?

It could be that you ordered non-functional crap, the board might be ok, but we can't confirm this now, that is just a vague assumption...

I'm not aware of any general issues with STLink-v2 programmers on macOS. Also note that up to now we haven't done anything with the `stlink` tools yet, as the hardware detection of the programmer is not related to the tools, as no driver is necessary for that (this is different for the older STLink-v1 though, now obsolete).

ihassin, 2020-04-06:
Thanks so much for your help thus far. I’ll get another set/programmer and let you know.

Itamar

Nightwalker-87, 2020-04-06:
@ihassin: Let me give you some final advice how to proceed further: As soon as you have the new device, simply plug it into an usb port on your computer _without any board connected to it_ and refer to the following:

> ok, and what does "System Profiler" say? (`/Programs/Utilities/System Profiler.app`)
> Take a look at "Hardware --> USB" in the sidebar within the program window.
> I expect to see something there like a device and manufacturer ID.

If this is the case, you can proceed as originally intended.

... and let me give you some general advise for further proceedings:
- **avoid** to connect wires while the programmer is plugged into the usb port and thus is powered on
- **don't** connect any pins from the programmer to the board other than VCC, GND, SWDCLK and SWDIO (as intended)

Nightwalker-87, 2020-04-06:
@ihassin: I hope to have helped you along with the provided feedback.
Thx also to @jcw for the given input.

I am closing this now as resolved, as this issue is not related to the `stlink` toolset itself.

closed 2020-04-06 by Nightwalker-87

ihassin, 2020-04-06:
I’d like to thank you for the time and effort you spent on this and how helpful you were with your explanations and support. Really commendable, thank you for help and attention!

Nightwalker-87, 2020-04-06:
You are welcome.

ihassin, 2020-04-25:
Just following up - I just got the new programmer from Adafruit, and it works well. Recognized and st-info --probe sees it. Thank you so much for spending time with me to resolve this. I will leave a comment on Amazon regarding the one I originally bought.

Nightwalker-87, 2020-04-25:
@ihassin: Thx for your feedback. That is good news and I'm happy that we were able to help.