stlink-org/stlink #854: Fails to flash NUCLEO-F103RB.

fadedbee, 2020-01-08:
- [X] Programmer/board type: e.g Stlink/v2-onboard
- [X] Programmer firmware version: From programmer's (MBED) USB Mass Storage NODE_F103RB/DETAILS.TXT: Version: 0221, Build:   Jun 23 2017 17:43:45.
- [X] Operating system: Linux, Ubuntu 16.04
- [X] Stlink tools version: 3690de9fddeed49d304a1afab6410ceb64c70e1b
- [X] Stlink commandline tool name: `st-util`
- [X] Target chip (and optional board): e.g STM32F103RB (NUCLEO-F103RB)

The flashing of the onboard STM32F103RB fails.

Output:

```
chris@purple:~/github/stlink$ ./build/Release/src/gdbserver/st-util
st-util 1.5.1-50-g3690de9
2020-01-08T09:35:16 INFO common.c: Loading device parameters....
2020-01-08T09:35:16 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2020-01-08T09:35:16 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
2020-01-08T09:35:16 INFO gdb-server.c: Chip ID is 00000410, Core ID is  1ba01477.
2020-01-08T09:35:16 INFO gdb-server.c: Listening at *:4242...
2020-01-08T09:49:00 INFO gdb-server.c: Found 6 hw breakpoint registers
2020-01-08T09:49:00 INFO gdb-server.c: GDB connected.
2020-01-08T09:49:06 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2020-01-08T09:49:06 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2020-01-08T09:49:06 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-01-08T09:49:06 INFO flash_loader.c: Successfully loaded flash loader in sram

2020-01-08T09:49:06 INFO common.c: Starting verification of write complete
2020-01-08T09:49:06 INFO common.c: Flash written and verified! jolly good!
2020-01-08T09:49:06 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134218752 (0x8000400)
Flash page at addr: 0x08000400 erased
2020-01-08T09:49:06 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2020-01-08T09:49:06 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-01-08T09:49:06 INFO flash_loader.c: Successfully loaded flash loader in sram
2020-01-08T09:49:06 ERROR flash_loader.c: write error, count == 511
2020-01-08T09:49:06 ERROR common.c: stlink_flash_loader_run(0x8000400) failed! == -1
^C
```

```
chris@purple:~/vx4000-platform/nucleo-f103rb-blinky$ arm-none-eabi-gdb build/nucleo-f103rb-blinky.elf 
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/nucleo-f103rb-blinky.elf...done.
(gdb) target extended-remote :4242
Remote debugging using :4242
0x080021d8 in ?? ()
(gdb) load
Loading section .isr_vector, size 0x10c lma 0x8000000
Loading section .text, size 0xe4c lma 0x800010c
Loading section .rodata, size 0x2c lma 0x8000f58
Loading section .init_array, size 0x4 lma 0x8000f84
Loading section .fini_array, size 0x4 lma 0x8000f88
Loading section .data, size 0xc lma 0x8000f8c
Error finishing flash operation
(gdb) quit
```

Expected/description:

I had expected the flashing to complete successfully.

fadedbee, 2020-01-08:
I was able to create an mbed-os-example-blinky project and drop into the nucleo's USB Mass Storage.  This then automatically flashed it onto the F103RB.

The blinky is now running, so I don't think that the F103RB is faulty.

SourceLink, 2020-02-19:
I also encountered the same problem, the program uses mdk to compile and download without problems

Nightwalker-87, 2020-03-21:
@fadedbee: Can you do a clean rebuild of v1.5.1 and retest? If this is not successful I suggest to test v1.6.0 afterwards. Please report to us about the result.

Nightwalker-87, 2020-06-24:
Please test this with Release v1.6.0 and v1.6.1 to allow for investigation.

Nightwalker-87, 2021-03-11:
I have tried to reproduce this issue with under mostly identical circumstances:

- v1.5.1 / commit: 3690de9fddeed49d304a1afab6410ceb64c70e1b
- Programmer: STLink/V2-clone Firmware V2J37S7
- Board: STM32F103C8T6

Looking at the output reveals that `st-util` and `gdb` appear to deliver the expected output:

```
$ ./build/Release/src/gdbserver/st-util 
st-util 1.5.1-50-g3690de9
2021-03-12T00:05:13 INFO common.c: Loading device parameters....
2021-03-12T00:05:13 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-12T00:05:13 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2021-03-12T00:05:13 INFO gdb-server.c: Chip ID is 00000410, Core ID is  2ba01477.
2021-03-12T00:05:13 INFO gdb-server.c: Listening at *:4242...
2021-03-12T00:16:36 INFO gdb-server.c: Found 6 hw breakpoint registers
2021-03-12T00:16:36 INFO gdb-server.c: GDB connected.
2021-03-12T00:16:42 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2021-03-12T00:16:42 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2021-03-12T00:16:42 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2021-03-12T00:16:42 INFO flash_loader.c: Successfully loaded flash loader in sram

2021-03-12T00:16:42 INFO common.c: Starting verification of write complete
2021-03-12T00:16:42 INFO common.c: Flash written and verified! jolly good!
2021-03-12T00:16:42 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134218752 (0x8000400)
Flash page at addr: 0x08000400 erased
2021-03-12T00:16:42 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2021-03-12T00:16:42 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2021-03-12T00:16:42 INFO flash_loader.c: Successfully loaded flash loader in sram

2021-03-12T00:16:42 INFO common.c: Starting verification of write complete
2021-03-12T00:16:42 INFO common.c: Flash written and verified! jolly good!
2021-03-12T00:16:42 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134219776 (0x8000800)
Flash page at addr: 0x08000800 erased
2021-03-12T00:16:42 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2021-03-12T00:16:42 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2021-03-12T00:16:42 INFO flash_loader.c: Successfully loaded flash loader in sram

2021-03-12T00:16:42 INFO common.c: Starting verification of write complete
2021-03-12T00:16:42 INFO common.c: Flash written and verified! jolly good!
2021-03-12T00:16:42 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134220800 (0x8000c00)
Flash page at addr: 0x08000c00 erased
2021-03-12T00:16:42 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2021-03-12T00:16:42 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2021-03-12T00:16:42 INFO flash_loader.c: Successfully loaded flash loader in sram

2021-03-12T00:16:42 INFO common.c: Starting verification of write complete
2021-03-12T00:16:42 INFO common.c: Flash written and verified! jolly good!
2021-03-12T00:17:27 ERROR gdb-server.c: cannot recv: -2
2021-03-12T00:17:27 INFO gdb-server.c: Listening at *:4242...

```

```
$ gdb-multiarch /[...]/Blinker.elf 
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /[...]/Blinker.elf...
(No debugging symbols found in /[...]/Blinker.elf)
(gdb) load
You can't do that when your target is `exec'
(gdb) target extended-remote :4242
Remote debugging using :4242
0xfffffffe in ?? ()
(gdb) load
Loading section .isr_vector, size 0x130 lma 0x8000000
Loading section .text, size 0xbac lma 0x8000130
Loading section .rodata, size 0x74 lma 0x8000cdc
Loading section .init_array, size 0x4 lma 0x8000d50
Loading section .fini_array, size 0x4 lma 0x8000d54
Loading section .data, size 0x68 lma 0x8000d58
Start address 0x080002b4, load size 3520
Transfer rate: 16 KB/sec, 586 bytes/write.
(gdb) quit
A debugging session is active.

	Inferior 1 [Remote target] will be killed.

Quit anyway? (y or n) y
```

Nightwalker-87, 2021-03-25:
@Ant-ON What about this one. Do you have an idea?

Ant-ON, 2021-03-25:
@Nightwalker-87 Most likely fixed by #1113

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87