stlink-org/stlink #855: Load HEX file onto STM32L072CZ

vicatcu, 2020-01-13:
- [x] Programmer/board type: Stlink/v2 Clone (specifically [this one](https://www.amazon.com/Aideepen-ST-Link-Programming-Emulator-Downloader/dp/B01J7N3RE6/ref=pd_sbs_147_t_0/131-8360886-6233443))
- [x] Programmer firmware version: V2.J34.S7 as reported by Windows STM32 ST-LINK Utility software
- [x] Operating system: Linux (specifically Ubuntu 18.04)
- [x] Stlink tools version and/or git commit hash: e.g master (3690de9f)
- [x] Stlink commandline tool name: `st-flash`
- [x] Target chip (and optional board): STM32L072CZ

I am looking to simply load a HEX file onto the target using this programmer. When I run the following command: `st-flash --format ihex write myapp.hex`

I get the following error trace / output:
```
st-flash 1.5.1-50-g3690de9
2020-01-13T16:19:45 INFO usb.c: -- exit_dfu_mode
2020-01-13T16:19:45 INFO common.c: Loading device parameters....
2020-01-13T16:19:45 WARN common.c: Invalid flash type, please check device declaration
  core status: unknown
Unknown memory region
```

And if I run `st-info --chipid` I just get: `0x0000`.

...and `st-info --probe` outputs the following when the programmer is plugged into USB:
```
Found 1 stlink programmers
 serial: 2c0f03013212354d434b4e00
openocd: "\x2c\x0f\x03\x01\x32\x12\x35\x4d\x43\x4b\x4e\x00"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device
```

I'm not sure what, if anything, I might be doing wrong. I built the project from source and I can obviously run the binaries without a problem. Please can someone help set on the right course? If there is an easier path than using the tools in this repository, I'm also open to hearing that.

EmBitz, 2020-01-21:
To check another tool if you can read the ID, you could try EBlink
https://github.com/EmBitz/EBlink

use:
./eblink -I stlink -S stm32-auto

that should give you a chipd ID. There is a Linux build. If EBlink also fails to read chip ID then I guess there is something wrong with your board.  If you have a valid ID you could try flashing

./eblink -I stlink -S stm32-auto -F file=myapp.hex

Nightwalker-87, 2020-03-21:
@vicatcu: Can you rule out any hardware issues?

Nightwalker-87, 2020-06-24:
> And if I run `st-info --chipid` I just get: `0x0000`.

To me this does not seem to be related to flashing a hex file then.
Can we test with later releases than v.1.5.1 ? Is there anybody with the same device?

Nightwalker-87, 2021-05-14:
This may be looked at in the context of #923 and #1108.

Nightwalker-87, 2021-05-26:
@wolsty7 Would you fancy to give this a test as well? Please let us know.

Ant-ON, 2021-05-27:
@Nightwalker-87 Most likely using `--connect-under-reset` will solve this issue

wolsty7, 2021-07-21:
my programmer is the V3SET that isn't supported by 1.5.1 and i'm on raspbian... --connect-under-reset helps in my case

using branch l0_soft_loader_fix 1.6.1-330-g679ab65  that works for #1138:
> st-flash --format ihex write /home/pi/Downloads/v1.hex
> st-flash 1.6.1-330-g679ab65
> Failed to enter SWD mode
> Failed to connect to target

using branch l0_soft_loader_fix 1.6.1-330-g679ab65  that works for #1138:
> st-flash --format=ihex write /home/pi/Downloads/v1.hex
> st-flash 1.6.1-330-g679ab65
> Failed to enter SWD mode
> Failed to connect to target

using branch l0_soft_loader_fix 1.6.1-330-g679ab65  that works for #1138:
> st-flash --connect-under-reset --format=ihex write /home/pi/Downloads/v1.hex
> st-flash 1.6.1-330-g679ab65
> 2021-07-21T11:28:03 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
> 2021-07-21T11:28:03 INFO common.c: Attempting to write 78900 (0x13434) bytes to stm32 address: 134217728 (0x8000000)
> 2021-07-21T11:28:03 INFO common.c: Flash page at addr: 0x08000000 erased
> 2021-07-21T11:28:03 INFO common.c: Flash page at addr: 0x08000080 erased
> ###removed for brevity###
> 2021-07-21T11:28:08 INFO common.c: Flash page at addr: 0x08013300 erased
> 2021-07-21T11:28:08 INFO common.c: Flash page at addr: 0x08013380 erased
> 2021-07-21T11:28:08 INFO common.c: Flash page at addr: 0x08013400 erased
> 2021-07-21T11:28:08 INFO common.c: Finished erasing 617 pages of 128 (0x80) bytes
> 2021-07-21T11:28:08 INFO common.c: Starting Flash write for L0
> 2021-07-21T11:28:08 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2021-07-21T11:28:08 INFO flash_loader.c: Clear DFSR
> 2021-07-21T11:28:08 ERROR flash_loader.c: write_buffer_to_sram() == -1
> 2021-07-21T11:28:08 WARN common.c: Failed to use flash loader, fallback to soft write
> 1232/1232 halfpages written
> 2021-07-21T11:28:14 INFO common.c: Starting verification of write complete
> 2021-07-21T11:28:15 INFO common.c: Flash written and verified! jolly good!

using branch master 1.7.0:
> st-flash --connect-under-reset --format=ihex write /home/pi/Downloads/v1.hex
> st-flash 1.7.0
> 2021-07-21T11:50:56 INFO usb.c: Unable to match requested speed 1800 kHz, using 1000 kHz
> 2021-07-21T11:50:56 WARN common.c: NRST is not connected
> 2021-07-21T11:50:56 ERROR common.c: Soft reset failed: timeout
> 2021-07-21T11:50:56 ERROR common.c: Can not connect to target. Please use 'connect under reset' and try again
> Failed to connect to target

So it appears to work correctly using the l0_soft_loader_fix branch

Nightwalker-87, 2021-07-28:
@Ant-ON Is this a regression?

Nightwalker-87, 2022-10-23:
This issue is now closed due to inactivity.
Please also note that any version prior to `v1.7.0` is unsupported according to our [#Security Policy](https://github.com/stlink-org/stlink/security/policy#security-policy).
Should the problem persist, please retry with the latest version in the `develop` branch.
If this is the case, one should then open a new ticket.

closed 2022-10-23 by Nightwalker-87