stlink-org/stlink #862: How do I get checksum?

vitormhenrique, 2020-02-24:
Hello,

Is there a way to get the checksum from a binary? like the st-link tool has on windows? I wanna compare two binaries generated on my computer and a standard one.

bmarvo, 2020-02-25:
Do you mean a checksum of the memory on the device? And compare that to a file on your computer?

vitormhenrique, 2020-02-25:
I was thinking of a binary file, on my use case I was having problems with a binary compiled on my computer and one downloaded, the source code was the same, but the binaries did not match, I was using a newer tool chain, so the new binary was corrupt 

on st-link tool when you load a binary, ti calculates a checksum, so I used that to figure out when I had the correct toolchain

https://i.ibb.co/5TS7twF/figure.png

bmarvo, 2020-02-25:
Ah okay. These tools don't do that for you as it stands. On Windows I can recommend HxD hex editor to checksum a binary file (or simply compare two binary files quickly). Are you on Windows?

Nightwalker-87, 2020-03-25:
@vitormhenrique: Unfortunately I can't read the content of the image you linked as the resolution is too low for that. Can you paste a larger screenshot or a relevant detail of it?

vitormhenrique, 2020-03-25:
no idea why that image was so small....

https://i.ibb.co/mq8vvpP/pic.png

Nightwalker-87, 2020-03-25:
I see - so we are actually looking for a feature request here.
The flashloader does verify the integrity of the flashed content behind the scenes, so I assume that this functionality can be extended. It wants to be looked at certainly.

chenguokai, 2020-04-12:
If no special algorithm limitations are given, I think it’s quite easy to integrate some checksum algorithm(md5, sha256, etc) into the file loading process. Maybe I can give it a try when I am free.

grevaillot, 2020-04-13:
stlink utility (the one in the capture) defines the checksum in its documentation (UM0892) as a simple arithmetic sum, truncated to 32bit ... (sigh)

> Memory checksum
>  Calculates the checksum value of a specified memory zone defined by the
> address and the size fields in the Memory display section of the main window.
> The checksum is calculated based on an arithmetic sum algorithm, byte per
> byte. The result is truncated to 32-bit word.
> The checksum value is displayed in the log window

While this is quite crude and not reaaally reliable, this is cheap to implement and maintaining "compatibility" with st-utility would be nice.

chenguokai, 2020-04-13:
@grevaillot Thanks for your information, I will add support for this on my PR.

Nightwalker-87, 2020-04-13:
Should we have some documentation for this in our tutorial?

chenguokai, 2020-04-13:
The algorithm used by ST’s windows program may need a documentation at least for developers I think.

Nightwalker-87, 2020-04-13:
Ok, can you add a section with a few lines to our tutorial then (at the end of the reviewed part)? That would be nice. Don't bother too much about formatting there though, as I am still working on this document anyway...

chenguokai, 2020-04-13:
I am still wondering, if the two methods should be merged all together.
If so, I may need to adjust the two functions for a better format. The current two commits would cause a somewhat too verbose output.

closed 2020-04-13 by Nightwalker-87