stlink-org/stlink #869: Compilation error on raspbian

koendv, 2020-02-28:
Compilation error on raspbian (arm 32-bit):

/home/koen/src/stm32duino-raspberrypi/STM32Tools/tools/src/stlink/src/usb.c: In function ‘_stlink_usb_version’:
/home/koen/src/stm32duino-raspberrypi/STM32Tools/tools/src/stlink/src/usb.c:192:18: error: comparison of integer expressions of different signedness: ‘ssize_t’ {aka ‘int’} and ‘uint32_t’ {aka ‘unsigned int’} [-Werror=sign-compare]
         if (size != rep_len) {
                  ^~
cc1: all warnings being treated as errors

Propose the following patch, to make sure that rep_len and size have the same type (usb.c).

diff --git a/src/usb.c b/src/usb.c
index 9df9d72..ad90baa 100644
--- a/src/usb.c
+++ b/src/usb.c
@@ -171,7 +171,7 @@ int _stlink_usb_version(stlink_t *sl) {
     unsigned char* const data = sl->q_buf;
     unsigned char* const cmd  = sl->c_buf;
     ssize_t size;
-    uint32_t rep_len = 6;
+    ssize_t rep_len = 6;
     int i = fill_command(sl, SG_DXFER_FROM_DEV, rep_len);
 
     cmd[i++] = STLINK_GET_VERSION;

Nightwalker-87, 2020-02-29:
Fixed in PR #870.

Nightwalker-87, 2020-03-21:
Fixed in PR #872.

closed 2020-03-21 by Nightwalker-87