stlink-org/stlink #871: Build failing with unknown compiler flags

bmeneguele, 2020-02-29:
Hi all, I couldn't find any issue reporting this same error, so I'm sorry in case there's already one.

I tried to build stlink from the source using GCC and got a building error pointing that the compiler didn't recognize "-Wmissing-variable-declaration", which in fact isn't supported by GCC, but only Clang. Then I changed my toolchain to clang and got the same error, but this time the warning option that wasn't recognized was "-Wmaybe-uninitialized", which is a GCC option, and not clang's.

It seems that in `cmake/CFlags.cmake` the check for compiler's flags support (CheckCCompilerFlag) isn't working properly, or some information is missing from `doc/compiling.md`.

System details:
cmake version 3.16.4
gcc version 9.2.1
clang version 9.0.1

I can help testing if necessary.

slyshykO, 2020-03-01:
Hello @bmeneguele.
Can you specify OS on which you try to build?
And, please, provide CMakeCache.txt.

bmeneguele, 2020-03-01:
@slyshykO ... I won't lie... I'm not being able to reproduce the issue again, even with a fresh repo clone: `cmake -DCMAKE_C_COMPILER=clang -DCMAKE_BUILD_TYPE=Debug ..` is working just fine.

Considering I'm not really aware of cmake's workflow, are there any "cache" file outside project's own folder that could be working around the reproduction of the issue?

I'm currently running it over Linux (Fedora 31) and you can find the CMakeCache file pasted in https://gist.github.com/bmeneguele/99cb5a739f6a7363a7c9ed832496b7fe. But unfortunately, as I said before, this CMakeCache is referencing the build that didn't fail :/, so you probably won't find any issue there.

Feel free to close this issue. In case I see it again I open a new one with all the information needed.

slyshykO, 2020-03-01:
@bmeneguele. Yes, cmake generate CMakeCache.txt in the current build directory. So, you provided a proper file, I think. One thing bothers me.
You set -DCMAKE_C_COMPILER=clang but in cache CMAKE_C_COMPILER:STRING=/usr/lib64/ccache/gcc. It is strange.
Try to point the full path to the compiler.
For example -DCMAKE_C_COMPILER=/usr/bin/clang-8.0

bmeneguele, 2020-03-01:
@slyshykO ah .. sorry!! I actually ran the command with "gcc" indeed. I also failed to copy/paste the correct line to you (I tried with both gcc and clang to check if any errors would shown up).

slyshykO, 2020-03-01:
@Nightwalker-87 , It look like we can close this issue.

bmeneguele, 2020-03-02:
Yes. Sorry for the noise guys, and thanks anyway.

closed 2020-03-02 by bmeneguele

Nightwalker-87, 2020-03-10:
@bmeneguele: No need to apologise - this resolved issue may be still be helpful to others.
@slyshykO: Thx for your support in this case.