stlink-org/stlink #873: how to use stlink to debug iap program

mttbx, 2020-03-09:
I want to debug iap program by stlink, however I don't know how to burn two program at different flash address through st-util / gdb.

Nightwalker-87, 2020-03-09:
Please describe your issue more precisely and specify the system you use to facilitate individual support.

mttbx, 2020-03-09:
I'm using stm32 chip to run c programs in Linux. I use gdb to connect remote st-util server so I can debug my program. My program has two parts. The first part is boot sub program, and the second is app sub program. In the boot program, it will do some work and then jump to the second program. The two programs can be compiled seperatly, and they both contains the main function. I can boot these two program into the flash of chip through gdb using load command (remote connect to st-util server). Here's the question, tt seems there's two sets of symbols, and how can I guide gdb to help me debug them. 

梅骁同济大学测绘与地理信息学院

mttbx, 2020-03-13:
I'm using stm32 chip to run c programs in Linux. I use gdb to connect remote st-util server so I can debug my program. My program has two parts. The first part is boot sub program, and the second is app sub program. In the boot program, it will do some work and then jump to the second program. The two programs can be compiled seperatly, and they both contains the main function. I can boot these two program into the flash of chip through gdb using load command (remote connect to st-util server). Here's the question, tt seems there's two sets of symbols, and how can I guide gdb to help me debug them.

Nightwalker-87, 2020-06-24:
This is actually a question related to the GNU Debugger (gdb) which is not part of this `stlink` toolset itself.
I'd suggest you to consult the IRC-Support channel of the gdb project with this.
It can be found here: https://www.gnu.org/software/gdb/irc/

As this question is not related to the `stlink` project and adequate advice has been given, this issue will now be closed.

closed 2020-06-24 by Nightwalker-87