stlink-org/stlink #874: PROJECT NEWS & UPDATES

Nightwalker-87, 2020-03-11:
This issue is dedicated to current project activities and project updates of general interest.
Comments related to specific announcements are always welcome.
**Please do not reuse this ticket for any other purpose.**

Nightwalker-87, 2020-03-11:
Review of old issues is about to continue in the upcoming days.
Current state: 219 issues and PRs remain...

Nightwalker-87, 2020-03-11:
**Upcoming Release:** **v1.6.1** (_planned for June 2020_)

Nightwalker-87, 2020-03-11:
**General ToDo:**
- Update CHANGELOG.md for Releases  v1.3.1+
-

Nightwalker-87, 2020-03-21:
@texane: Could you switch the default branch from `master` to `develop` for future PRs? Thx.

Nightwalker-87, 2020-03-24:
**Project Maintenance:**
- Review of old issues has completed by mid of March 2020
- Reviewed and updated missing features (24.03.2020)

Nightwalker-87, 2020-04-08:
**General notice:**

This project has now been transferred to a so called "**git-organisation**". Hosting within such a shared account will ease collaboration and offer new possibilities and features for further evolution of this project.
Please take into account that with this transition, the web address to this project has changed as well. As taken from documentation, GitHub will automatically redirect links to the former project page to the new space of this project (within our organisation **stlink-org**).

However it would be nice if old external references are updated whenever possible.
**Thanks in advance and also for your continued support and contribution to this project!**

Nightwalker-87, 2020-04-21:
**General notice:**

A build failure in the recent commits on the `develop` forced me to rollback the most recent commits to the last known stable state on unix based systems. This breaks building on windows again currently. **Please consider that no further PRs will be reviewed and no new issues shall be reported until this has been investigated and fixed.** Further updates will be posted here.

Nightwalker-87, 2020-04-21:
Ok, we appear to be back on track now... Thx for testing. :1st_place_medal:

Nightwalker-87, 2020-04-25:
**Upcoming Release: v1.6.1** (planned for 01.06.2020)

In order to have enough time for pre-release testing the **Deadline for feature-PRs** is the **15.05.2020**. After this date only bugfixes shall be resolved.
Please consider this and comment on individual issues tagged with the v1.6.1 milestone, if these are likely not addressable by the current assignees up to this date.
These topics will then move to the next milestone v1.6.2 by that date.
Thx for taking this into account.

martonmiklos, 2020-04-27:
I doubt that StLinkV3 support will fit to that timeframe, unless someone else look into it because I am busy with other things at the moment.

Nightwalker-87, 2020-05-06:
**Upcoming Release: v1.6.1** (planned for 01.06.2020)

In order to have enough time for pre-release testing the **Deadline for feature-PRs** is the **15.05.2020**. After this date only bugfixes shall be resolved.
Please consider this and _comment in the individual issues_ tagged with the v1.6.1 milestone, if these are likely not addressable by the current assignees up to this date.
These topics will then move to the next milestone v1.6.2 by that date.
Thx for taking this into account.

Nightwalker-87, 2020-05-10:
**Following Release: v1.6.2** (planned for 01.09.2020)

Contributions for issues tagged with the related milestone should use the codebase of v1.6.1 once released. Submissions will be accepted as from 15.06.2020.

Nightwalker-87, 2020-05-27:
**Upcoming Release: v1.6.1** (planned for 01.06.2020)

We still need reviewers and somebody with a STM32VL Discovery board and/or macOS would also be helpful to finally close the two remaining Pull-Request (which are completed). I'd like to see somebody else having a look at these as well. Please drop a note in the respective PRs, if you can contribute. Thx in advance.

Nightwalker-87, 2020-05-28:
**Following Release: v1.6.2** (currently planned for 01.09.2020)

Contributions for issues tagged with the related milestone should use the codebase of v1.6.1 once released. Submissions will be accepted as from 05.06.2020.

Nightwalker-87, 2020-06-18:
**List of maintainers for distribution repositories:**

- macOS (MacPorts): **chenguokai**
- Debian: **bluca**
- Ubuntu: **bluca**
- Fedora: **Vascom**
- EPEL (CentOS/RHEL): **Vascom**
- openSUSE: _(TBD)_ --> _Volunteers are welcome._
- Arch Linux: _(TBD)_ --> _Volunteers are welcome._
- ...

Nightwalker-87, 2020-06-24:
**NOTE: In order to support the evolution and progress of this project we are looking for ACTIVE DEVELOPERS to help with implementing new features and addressing bug reports. If you are reading this and are interested in contributing, please have a look at the issues currently open. Any help is highly appreciated.**

Nightwalker-87, 2020-06-27:
Content moved to #1000 which finally closes this issue.

closed 2020-06-27 by Nightwalker-87