stlink-org/stlink #875: reading and writing flash address with CLI

Hoel, 2020-03-14:
Hello, is there a way to read and write some flash addresses directly with the CLI without having to flash or read a whole binary file? I want to check few addresses on the flash and also overwrite few others, my application uses the binary directly (with CLI) as it is a javafx application.
Thank you.

Nightwalker-87, 2020-03-14:
Please describe more precisely what you intend to do.

Hoel, 2020-03-14:
I use STLink V2-1, on STM32F1xx device in this case. What i want to do is to write and read few bits at different addresses of the flash, without having to read the whole firmware and write it back. The prupose of this is to update some data in place, without having to read or write the whole thing.

Nightwalker-87, 2020-03-20:
Is there anybody who is able to contribute on this issue?

chenguokai, 2020-04-12:
Some facts may need a stress.

1. The internal flash is erased once a page(page size ranging from 1KB to at least 4KB). Any non-0xff bytes cannot be rewritten reliably. For F1XX, if I got it right, even 0xff bytes cannot be rewritten without an erase. That is, writing some random bytes causes the whole page to be erased.
2.  Considering this feature, a partial update may not reduce erase cycles without intentionally design that places all variable data inside a limited number of pages.
3. For an intentional design, manually specifying the start address and using a partial binary file is as effective as stlink on erase cycles and would reduce a read cycle that is a must if the diff process is done by stlink.

EmBitz, 2020-04-12:
@Hoel
Hi,

Now and then I visit this issue list, together with the one of OpenOCD, to check if users experience hardware problems with MCU's (great source for that). So I already noticed your request some days ago.

To implement reading and put this out on stdout shouldn't be difficult to implement in STlink. However, in-place modification is a bit harder because you need to cache the sector to be modified first and that is (was, I'm not sure the status today) not standard in STlink. Some 4 years ago I implemented some caching in a STlink fork to skip unmodified sectors but it wasn't that easy and straight forward.

I understand you question because I had the same need, to write a serial number at a particular place together with an updated checksum. So if you are in desperate need of this functionality you could check [EBlink](https://github.com/EmBitz/EBlink) which is build around a caching engine and can read memory locations from CLI (even on running targets) and can modify flash locations from CLI.

But EBlink can't use STlink V1, in that case you have to modify STlink (or openOCD)

Nightwalker-87, 2020-06-24:
What is the current state of this?

Hoel, 2020-06-25:
hello,
Regarding this issue i did us a workaround, a base firmware is stored in the application, some chunks are copied and patched when the target is programmed, st-flash is then used to flash these small chunks on the target.
This is not as elegant as it could be but it is fast since the chunks are small and there is no need to read the entire firmware from each target, so for this specific use case it is fine.

Nightwalker-87, 2020-06-28:
This could be helpful indeed.
Can you submit a PR so we can have a look at the implementation?

Nightwalker-87, 2021-03-12:
Ping @Hoel

geoffrey-vl, 2021-03-26:
I think this could be very useful, at least on ST's command line tool for Windows I'm using this for a couple of years now.
Example using this feature:

```
C:\> ST-LINK_CLI.EXE -c UR SWD freq=4000 -w32 0x080FF800 0x131D3FFC
```

For st-flash it could look as following:

```
$ st-flash write <variable-type> <variable> <address>
```

Example:

```
$ st-flash write -w32 0x131D3FFC 0x080FF800
```

Nightwalker-87, 2021-05-21:
@geoffrey-vl Can you contribute on this topic?

geoffrey-vl, 2021-05-21:
To be honest I have a couple of other project I'm also working on, so don't count on it.
But for starters I'd like to discuss how the command would look like. Can it be as described in https://github.com/stlink-org/stlink/issues/875#issuecomment-808199643 ?

Nightwalker-87, 2022-08-31:
I'd consider this a useful feature. Who can help out on this topic? Does anybody have an idea how to implement this?

Nightwalker-87, 2023-09-04:
Closing this feature request due to inactivity.

closed 2023-09-04 by Nightwalker-87