stlink-org/stlink #877: Maintainer for distribution repositories

Nightwalker-87, 2020-03-15:
Hi everybody. I would consider it useful to have somebody who is in charge of passing stable releases to several linux distribution repositories to ensure (see supported distributions), that users are not forced to build from source if they desire to run stable releases from 1.6.0 onwards. I don't know how this was organised in the recent years. Maybe someone has more to tell on this. Please have your say. Volunteers are welcome.

Nightwalker-87, 2020-03-24:
@bluca: I noted that v1.6.0 made it into the debian bullseye repository and that you are linked to it as a debian packet maintainer. Can you tell me more about the procedure(s) running behind the scenes?

Nightwalker-87, 2020-03-29:
List of maintainers:
- MacPorts: **chenguokai**
- Debian: **bluca**
- ...

bluca, 2020-03-29:
> @bluca: I noted that v1.6.0 made it into the debian bullseye repository and that you are linked to it as a debian packet maintainer. Can you tell me more about the procedure(s) running behind the scenes?

Sorry, not sure what you are looking for exactly? This describeds more or less how mainteinership works:

https://www.debian.org/doc/manuals/maint-guide/

Nightwalker-87, 2020-03-29:
@bluca: Thx for the link, but I mainly try to gather information on who is taking care of the package distribution to repositories for this project. Are you still active on this topic?

bluca, 2020-03-29:
I am, for Debian and Ubuntu

Vascom, 2020-04-20:
I am maintainer for Fedora and EPEL (CentOS/RHEL).
https://src.fedoraproject.org/rpms/stlink

Nightwalker-87, 2020-06-18:
Thanks for the feedback.
This information has now been added to #874 which allows for closing this issue as resolved.

closed 2020-06-18 by Nightwalker-87