stlink-org/stlink #881: STM32F730 is misrecognized.

rewolff, 2020-03-18:
I was starting to write "is not supported" but upgrading to master helps. Good job! 

`2020-03-18T13:59:18 INFO common.c: Device connected is: F72 device, id 0x10006452
`
```
diff --git a/src/chipid.c b/src/chipid.c
index df07d52..f04c3ea 100644
--- a/src/chipid.c
+++ b/src/chipid.c
@@ -27,7 +27,7 @@ static const struct stlink_chipid_params devices[] = {
         {
             //RM0431 and DS document was used to find these paramaters
             .chip_id = STLINK_CHIPID_STM32_F72XXX,
-            .description = "F72 device",
+            .description = "F72/F73 device",
             .flash_type = STLINK_FLASH_TYPE_F4,
             .flash_size_reg = 0x1ff07a22,      // section 35.2
             .flash_pagesize = 0x800,           // No flash pages

```

rewolff, 2020-03-18:
SOURCE: 
Page 1372 of the reference manual: https://www.st.com/resource/en/reference_manual/dm00305990-stm32f72xxx-and-stm32f73xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

The page starts with making clear that this page is valid both for F72xx and F73xx devices.

rewolff, 2020-03-19:
The patch I made for this was pulled with the fix for #290 But if you want the more wordy F72x/F73x instead of the F72/F73 that I used to stay more in line with the existing code I can create another pull request.

Nightwalker-87, 2020-03-19:
No, that's just fine. We can address more than one issue with a single PR.

closed 2020-03-20 by Nightwalker-87