stlink-org/stlink #891: Build fails

mmikkone, 2020-03-21:
Make ends up in following error:

/home/osboxes/shared/Ohjelmointi/stlink-master/src/usb.c: In function ‘_stlink_usb_version’:
/home/osboxes/shared/Ohjelmointi/stlink-master/src/usb.c:192:18: error: comparison of integer expressions of different signedness: ‘ssize_t’ {aka ‘int’} and ‘uint32_t’ {aka ‘unsigned int’} [-Werror=sign-compare]
         if (size != rep_len) {
                  ^~
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/stlink-static.dir/build.make:89: CMakeFiles/stlink-static.dir/src/usb.c.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:73: CMakeFiles/stlink-static.dir/all] Error 2
make[1]: *** [Makefile:152: all] Error 2
make: *** [Makefile:28: release] Error 2

slyshykO, 2020-03-21:
Hello @mmikkone. Look like this issue was already fixed in develop branch.

mmikkone, 2020-03-21:
Thank you. I was able to build it from develop branch.

Nightwalker-87, 2020-03-21:
Fixed with PR #872.

closed 2020-03-21 by Nightwalker-87