stlink-org/stlink #896: undefined reference to `libusb_set_option'

purjus74, 2020-03-25:
Hi all,

I will try to compile the project on Windows 10 with MinGW-w64 without success.
The error is:
`C:/stlink/src/sg.c:972: undefined reference to 'libusb_set_option'`

My environement is:
- Windows 10
- texane/stlink 1.6.0
- MinGW-w64 i686-8.1.0-posix-dwarf-rt_v6-rev0

Can you help me?

Complete log of the compilation: [log.txt](https://github.com/texane/stlink/files/4381036/log.txt)

Thank you in advance.

Ju

Nightwalker-87, 2020-03-25:
Please try the `develop` branch instead. I am aware of fixes related to MinGW compilation since Release v1.6.0.

purjus74, 2020-03-25:
The mentinned error isn't displayed but the compilation failed.

[log_w_develop_latest_commit_a09ef54.txt](https://github.com/texane/stlink/files/4381273/log_w_develop_latest_commit_a09ef54.txt)

purjus74, 2020-03-25:
I want to add a remark.

The last log indicates that I have already libusb in my system.
`-- Found Libusb: C:/Windows/System32/libusb-1.0.dll`
Sure this version is old.

What is your opinion about this.

slyshykO, 2020-03-25:
@purjus74 I think the problem with libusb, that you download manually.
Please remove it. Clear all builds. And try one more time.

purjus74, 2020-03-25:
Just to give more details:
Before to use your project, my system had already a libusb instance.
[image]

If I let the batch mingw64-build.bat to download the libusb-1.0.22.7z, the extraction fails:
```
...
-- Could NOT find Libusb (missing: LIBUSB_INCLUDE_DIR)
-- downloading libusb 1.0.22

7-Zip 19.00 (x64) : Copyright (c) 1999-2018 Igor Pavlov : 2019-02-21

Scanning the drive for archives:
1 file, 2601 bytes (3 KiB)

Extracting archive: C:\stlink-develop\scripts\libusb-1.0.22.7z
ERROR: C:\stlink-develop\scripts\libusb-1.0.22.7z
C:\stlink-develop\scripts\libusb-1.0.22.7z
Open ERROR: Can not open the file as [7z] archive

ERRORS:
Is not archive
...
```
It seems that the archive is file is corrupted :(
[image]

Nightwalker-87, 2020-03-25:
@slyshykO: You are right, I did not notice at first that an external `libusb` library was present.

purjus74, 2020-03-25:
I think this behavior (error during libusb-1.0.22.7z downloading) is due to my compagny protection :(
This f*#! protection creates the wanted archive file but it contains an error html page !!!

[image]

But I'm not sure that is a problem to download before the good libusb archive and put it inside the build folder.
My first log illustrates that the mingw64-build.bat finds the archive, extracts it and uses it:
```
-- libusb archive already in build folder

7-Zip 19.00 (x64) : Copyright (c) 1999-2018 Igor Pavlov : 2019-02-21

Scanning the drive for archives:
1 file, 980895 bytes (958 KiB)

Extracting archive: C:\stlink\scripts\libusb-1.0.22.7z
--
Path = C:\stlink\scripts\libusb-1.0.22.7z
Type = 7z
Physical Size = 980895
Headers Size = 682
Method = LZMA2:23 BCJ
Solid = +
Blocks = 2

Everything is Ok
```

Nightwalker-87, 2020-03-25:
@purjus74: Does your whole compilation succeed now?
@slyshykO: I think we should add a note to the instructions on not to use external `libusb` packages when installing on windows, as this is not the first issue of its kind.

slyshykO, 2020-03-25:
The problem is that windows have many options on how to install libusb, also many compilers with different ABI. That is why we should provide windows builds.

purjus74, 2020-03-25:
I just tried on other machine (without old libusb inside C:\Windows\System32) and it's work.
Therefore, the problem is due to other libusb version already stored and used by the others projects.

purjus74, 2020-03-25:
Thank you for your support and for your job ;)

Nightwalker-87, 2020-03-25:
Reopening this as documentation and related work remains outstanding...

commit 27aa8882 (2020-04-04): Updated compiling doc & version support

- Added info on version support
- Updated compiling instructions
- Updated minGW-w64 gcc-TC to v8.1.0
- Minor formatting fixes

(Closes #896, #897, #899)
closed 2020-04-03 by Nightwalker-87