stlink-org/stlink #898: Installation issues on Ubuntu 18.04

MarvinKlemp, 2020-03-26:
For me it's impossible to install
Can someone maybe try to get a hint what could be wrong?

```
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release ..
sudo make install
sudo ldconfig
```

```
./st-info --probe  
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
 serial: 303637304646353635313534363637383637323132393139
openocd: "\x30\x36\x37\x30\x46\x46\x35\x36\x35\x31\x35\x34\x36\x36\x37\x38\x36\x37\x32\x31\x32\x39\x31\x39"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device
```

Nightwalker-87, 2020-03-26:
Please provide info on your system and stlink-tools version, otherwise we will not be able to help.

MarvinKlemp, 2020-03-27:
I'm gonna close this as this was actually a MCU issue.
I checked with another compatible MCU which responded fine.

I'm really sorry!

closed 2020-03-27 by MarvinKlemp