stlink-org/stlink #903: Use of Core-IDs in the codebase for MCU identification

Nightwalker-87, 2020-04-01:
We have Core-IDs defined in `/include/stlink/stm32.h` which are used to detect chips correctly.
However this is implemented differently for the various STM32 product lines which is not very clever nor follows any visible logic structure. The current situation is as follows:

F0: Cortex M0 / Core-ID: 0x0bb11477 --> _not used for detection_
F1: Cortex M3 / Core-ID: 0x1ba01477 --> used for detection as `STM32VL_CORE_ID`
F2: Cortex M3 / Core-ID: 0x2ba01477 --> _not used for detection_ (also used by F1 clones)
F3: Cortex M4F / Core-ID: 0x2ba01477 --> _not used for detection_
F4: Cortex M4F / Core-ID: 0x2ba01477 --> _not used for detection_
F7: Cortex M7F / Core-ID: 0x5ba01477 --> used for detection as `STM32F7_CORE_ID`
H7: Cortex M7F / Core-ID: 0x6ba02477 --> _not supported yet_

other families: L0 / L1 / L4 / G0 / G4 --> _Core-ID not used for detection_

In reference to #756 and #761 this seems not to have been implemented well over time and should be commonly addressed.

grevaillot, 2020-04-12:
hmm, while evaluating codebase to support H7/L5, i extracted a bit of an old H7 support patch and reworked it to do somethink like that

it's here : https://github.com/grevaillot/stlink/commit/5b8ade21b39aceeae50cd23f0afdff0adcb951de

i need to have a look at chips RM instead of defaulting to some default address for identification, but, as it, it works for the few devices i have in hand.

Nightwalker-87, 2020-04-12:
I'd propose to identify every MCUs via the following hierarchical workflow:
1) Read Core-ID
2) Read Chip-ID (by use of a unified macro naming scheme)
3) Read specific parameters taken from `/src/chip-id.c`

grevaillot, 2020-06-29:
grmpf, i need to finish back on that one, been busy recently and nothing rebases anymore :(

Nightwalker-87, 2020-06-29:
Hi there. No problem.
After Release v1.6.1 the transition to the new subfolder structure was finalised.
This is likely the reason why there are some minor issues.
I suggest to re-fork from `develop`.

Nightwalker-87, 2020-12-12:
Is anybody interested on taking action on this issue? It will likely require some time and efforts, but I'd be pleased to share some ideas with interested contributors on how to improve the design of the codebase structure in this point.

Ant-ON, 2020-12-15:
@Nightwalker-87 Using Core-ID may break support for some clones of stm32. This should be done with caution and extensive testing. OpenOСD uses CPUID in some places. I think it would be a more safe way.

Nightwalker-87, 2020-12-15:
Well, one could think about a secondary loop to identify such if the regular Core-ID-based approach fails. This would not exclude clones from support, but instead even help to identify such more easily. A secondary loop could relate to the current process of identification.

The whole procedure would clearly strip down the whole process to a very unified linear approach with additional functionality.

Nightwalker-87, 2021-05-14:
This approach should be discussed again. Please have your say.

houzhuhouzhu, 2021-07-29:
how to deal with ?

Nightwalker-87, 2021-07-29:
Yeah indeed, I've been waiting for some feedback here as well...

Ant-ON, 2021-07-30:
@houzhuhouzhu 
@Nightwalker-87 
A similar implementation was done in PR #1120. Now it is used in release version. I have not heard any problem on the work yet...

houzhuhouzhu, 2021-07-30:
[image]

oh look it，maybe,I can rewrite it

Nightwalker-87, 2021-07-31:
@Ant-ON Yes, but as far as I noticed there are even more chip IDs that are currently not used for detection.

PR #1216 (2022-01-28): [refactoring] General maintenance for code structure
List of changes and improvements:

- Revised device-specific parametres in the .chip files
- Aligned naming for .chip files and defines
- Unified formatting for .chip files
- Removed function `dump_a_chip` and old device database from `common.c`
- Moved-out STM32-specific defines to `stm32.h`
- Updated MCU core-ids
- Removed additional `extern "C"` linkage specs

(Closes #903) (Closes #1090) (Closes #1199) (Closes #1212)
closed 2022-01-28 by Nightwalker-87