stlink-org/stlink #909: Clean code with unified variable type

chenguokai, 2020-04-06:
As far as I can see, many functions use fixed-length variables. It is indeed useful for functions that need to deal with fixed data format. However the abuse of fixed-length variables are causing potential build failures and inessential casts.

For example in src/usb.c:
[image]

send_recv accept `txsize` and `rxsize` as `size_t` but in the actual context, those arguments have to be casted to `int` or `unsigned int`, without a fixed length, and whoever calls it provides arguments of a different type(`uint32_t` or magic numbers in this case). unintentional cast from 32 bit to 64 bit is applied at the calling process. Also, to prevent a build failure, many explicit casts are required.

I suggest replacing fixed length variables according to their actual usages. 
Staying fuzzy to be more portable.

I would like to give it a try if others find it meaningful.

Nightwalker-87, 2020-04-06:
Thx for putting this up on the agenda. As it is a general problem throughout the codebase, I'm not sure at the moment if we should put that into v1.6.1 straight away. @martonmiklos previously requested to do some cleanup related to code style, which I pushed into the v1.6.2 milestone. Maybe this would be a good place for this issue as well, though it will push it out a bit further.

chenguokai, 2020-04-06:
Maybe I can do some cleanup locally first and follow up the develop process.😃

Nightwalker-87, 2020-04-06:
Well of course you can, but please ensure to branch off from `develop` and keep following this branch, instead of `master`, which is no longer used for frequent changes, but only for releases and hotfixes.

Nightwalker-87, 2020-05-09:
With respect to compatibility between ILP32, LLP64 and LP64 I think it would be a good idea to replace the data type `long` with `int32_t`, but if one wants to rule out surprises in the future related to conversion, we may consider to move to fixed width integer types wherever possible...

Nightwalker-87, 2020-06-02:
@chenguokai: I'll leave this to you. You may contribute, if you fancy and have some some time.

chenguokai, 2020-06-07:
Sure I will, before the release. Getting busy now.😔

Nightwalker-87, 2021-05-14:
Is there anyone else who can deal with this?

Nightwalker-87, 2023-05-08:
As there is no common quality level of coding throughout the codebase, the mentioned explicit casting may be excessively used in some parts of the code while others appear to be designed and written quite well.

We are going to use fixed width data types for the first step in order to achieve a unified scheme on this side and most common appearance on different architectures. Further changes related to casting should be individually addressed in a second step (where necessary). Volunteers are welcome to review certain parts and contribute with their ideas and improvements to do so.

commit 5e85fd06 (2023-05-08): Replace data types with fixed width typedefs (C99)

- Unified variable types (Closes #909)
short --> int16_t
unsigned short --> uint16_t
int --> int32_t
unsigned int --> uint32_t
long --> int32_t
unsigned long --> uint32_t
long long --> int64_t
unsigned long long --> uint64_t

- Added missing header includes
closed 2023-05-08 by Nightwalker-87