stlink-org/stlink #914: Not building

Krasnov-Max, 2020-04-09:
/tmp/stlink $ sudo ./travis.sh 
-- C compilers available
/usr/bin/gcc
/usr/bin/gcc-5
/usr/bin/gcc-ar
/usr/bin/gcc-ar-5
/usr/bin/gcc-nm
/usr/bin/gcc-nm-5
/usr/bin/gcc-ranlib
/usr/bin/gcc-ranlib-5
/usr/bin/clang
/usr/bin/clang++
/usr/bin/clang-3.8
/usr/bin/clang++-3.8
/usr/bin/clang-apply-replacements
/usr/bin/clang-apply-replacements-3.8
/usr/bin/clang-check
/usr/bin/clang-check-3.8
/usr/bin/clang-cl-3.8
/usr/bin/clang-query
/usr/bin/clang-query-3.8
/usr/bin/clang-rename-3.8
/usr/bin/clang-tblgen
/usr/bin/scan-build
/usr/bin/scan-build-3.8
----
./travis.sh: строка 12: brew: команда не найдена
=== Building Debug
-- The C compiler identification is unknown
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/bin/cc" is not able to compile a simple test program.

  It fails with the following output:

   Change Dir: /home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeTmp

  

  Run Build Command:"/usr/bin/make" "cmTC_ea4f9/fast"

  /usr/bin/make -f CMakeFiles/cmTC_ea4f9.dir/build.make
  CMakeFiles/cmTC_ea4f9.dir/build

  make[1]: вход в каталог
  «/home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeTmp»

  Building C object CMakeFiles/cmTC_ea4f9.dir/testCCompiler.c.o

  /usr/bin/cc -o CMakeFiles/cmTC_ea4f9.dir/testCCompiler.c.o -c
  /home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeTmp/testCCompiler.c

  as: неизвестный ключ «--64»

  CMakeFiles/cmTC_ea4f9.dir/build.make:65: ошибка выполнения
  рецепта для цели
  «CMakeFiles/cmTC_ea4f9.dir/testCCompiler.c.o»

  make[1]: *** [CMakeFiles/cmTC_ea4f9.dir/testCCompiler.c.o] Ошибка 1

  make[1]: выход из каталога
  «/home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeTmp»

  Makefile:126: ошибка выполнения рецепта для
  цели «cmTC_ea4f9/fast»

  make: *** [cmTC_ea4f9/fast] Ошибка 2

  

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "/home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeOutput.log".
See also "/home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeError.log".
=== Building Release
-- The C compiler identification is unknown
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/bin/cc" is not able to compile a simple test program.

  It fails with the following output:

   Change Dir: /home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeTmp

  

  Run Build Command:"/usr/bin/make" "cmTC_d4cba/fast"

  /usr/bin/make -f CMakeFiles/cmTC_d4cba.dir/build.make
  CMakeFiles/cmTC_d4cba.dir/build

  make[1]: вход в каталог
  «/home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeTmp»

  Building C object CMakeFiles/cmTC_d4cba.dir/testCCompiler.c.o

  /usr/bin/cc -o CMakeFiles/cmTC_d4cba.dir/testCCompiler.c.o -c
  /home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeTmp/testCCompiler.c

  as: неизвестный ключ «--64»

  CMakeFiles/cmTC_d4cba.dir/build.make:65: ошибка выполнения
  рецепта для цели
  «CMakeFiles/cmTC_d4cba.dir/testCCompiler.c.o»

  make[1]: *** [CMakeFiles/cmTC_d4cba.dir/testCCompiler.c.o] Ошибка 1

  make[1]: выход из каталога
  «/home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeTmp»

  Makefile:126: ошибка выполнения рецепта для
  цели «cmTC_d4cba/fast»

  make: *** [cmTC_d4cba/fast] Ошибка 2

  

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "/home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeOutput.log".
See also "/home/maxx/tmp/stlink/build/Debug/CMakeFiles/CMakeError.log".

 ~/tmp/stlink $ uname -a
Linux maxx-pc 4.15.0-76-generic #86~16.04.1-Ubuntu SMP Mon Jan 20 11:02:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

slyshykO, 2020-04-09:
@Krasnov-Max , скрипт ./travis.sh предназначен для сборки системой последовательной интеграции (CI).
Если тебе интерстно как собрать из исходников, пожалуйста, ознакомься с https://github.com/stlink-org/stlink/blob/develop/doc/compiling.md .

Также, ознакомься с правилами оформления багов - https://github.com/stlink-org/stlink/blob/develop/.github/ISSUE_TEMPLATE/bug-report.md

Krasnov-Max, 2020-04-09:
> @Krasnov-Max , скрипт ./travis.sh предназначен для сборки системой последовательной интеграции (CI).
> Если тебе интерстно как собрать из исходников, пожалуйста, ознакомься с [...]

Ознакомился. Спасибо. Но это не помогло. не собирается и все тут. Ошибки на первом шаге make release.

slyshykO, 2020-04-09:
Установил нужные *-dev пакеты ?
```
libusb-1.0-0-dev
libgtk-3-dev
```

Какую версию _stlink_ ты собираешь ?
Так же хотелось бы увидеть лог ошибки.

Krasnov-Max, 2020-04-09:
> Установил нужные *-dev пакеты ?
> 
> https://github.com/stlink-org/stlink/blob/develop/doc/compiling.md#common-requirements-1
> 
> Какую версию _stlink_ ты собираешь ?
> Так же хотелось бы увидеть лог ошибки.

Версия 1.6.

Логи ошибок:

/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeError.log                                                                                        
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /usr/bin/cc
Build flags:
Id flags:

The output was:
1
as: неизвестный ключ «--64»

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /usr/bin/cc
Build flags:
Id flags: -c

The output was:
1
as: неизвестный ключ «--64»

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /usr/bin/cc
Build flags:
Id flags: -Aa

The output was:
1
<command-line>:0:1: error: missing '(' after predicate

Checking whether the C compiler is IAR using "" did not match "IAR .+ Compiler":
cc: fatal error: no input files
compilation terminated.
Determining if the C compiler works failed with the following output:
Change Dir: /home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_6d47a/fast"
make[1]: вход в каталог «/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp»
/usr/bin/make -f CMakeFiles/cmTC_6d47a.dir/build.make CMakeFiles/cmTC_6d47a.dir/build
make[2]: вход в каталог «/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp»
Building C object CMakeFiles/cmTC_6d47a.dir/testCCompiler.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_6d47a.dir/testCCompiler.c.o   -c /home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp/testCCompiler.c
as: неизвестный ключ «--64»
CMakeFiles/cmTC_6d47a.dir/build.make:65: ошибка выполнения рецепта для цели «CMakeFiles/cmTC_6d47a.dir/testCCompiler.c.o»
make[2]: *** [CMakeFiles/cmTC_6d47a.dir/testCCompiler.c.o] Ошибка 1
make[2]: выход из каталога «/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp»
Makefile:126: ошибка выполнения рецепта для цели «cmTC_6d47a/fast»
make[1]: *** [cmTC_6d47a/fast] Ошибка 2
make[1]: выход из каталога «/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp»

/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeOutput.log                                                                                         
The system is: Linux - 4.15.0-76-generic - x86_64

Krasnov-Max, 2020-04-09:
> Установил нужные *-dev пакеты ?
> 
> https://github.com/stlink-org/stlink/blob/develop/doc/compiling.md#common-requirements-1
Да нужные пакету установленны

slyshykO, 2020-04-09:
1 - покажи вывод команды 
```sh
gcc -v
```

2 выполни команду 
```sh
make release 2>&1 | tee buildlog.txt
```
и прикрепи к ответу файл **buildlog.txt**

Krasnov-Max, 2020-04-09:
maxx@maxx-pc ~/tmp/stlink $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu
5.4.0-6ubuntu1~16.04.12'
--with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs
--enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr
--program-suffix=-5 --enable-shared --enable-linker-build-id
--libexecdir=/usr/lib --without-included-gettext --enable-threads=posix
--libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu
--enable-libstdcxx-debug --enable-libstdcxx-time=yes
--with-default-libstdcxx-abi=new --enable-gnu-unique-object
--disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib
--disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo
--with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home
--with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64
--with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64
--with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar
--enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686
--with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib
--with-tune=generic --enable-checking=release --build=x86_64-linux-gnu
--host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)

[RELEASE]
Makefile:26: ошибка выполнения рецепта для цели «release»

slyshykO, 2020-04-09:
Еще раз и внимательно.
```sh
make clean; make release 2>&1 | tee buildlog.txt
```
**Выполни** команду и **приложи** к ответу **buildlog.txt**

Krasnov-Max, 2020-04-09:
-- The C compiler identification is unknown
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/bin/cc" is not able to compile a simple test program.

  It fails with the following output:

   Change Dir: /home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp

  

  Run Build Command:"/usr/bin/make" "cmTC_d70ca/fast"

  make[1]: вход в каталог
  «/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp»

  /usr/bin/make -f CMakeFiles/cmTC_d70ca.dir/build.make
  CMakeFiles/cmTC_d70ca.dir/build

  make[2]: вход в каталог
  «/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp»

  Building C object CMakeFiles/cmTC_d70ca.dir/testCCompiler.c.o

  /usr/bin/cc -o CMakeFiles/cmTC_d70ca.dir/testCCompiler.c.o -c
  /home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp/testCCompiler.c

  as: неизвестный ключ «--64»

  CMakeFiles/cmTC_d70ca.dir/build.make:65: ошибка выполнения
  рецепта для цели
  «CMakeFiles/cmTC_d70ca.dir/testCCompiler.c.o»

  make[2]: *** [CMakeFiles/cmTC_d70ca.dir/testCCompiler.c.o] Ошибка 1

  make[2]: выход из каталога
  «/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp»

  Makefile:126: ошибка выполнения рецепта для
  цели «cmTC_d70ca/fast»

  make[1]: *** [cmTC_d70ca/fast] Ошибка 2

  make[1]: выход из каталога
  «/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeTmp»

  

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeOutput.log".
See also "/home/maxx/tmp/stlink/build/Release/CMakeFiles/CMakeError.log".
Makefile:41: ошибка выполнения рецепта для цели «build/Release»
make: *** [build/Release] Ошибка 1

chenguokai, 2020-04-09:
From what the logs indicate, I suspect that the compiler on your machine is corrupted or misconfigured.
`as` is the program that generate binarys from assembly sources. Quite rare that you are experiencing `--64` error.
I suspect that some ENVs on your machine cause the C compiler to call another assembler rather than its corresponding one.
May you write a small program like hello world and try compiling it?

Krasnov-Max, 2020-04-09:
yes. Test programm is noc bulding error --64.

slyshykO, 2020-04-09:
@chenguokai . I don't know why but in Max's machine /usr/bin/cc is invalid

chenguokai, 2020-04-09:
@slyshykO  Yes, that's what I am trying to figure out.
From the log I can see multiple available compilers.
Problems will happen when something like this happens:
cc is pointed to gcc and as is pointed to llvm-as

I experienced similar problems a few days ago when I wanted to use GNU AS and LLVM/Clang at the same time.

slyshykO, 2020-04-09:
@Krasnov-Max 
```sh
mkdir -p build/release
cd ./build/release/
cmake -Wno-dev -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=gcc ../../
cmake --build .
```

slyshykO, 2020-04-09:
@chenguokai 
according to this 
```
-- Check for working C compiler: /usr/bin/cc -- broken
```
cc pointing to nothing.

Nightwalker-87, 2020-04-09:
Please **discuss in English** otherwise it is not possible or at least very complicated for others to follow as everything needs a translator.

chenguokai, 2020-04-09:
@slyshykO  In my view, it may indeed point to a compiler. The error indicates that `cc` cannot produce a working binary. This test is usually performed by trying compiling a small test program. The only known information concluded from the error is that either the compiler exited with error or its output is not valid.  Either a misconfigured compiler or an invalid soft link can lead to this program.

slyshykO, 2020-04-09:
@Krasnov-Max 
run 
```sh
/usr/bin/cc -v
```
ans show as output

chenguokai, 2020-04-09:
@Krasnov-Max May you provide an error log or the output of the following command?

`/usr/bin/cc -S YOUR_SMALL_PROGRAM`

Also slyshykO's output is useful.

Krasnov-Max, 2020-04-09:
Porogramm building. Generate hello.s
Output empty.

Krasnov-Max, 2020-04-09:
maxx@maxx-pc ~/tmp/stlink $ /usr/bin/cc -v
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu
5.4.0-6ubuntu1~16.04.12'
--with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs
--enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr
--program-suffix=-5 --enable-shared --enable-linker-build-id
--libexecdir=/usr/lib --without-included-gettext --enable-threads=posix
--libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu
--enable-libstdcxx-debug --enable-libstdcxx-time=yes
--with-default-libstdcxx-abi=new --enable-gnu-unique-object
--disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib
--disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo
--with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home
--with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64
--with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64
--with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar
--enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686
--with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib
--with-tune=generic --enable-checking=release --build=x86_64-linux-gnu
--host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)

chenguokai, 2020-04-09:
Seems not a problem with stlink.
I suggest a reboot or reinstall your compiler/toolchain to ensure a working compiler available.
If other compilers are proven to be good to compile your test program, you may also try specifying compilers to cmake

Krasnov-Max, 2020-04-09:
> @Krasnov-Max
> 
> ```shell
> mkdir -p build/release
> cd ./build/release/
> cmake -Wno-dev -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=gcc ../../
> cmake --build .
> ```

maxx@maxx-pc ~/tmp/stlink/build/release $ cmake -Wno-dev -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=gcc ../../
-- The C compiler identification is unknown
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/bin/gcc" is not able to compile a simple test program.

  It fails with the following output:

   Change Dir: /home/maxx/tmp/stlink/build/release/CMakeFiles/CMakeTmp

  

  Run Build Command:"/usr/bin/make" "cmTC_dc222/fast"

  /usr/bin/make -f CMakeFiles/cmTC_dc222.dir/build.make
  CMakeFiles/cmTC_dc222.dir/build

  make[1]: вход в каталог
  «/home/maxx/tmp/stlink/build/release/CMakeFiles/CMakeTmp»

  Building C object CMakeFiles/cmTC_dc222.dir/testCCompiler.c.o

  /usr/bin/gcc -o CMakeFiles/cmTC_dc222.dir/testCCompiler.c.o -c
  /home/maxx/tmp/stlink/build/release/CMakeFiles/CMakeTmp/testCCompiler.c

  as: неизвестный ключ «--64»

  CMakeFiles/cmTC_dc222.dir/build.make:65: ошибка выполнения
  рецепта для цели
  «CMakeFiles/cmTC_dc222.dir/testCCompiler.c.o»

  make[1]: *** [CMakeFiles/cmTC_dc222.dir/testCCompiler.c.o] Ошибка 1

  make[1]: выход из каталога
  «/home/maxx/tmp/stlink/build/release/CMakeFiles/CMakeTmp»

  Makefile:126: ошибка выполнения рецепта для
  цели «cmTC_dc222/fast»

  make: *** [cmTC_dc222/fast] Ошибка 2

  

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "/home/maxx/tmp/stlink/build/release/CMakeFiles/CMakeOutput.log".
See also "/home/maxx/tmp/stlink/build/release/CMakeFiles/CMakeError.log".

chenguokai, 2020-04-09:
I saw that you have clang installed. Give clang a try. It may work.

Krasnov-Max, 2020-04-09:
> Seems not a problem with stlink.
> I suggest a reboot or reinstall your compiler/toolchain to ensure a working compiler available.
> If other compilers are proven to be good to compile your test program, you may also try specifying compilers to cmake

Restart PC. Not bulding

Krasnov-Max, 2020-04-09:
> I saw that you have clang installed. Give clang a try. It may work.

maxx@maxx-pc ~/tmp/stlink $ sudo apt-get install clang
[sudo] пароль для maxx: 
Чтение списков пакетов… Готово
Построение дерева зависимостей       
Чтение информации о состоянии… Готово
Уже установлен пакет clang самой новой версии (1:3.8-33ubuntu3.1).
Обновлено 0 пакетов, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 24 пакетов не обновлено.

chenguokai, 2020-04-09:
@Krasnov-Max  I mean, try using clang to compile your test program, instead of the buggy gcc.
If clang is also corrupted, I am afraid that we cannot move further here.

Krasnov-Max, 2020-04-09:
> @Krasnov-Max I mean, try using clang to compile your test program, instead of the buggy gcc.
> If clang is also corrupted, I am afraid that we cannot move further here.

Ok
Test program not building.  OutputsCLang:

maxx@maxx-pc ~/tmp/stlink $ 
clang hello.c 
/usr/bin/ld: unrecognised emulation mode: elf_x86_64
Supported emulations: armelf
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Krasnov-Max, 2020-04-09:
maxx@maxx-pc ~/tmp/stlink $ clang -v hello.c 
clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/5.4.0
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/6.0.0
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/6.0.0
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/5.4.0
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/6.0.0
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/5.4.0
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/6.0.0
Selected GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0
Candidate multilib: .;@m64
Selected multilib: .;@m64
 "/usr/lib/llvm-3.8/bin/clang" -cc1 -triple x86_64-pc-linux-gnu -emit-obj -mrelax-all -disable-free -disable-llvm-verifier -main-file-name hello.c -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -munwind-tables -fuse-init-array -target-cpu x86-64 -v -dwarf-column-info -debugger-tuning=gdb -resource-dir /usr/lib/llvm-3.8/bin/../lib/clang/3.8.0 -internal-isystem /usr/local/include -internal-isystem /usr/lib/llvm-3.8/bin/../lib/clang/3.8.0/include -internal-externc-isystem /usr/include/x86_64-linux-gnu -internal-externc-isystem /include -internal-externc-isystem /usr/include -fdebug-compilation-dir /home/maxx/tmp/stlink -ferror-limit 19 -fmessage-length 80 -fobjc-runtime=gcc -fdiagnostics-show-option -fcolor-diagnostics -o /tmp/hello-1127ca.o -x c hello.c
clang -cc1 version 3.8.0 based upon LLVM 3.8.0 default target x86_64-pc-linux-gnu
ignoring nonexistent directory "/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/include
 /usr/lib/llvm-3.8/bin/../lib/clang/3.8.0/include
 /usr/include/x86_64-linux-gnu
 /usr/include
End of search list.
 "/usr/bin/ld" --eh-frame-hdr -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 -o a.out /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/crt1.o /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/crti.o /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/crtbegin.o -L/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0 -L/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu -L/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../lib64 -L/lib/x86_64-linux-gnu -L/lib/../lib64 -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib64 -L/usr/lib/x86_64-linux-gnu/../../lib64 -L/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../.. -L/usr/lib/llvm-3.8/bin/../lib -L/lib -L/usr/lib /tmp/hello-1127ca.o -lgcc --as-needed -lgcc_s --no-as-needed -lc -lgcc --as-needed -lgcc_s --no-as-needed /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/crtend.o /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/crtn.o
/usr/bin/ld: unrecognised emulation mode: elf_x86_64
Supported emulations: armelf
clang: error: linker command failed with exit code 1 (use -v to see invocation)
maxx@maxx-pc ~/tmp/stlink $

chenguokai, 2020-04-09:
@Krasnov-Max I suggest you close this issue for now.
After you have a working toolchain to compile at least your test program, if build problem remains, you may reopen this issue.

Krasnov-Max, 2020-04-09:
> @Krasnov-Max I suggest you close this issue for now.
> After you have a working toolchain to compile at least your test program, if build problem remains, you may reopen this issue.
> 
> Your logs indicates that you have a broken or not corresponding linker for your host system.

Ok How to close this issue?

Krasnov-Max, 2020-04-09:
Ok Reair gcc. 

maxx@maxx-pc ~/tmp/stlink $ make
-- The C compiler identification is GNU 6.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- stlink version: 1.6.0-145-g38c8b8b
-- Major 1 Minor 6 Patch 0
-- Performing Test C_SUPPORTS_STD_GNU11
-- Performing Test C_SUPPORTS_STD_GNU11 - Success
-- Performing Test C_SUPPORTS_WALL
-- Performing Test C_SUPPORTS_WALL - Success
-- Performing Test C_SUPPORTS_WEXTRA
-- Performing Test C_SUPPORTS_WEXTRA - Success
-- Performing Test C_SUPPORTS_WSHADOW
-- Performing Test C_SUPPORTS_WSHADOW - Success
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2 - Success
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING - Success
-- Performing Test C_SUPPORTS_WUNDEF
-- Performing Test C_SUPPORTS_WUNDEF - Success
-- Performing Test C_SUPPORTS_WFORMAT
-- Performing Test C_SUPPORTS_WFORMAT - Success
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY - Success
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED - Success
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS - Failed
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32 - Failed
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION - Success
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS - Success
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Success
-- Performing Test C_SUPPORTS_O2
-- Performing Test C_SUPPORTS_O2 - Success
-- Performing Test C_SUPPORTS_WERROR
-- Performing Test C_SUPPORTS_WERROR - Success
-- Found Libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- found libusb library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.18.9
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for __stack_chk_fail in ssp
-- Looking for __stack_chk_fail in ssp - not found
-- STLINK_LIB_SHARED: stlink
-- PROJECT_VERSION_MAJOR: 1
-- VERSION: 1.6.0
-- Manpage generation disabled
-- Debian-based Linux OS detected
-- Configuring done
-- Generating done
-- Build files have been written to: /home/maxx/tmp/stlink/build/Release
[RELEASE]
Scanning dependencies of target stlink
[  2%] Building C object CMakeFiles/stlink.dir/src/chipid.c.o
[  5%] Building C object CMakeFiles/stlink.dir/src/common.c.o
[... 129 lines cut ...]
/usr/include/sys/_types.h:40:20: note: previous declaration of ‘__fsfilcnt_t’ was here
 typedef __uint32_t __fsfilcnt_t;
                    ^~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/sys/mman.h:23:0,
                 from /home/maxx/tmp/stlink/include/stlink/mmap.h:5,
                 from /home/maxx/tmp/stlink/src/common.c:16:
/usr/include/x86_64-linux-gnu/bits/types.h:181:19: error: conflicting types for ‘__loff_t’
 typedef __off64_t __loff_t; /* Type of file sizes and offsets (LFS).  */
                   ^~~~~~~~
In file included from /usr/include/sys/reent.h:15:0,
                 from /usr/include/stdio.h:60,
                 from /home/maxx/tmp/stlink/src/common.c:4:
/usr/include/sys/_types.h:103:18: note: previous declaration of ‘__loff_t’ was here
 typedef _off64_t __loff_t;
                  ^~~~~~~~
CMakeFiles/stlink.dir/build.make:86: ошибка выполнения рецепта для цели «CMakeFiles/stlink.dir/src/common.c.o»
make[3]: *** [CMakeFiles/stlink.dir/src/common.c.o] Ошибка 1
CMakeFiles/Makefile2:104: ошибка выполнения рецепта для цели «CMakeFiles/stlink.dir/all»
make[2]: *** [CMakeFiles/stlink.dir/all] Ошибка 2
Makefile:149: ошибка выполнения рецепта для цели «all»
make[1]: *** [all] Ошибка 2
Makefile:26: ошибка выполнения рецепта для цели «release»
make: *** [release] Ошибка 2

maxx@maxx-pc ~/tmp/stlink $

Nightwalker-87, 2020-04-09:
@chenguokai: Let me please have a word here. If several people here close tickets just because they are able to clearly doesn't help. I have not even had the chance to read this ticket and a new one on the same topic has been reopened already. I can't manage to keep track and sort everything if it goes like this. So please leave that to one person, otherwise we will end up in a state we had some time ago where tickets could not be allocated any more. This had to be done later on which was very time consuming and a lot of work. So please - calm down a bit - this is not a race. :-/

Nightwalker-87, 2020-04-09:
Ok, so going though this (actually invalid) ticket reveals that the original issue is not related to the `stlink` toolset and goes back to a broken local toolchain. As I can see @Krasnov-Max has managed to solve this problem with a clean GNU 6.5.0 C-toolchain and has subsequently opened a new ticket in the intended way. Thx for that. Thus I am closing this now as resolved.

@chenguokai: You may tag new tickets though as soon as they appear with respect to the provided information. This would be helpful. I'll then define a suitable milestone afterwards.

closed 2020-04-09 by Nightwalker-87