stlink-org/stlink #915: License of some flashloaders source files are still missing

Nightwalker-87, 2020-04-09:
Pick up of an issue mentioned before by @xor-gate in #682:

> we need to update LICENSE and add the GPL part also for those files. I had some discussion in an email with the original authors which seemed to be copied from the OpenOCD project into stlink project. So we have a mix of GPL van BSD licensed code. If something is unclear please post a comment.

Nightwalker-87, 2020-04-09:
I'm going to look at this now.
Generally speaking I would not add such additional notes on licensing to the LICENSE file because that may possibly break the correct license recognition and display in git-header bar of the project, (text does not match the template any more) - not definitely sure, but have seen something similar in another project before. Thus this should better go into our README, but somewhere in the header part, that it is easy to spot.

Nightwalker-87, 2020-04-09:
@xor-gate: Did you only mean the two flashloaders for which the source files have a header, or did you have in mind that (in addition to these) there were also others (without any header) copied from openocd, but without any reference? If the latter is the case, one would have to compare line by line to identify. Regarding the former ones: These were indeed correctly declared - but somewhere at the very bottom of our README file. I'll move this important note to the top.

Nightwalker-87, 2020-04-12:
ping @xor-gate

chenguokai, 2020-04-15:
Maybe we can just clean room the source code to make a pure BSD one.
I checked to confirm that these assemblies are small enough that a clean room can be finished in a short time. May need a volunteer  who has not read the GPL ones to do the actual coding part.

Nightwalker-87, 2020-04-15:
Good idea. Indeed it would be nice to have an independent solution for this project.

chenguokai, 2020-04-15:
Volunteer found.

I will take part in this process.
Before that, I wonder which file(s) will need a  clean room.
The one for F0 and lx seems to be more complex and may be not from the same source. Should all the assemblies be rewritten? A complete list of files that need a clean room would be appreciated.

Nightwalker-87, 2020-04-15:
Well at first I'd suggest to review what is currently there and whether there are any redundant files or sections. Here are the related references in git history, when the loader files appeared:

8151bf5c4a443608d4fec15cf7dfe79a02cdd096 F2/F4
e8b15c0e787c7d5fc90c35d53165fcf36afb39e3 F0
94c7ac86dcb2a18ab4b2b60309892f6102a53205 Lx
44c645b7d74b7669b0e3988155528c2edcfbb05e L0x
71535104e47f2a96ce6a87a6da0fc4519bc64d15 F4lv
0419b8bec68b08acee490b4f0fcd35f9c7f4a496 L4
bc48e99b194dba002628a3b841fd18cb4dd2762e F7/F7lv

Some don't have any documentation within the file. I would find it desirable to have a header in each which explains the file content or at least holds a reference to an official STMicroelectronics document.

You may also join https://hackaday.io with your github-login, to consult on details, if you like.

chenguokai, 2020-04-15:
The src for F0 seems to be verbose. I will try optimize it in a unified way that others are mostly taken.
The others are mostly doing a copy and paste work in a clear way that is not so realistic to be more optimized.

The common problem is indeed the lack of references. Should be easy to solve thanks to commit 44c645b and 7153510.

Nightwalker-87, 2020-04-22:
@chenguokai: I am currently updating the deb packaging, including all required additional files, following official Debian policies. Taking this issue already into account, I'll update the license file to BSD-3 only. This would imply that there are no old flashloader source files present with incompatible licensing by the time of merging the enhanced cmake packaging stuff.

Nightwalker-87, 2020-04-27:
If it is not too much effort one may have a look at some of the older flash loader issues, which have been grouped in two separate milestones. It is possible that some issues have resolved with this rewrite as well, either due to an alternative implementation or fixes within the order of command calls.

chenguokai, 2020-04-27:
One problem makes nearly all reported error logs of flashloader related ones useless: No device dump of the error state.

Currently no info is dumped from device, no register values, no RAM datas, etc. Thus we cannot identify what has gone wrong in the underlying process.

I am going to be busy recently and I may not have time to implement a more detailed error log, while in my view, it would be essential for solving any flashloader related issues.

Nightwalker-87, 2020-04-27:
Ok, so I'd suggest to address that at a later point in time, and by then close all flash loader issues, which can no longer be investigated, to have a clean reference there. In the meanwhile I will check, if some of the original issuers are still active and available for any further feedback (but I doubt this will be the case).

closed 2020-05-06 by Nightwalker-87