stlink-org/stlink #916: stlink v1.6.0

Krasnov-Max, 2020-04-09:
Thank you for giving feedback to the stlink project.

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: [stlink v3] (e.g Stlink /v1, /v2, /v2-clone, /v2-1)
- Programmer firmware version: [enter here] (e.g STSW-LINK007 2.27.15)
- Operating system and version: [LINUX] (e.g Linux, Mac OS X, Windows)
- **Stlink tools version** and/or git commit hash: [1.6.0] (e.g v1.1.0/git-c722056)
not bulding 
maxx@maxx-pc ~/tmp/stlink $ make
-- The C compiler identification is GNU 6.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- stlink version: 1.6.0-145-g38c8b8b
-- Major 1 Minor 6 Patch 0
-- Performing Test C_SUPPORTS_STD_GNU11
-- Performing Test C_SUPPORTS_STD_GNU11 - Success
-- Performing Test C_SUPPORTS_WALL
-- Performing Test C_SUPPORTS_WALL - Success
-- Performing Test C_SUPPORTS_WEXTRA
-- Performing Test C_SUPPORTS_WEXTRA - Success
-- Performing Test C_SUPPORTS_WSHADOW
-- Performing Test C_SUPPORTS_WSHADOW - Success
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2 - Success
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING - Success
-- Performing Test C_SUPPORTS_WUNDEF
-- Performing Test C_SUPPORTS_WUNDEF - Success
-- Performing Test C_SUPPORTS_WFORMAT
-- Performing Test C_SUPPORTS_WFORMAT - Success
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY - Success
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED - Success
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS - Failed
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32 - Failed
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION - Success
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS - Success
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Success
-- Performing Test C_SUPPORTS_O2
-- Performing Test C_SUPPORTS_O2 - Success
-- Performing Test C_SUPPORTS_WERROR
-- Performing Test C_SUPPORTS_WERROR - Success
-- Found Libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- found libusb library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.18.9
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for __stack_chk_fail in ssp
-- Looking for __stack_chk_fail in ssp - not found
-- STLINK_LIB_SHARED: stlink
-- PROJECT_VERSION_MAJOR: 1
-- VERSION: 1.6.0
-- Manpage generation disabled
-- Debian-based Linux OS detected
[... 136 lines cut ...]
/usr/include/sys/_types.h:40:20: note: previous declaration of ‘__fsfilcnt_t’ was here
 typedef __uint32_t __fsfilcnt_t;
                    ^~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/sys/mman.h:23:0,
                 from /home/maxx/tmp/stlink/include/stlink/mmap.h:5,
                 from /home/maxx/tmp/stlink/src/common.c:16:
/usr/include/x86_64-linux-gnu/bits/types.h:181:19: error: conflicting types for ‘__loff_t’
 typedef __off64_t __loff_t; /* Type of file sizes and offsets (LFS).  */
                   ^~~~~~~~
In file included from /usr/include/sys/reent.h:15:0,
                 from /usr/include/stdio.h:60,
                 from /home/maxx/tmp/stlink/src/common.c:4:
/usr/include/sys/_types.h:103:18: note: previous declaration of ‘__loff_t’ was here
 typedef _off64_t __loff_t;
                  ^~~~~~~~
CMakeFiles/stlink.dir/build.make:86: ошибка выполнения рецепта для цели «CMakeFiles/stlink.dir/src/common.c.o»
make[3]: *** [CMakeFiles/stlink.dir/src/common.c.o] Ошибка 1
CMakeFiles/Makefile2:104: ошибка выполнения рецепта для цели «CMakeFiles/stlink.dir/all»
make[2]: *** [CMakeFiles/stlink.dir/all] Ошибка 2
Makefile:149: ошибка выполнения рецепта для цели «all»
make[1]: *** [all] Ошибка 2
Makefile:26: ошибка выполнения рецепта для цели «release»
make: *** [release] Ошибка 2

maxx@maxx-pc ~/tmp/stlink $

Nightwalker-87, 2020-04-09:
ok, so I consider this as a valid ticket in place for the previous one.

1) Looking at your header reveals that you are trying to use the ST-Link-v3 programmer.
This one is not yet supported by the tools. So far there have only been a few preceded fixes in the codebase. Support for the ST-Link-v3 is currently planned for the next release, depending on progress on this topic. Please track #820 for this.

2) The compilation error still needs a review. I'll have a look soon and will get back to it.

chenguokai, 2020-04-09:
Similar problem with ubuntu:
[https://github.com/idc/ps4-payload-sdk/issues/8](https://github.com/idc/ps4-payload-sdk/issues/8)

I will check on my ubuntu 18.04 machine.

Nightwalker-87, 2020-04-09:
If so, this would not be specifically related to `st-link`.

chenguokai, 2020-04-09:
Similar problems can be found all over the internet, though no convincing solutions have been found.
A further test could be:

Write a small program that uses one of those conflicting types, include the headers that causes the conflict. If this program compile with errors, problems lie in the standard C library rather than stlink, otherwise we should recheck the way stlink include headers.

Nightwalker-87, 2020-04-09:
As hard as this may sound for the people who run into such problems, but I'm afraid to say that in my view this is not a general operating system or toolchain support forum. If we open up for all that and dig deeper into such things, we can not focus on `st-link` and ensure that at least our toolset does not introduce any issues...

So should it turn out that this is such a problem, then please provide an appropriate reference for @Krasnov-Max where he can find the help he deserves with this (e.g. a Debian or Ubuntu help chat). Regardless to this we are open for further help of course, if things get to the toolset level.

chenguokai, 2020-04-09:
For a Debian system,  check [https://www.debian.org/support](https://www.debian.org/support)
For Ubuntu, [https://ubuntu.com/support/community-support](https://ubuntu.com/support/community-support)

If none of the two works, the stlink package maintainer of the distro might be more experienced  in solving things of this kind.

slyshykO, 2020-04-09:
@Nightwalker-87 think this issue have nothing with stlinkV3

Nightwalker-87, 2020-04-09:
Not, so far but as he filled it out in the opening post, this would have been the next problem he would have run into. That's why I mentioned that in the first point.

Nightwalker-87, 2020-04-10:
Ok, so as it turns out, this compilation failure is indeed related to the local toolchain. Further we can't reproduce it on a comparable ubuntu operating system and thus are not able to confirm any correlation to the `stlink` toolset. As sufficient help has been provided to the author be able to proceed, I am closing this ticket now. Thanks everybody for helping out!

closed 2020-04-10 by Nightwalker-87