stlink-org/stlink #919: develop branch is broken on 32 bit system again

chenguokai, 2020-04-11:
Thank you for giving feedback to the stlink project.

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:
- Operating system and version: [Mac OS X] (e.g Linux, Mac OS X, Windows)
- **Stlink tools version** and/or git commit hash: [ecbbd6db7aa8dff983ee43f629ac920f20995d12] (e.g v1.1.0/git-c722056)

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

```
...
Scanning dependencies of target stlink-static
[  3%] Building C object CMakeFiles/stlink-static.dir/src/chipid.c.o
[  6%] Building C object CMakeFiles/stlink-static.dir/src/common.c.o
cc1: warnings being treated as errors
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c: In function ‘set_flash_cr_pg’:
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:440: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c: In function ‘write_flash_cr_bker_pnb’:
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:758: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:759: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:763: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:763: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c: In function ‘calculate_L4_page’:
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:1723: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:1734: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c: In function ‘stlink_erase_flash_page’:
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:1910: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:1919: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:1926: warning: implicit conversion shortens 64-bit value into a 32-bit value
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c: In function ‘stlink_erase_flash_mass’:
/Users/xim/Desktop/Parallels Shared Folders/Home/Documents/macports-maintain/stlink/src/common.c:2003: warning: implicit conversion shortens 64-bit value into a 32-bit value
make[3]: *** [CMakeFiles/stlink-static.dir/src/common.c.o] Error 1
make[2]: *** [CMakeFiles/stlink-static.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [release] Error 2
```

Expected/description:

`Should compile on a 32 bit system`

I will make a patch for this.

Similar to https://github.com/stlink-org/stlink/issues/629

**NOTICE: This bug report may be closed without further notice, if not enough information is provided!**

Thank you for your support.

The stlink project maintainers

slyshykO, 2020-04-11:
Copy from https://github.com/stlink-org/stlink/issues/629#issuecomment-612415954

In my opinion, using **mmap** function brings us many problems. Maybe we should stop using it.
Maybe better for all will be using functions from _libc_ for work with files? This shouldn't impact on app speed dramatically.

chenguokai, 2020-04-11:
@slyshykO Agree with your idea, at least a self-implemented `mmap` is not so good for any POSIX-compliant systems, like Linux, macOS and BSDs. 
when talking about this issue, it is not related to `mmap` actually.

closed 2020-04-12 by Nightwalker-87