stlink-org/stlink #923: STM32L422CB: st-flash fails to flash ihex image

lulle2007200, 2020-04-12:
Trying to flash a small hex file to a STM32L422CB. The hex file is fine, flashes successfully with st-link utility. st-flash erases the first 2kb page of flash successfully, but doesnt write the the hex file at all.

- Programmer/board type: STLink V2
- Programmer firmware version: V2.J34.S7
- Operating system and version: Windows 10
- Stlink commandline tool name: st-flash v.1.6.0
- Target chip (and board if applicable): STM32L422CB

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

```
PS C:\Users\xxx\Documents\Code\Test> st-flash --format ihex write ./build/test.hex
st-flash 1.6.0
2020-04-12T18:28:57 INFO common.c: Loading device parameters....
2020-04-12T18:28:57 INFO common.c: Device connected is: L41x device, id 0x10006464
2020-04-12T18:28:57 INFO common.c: SRAM size: 0xa000 bytes (40 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 2048 bytes
2020-04-12T18:28:57 INFO common.c: Attempting to write 592 (0x250) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2020-04-12T18:28:57 INFO common.c: Finished erasing 1 pages of 2048 (0x800) bytes
2020-04-12T18:28:57 INFO common.c: Starting Flash write for F2/F4/L4
2020-04-12T18:28:57 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 592
2020-04-12T18:28:57 INFO common.c: Starting verification of write complete
2020-04-12T18:28:57 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
PS C:\Users\xxx\Documents\Code\Test>
```

Expected/description:

`Should flash the hex file without errors`

Nightwalker-87, 2020-04-18:
Who can take a closer look at this?

chenguokai, 2020-04-25:
I suggest a check with the same board and other flash tools so that we can confirm this is not a hardware problem.
The verification failed at the very beginning so if it is not a hardware problem, there might be some problems in flashloaders. Unfortunately error code check in flashloaders (if possible) is not applied on most models. I would suggest print out all register values after the failed flash, with `st-util` and (`gdb` or `lldb`).

lulle2007200, 2020-04-27:
I tried with both, a genuine st link v2 and one of those cheap copies from china. It fails on both. I will look at the registers later when i have some more time.

chenguokai, 2020-04-27:
The "flash tools" I mentioned refers to openocd or the official tools by ST. Sorry for the ambiguous expression. When you check the register values, it would be better to print out `STM32_FLASH_SR` register value, located in STM32_FLASH_REGS_BASE + 0x10. STM32_FLASH_REGS_BASE is 0x40022000 on STM32L4.

lulle2007200, 2020-05-04:
Sorry for replying so late.
I tried to flash a few different hex files of different sizes, st-flash successfully erases the the needed pages (verified by examining memory content with st link utility), but fails to flash as i said in initial post.
Flash is successful when flashing the hex with st link utility or the corresponding dfu with the st dfu tool, so i'd say that it's not a hardware issue.

Here is what STM32_FLASH_SR reads:
[image]

chenguokai, 2020-05-04:
The clue is that FLASH_SR shows PGSERR and PGAERR are set, that is, there are some alignment issues. If you can provide the general purpose register values after a failed programming (with `gdb`, `info registers` without reseting the board before attaching) and launch `st-flash` with `--debug` on, I think we will be able to further investigate.

lulle2007200, 2020-05-04:
[image]

these are the registers after failed flash (connected without reset).

If i run st-flash with the debug option being set, it doesnt fail.

lulle2007200, 2020-05-04:
(dont have gdb,

chenguokai, 2020-05-04:
@lulle2007200 in the mail I received you said that with —debug it flashes without error. I noticed that you edited it.
Was there indeed a successful flashing, with debug on?

If so, it may be a timing problem, just like that of f0 series.

From the log, things seem to be fine before running. the general purpose registers hold values that are hard to explain. If r0 was zero at the very beginning, maybe some issue with functions that write values to registers?

lulle2007200, 2020-05-04:
It didnt work, i messed up while trying to redirect output of st-flash to file. turned out it didnt run at all, but i had flashed it before using st link utilty...
Also, i hit close and comment instead of comment by accident, sorry for the confusion.
I will take another look tomorrow.

Nightwalker-87, 2020-07-05:
@lulle2007200: Any update on this?

lulle2007200, 2020-07-06:
I kind of forgot about this, life got in the way.
st-flash fails to erase and flash the provided hex file.
I multiple files of different sizes, doesn't work with any of them. Note that flashing those with the official ST-Link Utility works perfectly fine.
I have attached a log of the st-flash output and the values of the general purpose registers and STM32_FLASH_SR, both, before and after the flash attempt.
[log.txt](https://github.com/stlink-org/stlink/files/4881133/log.txt)
[image]

Nightwalker-87, 2020-07-08:
Now we are looking for a skilled developer who his somehow familiar with the L4 series to have the chance for a fix.

Nightwalker-87, 2023-01-15:
@grevaillot I've noticed that you are still involved in STM32 projects and notably contributed to this project previously on some advanced topics. If there is anything you can help us with (e.g. this issue, or even anything else) , please let me know. Even an idea or draft can turn out to be helpful.

eschoker, 2023-07-19:
I can confirm same issue on L0. 
I am on STM32L031 the nucleo-L031K6 board. I flashes normally via STM32 Programmer no issues. But when using stlink I get the following output for .hex

`st-flash 1.7.0-11-ge662da0
2023-07-19T10:17:53 INFO common.c: L0xx Category 2: 8 KiB SRAM, 32 KiB flash in at least 128 byte pages.
file Cube/build/stm32l0xxTemplate.hex md5 checksum: 21f17d8ec7f4da9be23f3145597739, stlink checksum: 0x000bc742
2023-07-19T10:17:53 INFO common.c: Attempting to write 14752 (0x39a0) bytes to stm32 address: 134217728 (0x8000000)
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000000 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000080 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000100 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000180 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000200 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000280 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000300 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000380 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000400 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000480 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000500 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000580 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000600 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000680 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000700 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000780 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000800 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000880 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000900 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000980 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000a00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000a80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000b00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000b80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000c00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000c80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000d00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000d80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000e00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000e80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000f00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08000f80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001000 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001080 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001100 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001180 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001200 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001280 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001300 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001380 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001400 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001480 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001500 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001580 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001600 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001680 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001700 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001780 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001800 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001880 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001900 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001980 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001a00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001a80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001b00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001b80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001c00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001c80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001d00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001d80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001e00 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001e80 erased
2023-07-19T10:17:53 INFO common.c: Flash page at addr: 0x08001f00 erased
[... 102 lines cut ...]
2023-07-19T10:22:35 INFO common.c: Flash page at addr: 0x08000f00 erased
2023-07-19T10:22:35 INFO common.c: Flash page at addr: 0x08000f80 erased
2023-07-19T10:22:35 INFO common.c: Flash page at addr: 0x08001000 erased
2023-07-19T10:22:35 INFO common.c: Flash page at addr: 0x08001080 erased
[... cut ...]
2023-07-19T10:22:35 INFO common.c: Go to Thumb mode
2023-07-19T10:22:35 ERROR flash_loader.c: Flash loader run error
2023-07-19T10:22:35 WARN flash_loader.c: Loader state: R2 0xF1000003 R15 0xF1000003
2023-07-19T10:22:35 WARN flash_loader.c: MCU state: DHCSR 0x1080009 DFSR 0x0 CFSR 0x0 HFSR 0x0
2023-07-19T10:22:35 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2023-07-19T10:22:35 WARN common.c: 
write_half_pages failed == -1`

After an unsuccessful flash I can only restart the board.

Nightwalker-87, 2023-07-20:
There could be some entries missing in `common_flash.c`  L967ff.
I believe L4-specific routines in `int32_t stlink_erase_flash_page()` should include all L4 devices:

```
    // select the page to erase
    if ((sl->chip_id == STM32_CHIPID_L4) ||
        (sl->chip_id == STM32_CHIPID_L41x_L42x) ||
        (sl->chip_id == STM32_CHIPID_L43x_L44x) ||
        (sl->chip_id == STM32_CHIPID_L45x_L46x) ||
        (sl->chip_id == STM32_CHIPID_L47x_L48x) ||
        (sl->chip_id == STM32_CHIPID_L496x_L4A6x) ||
        (sl->chip_id == STM32_CHIPID_L4Px_L4Qx) ||
        (sl->chip_id == STM32_CHIPID_L4Rx)) {
```

Some of these were missing. Please retry for the L4.
Will check on the L0 as well, but the way of implementation may be different there.

Nightwalker-87, 2023-09-02:
@Ant-ON What are your thoughts when looking at this?

Nightwalker-87, 2023-09-16:
@eschoker This is a different issue and not related to this ticket.
Please refer to #681 instead which covers this topic.
PR #1330 is the related fix and also lists all related duplicate tickets.

Nightwalker-87, 2023-09-16:
This issue may have been fixed in #1113.
Please verify this, if you have the same or a very similar chip at hand.

Nightwalker-87, 2023-11-26:
@Ant-ON Is this fixed by now?

Nightwalker-87, 2023-12-24:
One can assume that this issue has been fixed in #1113.
Unfortunately we are not able to verify or further investigate this due to missing feedback and contribution.
Against this background I am closing this ticket now as resolved.
Please open a new ticket should the problem persists.

closed 2023-12-24 by Nightwalker-87