stlink-org/stlink #926: make test target broken

grevaillot, 2020-04-14:
Hi,

Apparently, unitary tests are broken - i've no idea of the current level of brokenness, but, at least they do not run anymore due to some binary path change. I think this was a recent modification.

> 
>  % make test
> [DEBUG]
> [ 22%] Built target stlink
> [ 27%] Built target st-info
> [ 36%] Built target st-flash
> [ 58%] Built target stlink-static
> [ 69%] Built target st-util
> [ 75%] Built target stlink-gui
> [ 80%] Built target stlink-gui-local
> [ 88%] Built target flash
> [ 94%] Built target sg
> [100%] Built target usb
> Running tests...
> Test project /home/moot/Developpement/tools/stlink/build/Debug
>     Start 1: usb
> Could not find executable /home/moot/Developpement/tools/stlink/build/Debug/tests/usb
> Looked in the following places:
> /home/*/Developpement/tools/stlink/build/Debug/tests/usb
> /home/*/Developpement/tools/stlink/build/Debug/tests/usb
> /home/*/Developpement/tools/stlink/build/Debug/tests/Release/usb
...
> Unable to find executable: /home/moot/Developpement/tools/stlink/build/Debug/tests/usb

etc..

i'm running current develop branch on debian testing.

chenguokai, 2020-04-14:
Reproducible on macOS.
The master branch is only broken with Test 3, while the develop branch is much more broken.

Nightwalker-87, 2020-04-14:
Thx for reporting. Interesting... I'll do some research with older commits to find out. No clue yet where it exactly happened and why.

chenguokai, 2020-04-14:
On the develop branch, the built test binaries resident in `build/Debug/bin`.
On the master branch, they are in `build/Debug/tests`.

Nightwalker-87, 2020-04-14:
To me it would actually make sense to place then in `build/Debug/bin/tests`.

Nightwalker-87, 2020-04-14:
Test 3 already fails on v1.3.1, so nobody seems to have taken any notice thereof...

This merge-commit 946b7f3b0419871eb8c334eccf60ee67980dea85 breaks Tests 1 and 2 on `develop` as well. Sry for that. :disappointed: 
I'll investigate the merged branch to find the related commit.

Nightwalker-87, 2020-04-16:
Can't go back any further than v1.3.1 as build fails on v1.3.0 and v1.2.0 used autotools for building.
Who can help finding and fixing this?

chenguokai, 2020-04-17:
Maybe I have figured out the breakage.

The first two breakages are caused by one commit between master and develop.

```
[ERROR] (-1) --debug --reset read /dev/sg0 test.bin 0x80000000 0x1000
[ERROR] (-1) --debug --reset write /dev/sg0 test.bin 0x80000000
```

The possible reason is the removal of actual usage of `devname` in struct `flash_opts`. In master branch, `src/tools/flash_opts.c` and `src/tools/flash.c` deal with `devname`. In the develop branch, they are removed. 

The last two breakages are caused by the change of serial argument layout in memory. The latest serial buffer begins with no leading zeros which the test expects.

[image]

With this patch, all tests passed on master branch.
[image]

Nightwalker-87, 2020-04-17:
Good job @chenguokai! I can verify that you are right with the first two failures and also tested the second issue successfully. So indeed this fixes that bug. I'll include the fix soon. Thank you.

commit aa38587e (2020-04-17): Restored functionality of make test builds

(Fixes #926)
closed 2020-04-17 by Nightwalker-87