stlink-org/stlink #928: [Ubuntu 14.04]:  #error unsupported libusb version but I installed the latest libusb version and libusb dev?

AveryChiu64, 2020-04-16:
- Operating system and version: [Linux (via Vagrant virtual machine)]
- **Stlink tools version** and/or git commit hash: [6f934a3] 
- Target chip (and board if applicable): [stm32f103c8t6] 

Commandline-Output:

In file included from /vagrant/stlink/include/stlink/sg.h:11:0,
                 from /vagrant/stlink/include/stlink.h:242,
                 from /vagrant/stlink/src/chipid.c:1:
/vagrant/stlink/include/stlink/stlinkusb.h:42:7: error: "LIBUSB_API_VERSION" is not defined [-Werror=undef]
 #if ( LIBUSB_API_VERSION < MINIMAL_API_VERSION )
       ^
/vagrant/stlink/include/stlink/stlinkusb.h:43:6: error: #error unsupported libusb version
     #error unsupported libusb version
      ^
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/stlink.dir/src/chipid.c.o] Error 1
make[2]: *** [CMakeFiles/stlink.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [release] Error 2
Expected/description:

Description
I am trying to run "make" so I can install it, but it keeps giving me this error and says the libusb version is unsupported. 
I ran sudo apt-get install libusb-1.0-0-dev and tried reinstalling it as well however I still keep getting this error
I have installed cmake, gcc, libusb, and libusb dev as instructed.

Thanks

Nightwalker-87, 2020-04-16:
Sorry, we dropped support for Ubuntu 14.04 and some other outdated systems without official support since Release v1.6.0. See `/doc/version_support.md` for more information on supported operating systems. If you can't upgrade you will have to stay on Release v1.6.0.

Nightwalker-87, 2020-04-16:
I've added a note to our README in commit c2c480846a493ae5493a5aecda41ad53f0b83bb2 which resolves this issue.

closed 2020-04-16 by Nightwalker-87