stlink-org/stlink #930: stlink: v 1.6.0 does not recognize "crashed" stlink .

rewolff, 2020-04-17:
when powering the target from the STLINK, low ESR decoupling caps on the target may crash the STLINK. I remember getting messages to this effect in the past, but just now I spent a couple of days failing to bring up a bunch of boards ("All bad") just because  st-flash reported "failed to start flashloader" instead of the usual: "your STLINK has crashed" (whatever the message was). 

I thought the newly populated boards were all flawed in some way but couldn't find any problem in the design.... Turns out that trying my tried-and-tested desgin resulted in the same problem. Sigh.

Nightwalker-87, 2020-04-17:
Does this appear in earlier versions as well?

rewolff, 2020-04-17:
I can't compile versions between 1.<veryold> and 1.6.0, so I can't check for you. Sorry.

Nightwalker-87, 2020-04-17:
You may try the 1.x.x-patch branches, which fix a compile error due to versioning, if you find some time and fancy in doing so.

Nightwalker-87, 2020-04-29:
@rewolff: I assume there will be no further activity here, if no additional testing and/or investigation takes place... We need at least a hint where to look at, if this ticket shall not go down the sink.

Nightwalker-87, 2020-06-03:
ping @rewolff

rewolff, 2020-06-03:
Hi, 
Sorry. 1) Can't find the time to work on this. 
2) It seems that after 1.6.0 but before 1.6.1 you changed something that I can no longer compile it on my workstation at work. :-( That system is ubuntu 16.04, due to 18.04 not having a reliable ARM-compiler. 20.04 is: "same as 18.04"....

Nightwalker-87, 2020-06-03:
> Hi,
> Sorry. 1) Can't find the time to work on this.
> 2) It seems that after 1.6.0 but before 1.6.1 you changed something that I can no longer compile it on my workstation at work. :-( That system is ubuntu 16.04, due to 18.04 not having a reliable ARM-compiler. 20.04 is: "same as 18.04"....

You may consider not to open topics that appear specific to the projects you work on, if you are not able to help out while others spent efforts to resolve these with only a minimal amount of support available. ;-) There are quite a few issues I found by now where the probability is very low that somebody else is able to step in, but which still remain with their original reason for being.

For v1.6.1 we did some testing on several systems. I can't tell what causes the problem you mention. However I propose you analyse it more precisely, evaluate if it could be of general interest and open a ticket with a tangible hint, if necessary.

When you need to rely on a toolset in a commercial environment, you may also give the official STM32 ST-LINK Utility a try (e.g. by running it in a VM).

Nightwalker-87, 2020-07-19:
Has anyone else experienced anything similar?
If able to help, please try to reproduce this with our `v1.5.1-patch` branch and v1.6.1 as well.

rewolff, 2020-07-19:
Hi Nightwalker, 

In open source projects, people's time and effort they put into your project should be valued, no matter how little it is. 

Simply reporting bugs is also a contribution that is valuable to the project. Demanding that people fix the bugs they find is not going to work in general. It pisses off the ones who can (me) and it pisses off the ones who cant. 

If you as a maintainer decide that "only REW has this problem, nobody else", feel free to create some "back-burner" category of bugs and put it there.

I prefer to have my projects be "easy-to-compile". i.e. work on a wide range of host systems. I've told you before that I could compile 1.3.1 and then nothing until 1.6.x . If in the period between those, I had tried to compile the latest version, I would've been stuck possibly having to go back to backups etc.  Since then, something in the build system has changed again, and now I'm unable to compile recent sources again. 

So it is kind of tricky for me to help: If I compile the latest source, perform a quick test and think it works-for-me, and install it, I could run into a showstopper later, to be stuck with having to figure out what previous version did compile and worked for all my use-cases. What if the compilation issue ends up being an evolution of my build-system, so I can't go back and compile the version that did work for me?

So the build-system issues are a concern for me that prevent me from "freely" working on the project. I can find a workaround for most problems. So as a workaround for THIS issue I programmed 50 boards the last week by putting the STLINK on a USB cable with a momentary switch. Have the script while(1) {program-it}  running and after hooking up the board with one hand, press and hold hte button with the other hand for the duration of the programming session. 

IMHO, there are several things that need to be improved structurally in the stlink package: 

1)  build system and build dependencies.  Test-compile on a wider range of systems before you accept a patch. When it doesn't compile on common systems you lose audience in both potential users and developers. 
2) Simplify the branches. One stable branch tagged with the latest release and one develop. 
3) error handling. Some errors are retried infinitely but they are of the class: Won't fix themselves. And others that WILL resolve themselves automatically are not retried. 

To solve 3, just sitting down for an afternoon, you can create some infrastructure that improves the error handling. Make it easy for the rest of the code to retry an operation X times or immediately declare defeat when something goes wrong or maybe issue a warning and hope things can work out with this one thing failing.

Nightwalker-87, 2020-07-19:
> Simply reporting bugs is also a contribution that is valuable to the project. Demanding that people fix the bugs they find is not going to work in general. It pisses off the ones who can (me) and it pisses off the ones who cant.

Reporting bugs is indeed helpful if people do query the issue database to look if the respective bug has been reported already.
It was never requested in general that bug reporters fix their own bugs, but I people indicate that they can contribute themselves then this suggests that the issue is dealt with. So if there is neither time nor the background to do so, then I would personally don't indicate there is.

> If you as a maintainer decide that "only REW has this problem, nobody else", feel free to create some "back-burner" category of bugs and put it there.

We had this before and it leads to issues not being looked after at all. This is why we should not go back to this as it counteracts any efforts to keep track of issues properly.

> So it is kind of tricky for me to help: If I compile the latest source, perform a quick test and think it works-for-me, and install it, I could run into a showstopper later, to be stuck with having [...]

That's why I asked if somebody else could do some testing (preferably in a testing environment). Testing should not occur in productive environments - that has never been put into question.

> IMHO, there are several things that need to be improved structurally in the stlink package:
>     1. build system and build dependencies.  Test-compile on a wider range of systems before you accept a [...]

That's **why** I recommended the use of the official ST-Link Utility in your case, because we can't offer or guarantee any of these three points in this project currently. This is because there are no regular and permanent contributors apart from project maintenance in this project. This was the case for many years, and still is the case since I stepped in to continue maintaining this project. I invested many hours and still spent lot of efforts to improve issue tracking, release management, documentation code structure, and package distribution. However my free time I can invest in this _as a hobby_ is limited somewhere. We don't have a commercial background in this project nor any external support.

So please consider this - there is no offence taken from my side, just this intended good advice for your needs.

Nightwalker-87, 2021-03-11:
This issue can likely not be reproduced by anyone else, as there is to little information provided on the used hardware and the specific application in a commercial environment. Against this background and the specific use case, the official STLINK-Utility from ST-Microelectronics should be used to ensure best results and extended commercial support.

closed 2021-03-11 by Nightwalker-87