stlink-org/stlink #936: win32: Wrong command line for 7z.exe in Findlibusb.cmake

slyshykO, 2020-04-19:
**Stlink tools version** 1.6.0-207-gcb3a2b6

Commandline-Output:

```
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.17/Modules/GNUInstallDirs.cmake:225 (message):
  Unable to determine default CMAKE_INSTALL_LIBDIR directory because no
  target architecture is known.  Please enable at least one language before
  including GNUInstallDirs.
Call Stack (most recent call first):
  CMakeLists.txt:4 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 8.1.0
-- Check for working C compiler: E:/test-st/i686-8.1.0-release-posix-sjlj-rt_v6-rev0/mingw32/bin/gcc.exe
-- Check for working C compiler: E:/test-st/i686-8.1.0-release-posix-sjlj-rt_v6-rev0/mingw32/bin/gcc.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.26.1.windows.1") 
-- stlink version: 1.6.0-207-gcb3a2b6
-- Major 1 Minor 6 Patch 0
-- Performing Test C_SUPPORTS_STD_GNU11
-- Performing Test C_SUPPORTS_STD_GNU11 - Success
-- Performing Test C_SUPPORTS_WALL
-- Performing Test C_SUPPORTS_WALL - Success
-- Performing Test C_SUPPORTS_WEXTRA
-- Performing Test C_SUPPORTS_WEXTRA - Success
-- Performing Test C_SUPPORTS_WSHADOW
-- Performing Test C_SUPPORTS_WSHADOW - Success
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2 - Success
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING - Success
-- Performing Test C_SUPPORTS_WUNDEF
-- Performing Test C_SUPPORTS_WUNDEF - Success
-- Performing Test C_SUPPORTS_WFORMAT
-- Performing Test C_SUPPORTS_WFORMAT - Success
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY - Success
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED - Success
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS - Failed
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32 - Failed
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION - Success
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS - Success
-- Performing Test C_SUPPORTS_O2
-- Performing Test C_SUPPORTS_O2 - Success
-- Performing Test C_SUPPORTS_WERROR
-- Performing Test C_SUPPORTS_WERROR - Success
-- downloading libusb 1.0.23

Command Line Error:
Too short switch:
-o
-- Missing libusb library has been installed
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (Libusb)
  does not match the name of the calling package (libusb).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/Findlibusb.cmake:134 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:56 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find Libusb (missing: LIBUSB_LIBRARY LIBUSB_INCLUDE_DIR) 
-- Looking for sys/mman.h
-- Looking for sys/mman.h - not found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for __stack_chk_fail in ssp
-- Looking for __stack_chk_fail in ssp - found
-- STLINK_LIB_SHARED: stlink-shared
-- PROJECT_VERSION_MAJOR: 1
-- VERSION: 1.6.0
-- GTK3 not found!
-- Manpage generation disabled
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory E:/test-st/stlink-git
   used as include directory in directory E:/test-st/stlink-git/src/gdbserver
   used as include directory in directory E:/test-st/stlink-git/src/stlink-gui
   used as include directory in directory E:/test-st/stlink-git/debian/pkg-config
   used as include directory in directory E:/test-st/stlink-git/include
   used as include directory in directory E:/test-st/stlink-git/doc/man
   used as include directory in directory E:/test-st/stlink-git/tests
LIBUSB_LIBRARY (ADVANCED)
    linked by target "stlink-shared" in directory E:/test-st/stlink-git
    linked by target "stlink-shared" in directory E:/test-st/stlink-git
    linked by target "stlink-static" in directory E:/test-st/stlink-git

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "stlink-shared":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "stlink-shared":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "stlink-static":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "stlink-static":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of "st-flash":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of "st-flash":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of "st-info":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of "st-info":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in src/gdbserver/CMakeLists.txt:
  Found relative path while evaluating include directories of "st-util":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in src/gdbserver/CMakeLists.txt:
  Found relative path while evaluating include directories of "st-util":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in tests/CMakeLists.txt:
  Found relative path while evaluating include directories of "flash":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in tests/CMakeLists.txt:
  Found relative path while evaluating include directories of "flash":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in tests/CMakeLists.txt:
  Found relative path while evaluating include directories of "usb":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in tests/CMakeLists.txt:
  Found relative path while evaluating include directories of "usb":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in tests/CMakeLists.txt:
  Found relative path while evaluating include directories of "sg":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

CMake Error in tests/CMakeLists.txt:
  Found relative path while evaluating include directories of "sg":

    "LIBUSB_INCLUDE_DIR-NOTFOUND"

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
Scanning dependencies of target stlink-shared
[  2%] Building C object CMakeFiles/stlink-shared.dir/src/chipid.c.obj
E:\test-st\stlink-git\src\chipid.c:1:10: fatal error: stlink.h: No such file or directory
 #include "stlink.h"
          ^~~~~~~~~~
compilation terminated.
mingw32-make.exe[2]: *** [CMakeFiles\stlink-shared.dir\build.make:82: CMakeFiles/stlink-shared.dir/src/chipid.c.obj] Error 1
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:231: CMakeFiles/stlink-shared.dir/all] Error 2
mingw32-make.exe: *** [Makefile:171: all] Error 2
[  2%] Building C object CMakeFiles/stlink-shared.dir/src/chipid.c.obj
E:\test-st\stlink-git\src\chipid.c:1:10: fatal error: stlink.h: No such file or directory
 #include "stlink.h"
          ^~~~~~~~~~
compilation terminated.
mingw32-make.exe[2]: *** [CMakeFiles\stlink-shared.dir\build.make:82: CMakeFiles/stlink-shared.dir/src/chipid.c.obj] Error 1
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:231: CMakeFiles/stlink-shared.dir/all] Error 2
mingw32-make.exe: *** [Makefile:171: all] Error 2

```

Expected/description:

Can't build project on windows cause downloaded libusb archive didn't unarchive properly.

Nightwalker-87, 2020-04-20:
I could reproduce on windows, and found what causes this.
This is as small issue only. Fix will be pushed soon.

slyshykO, 2020-04-20:
The problem still exists.

Nightwalker-87, 2020-04-20:
Additional compilation error:

```
Command Line Error:
Too short switch:
-o
```

which derives from 7zip archive extraction, has been fixed in commit 7640daae4c0ade48a4179b3a7750272471b8c8e5.

Nightwalker-87, 2020-04-21:
@slyshykO: Please try the temporary `test_a` branch to verify if everything is ok with it on Windows now. Don't stay with it tough and move back to develop after testing this, as it is going to be deleted as soon as develop is proved to be working again.

slyshykO, 2020-04-21:
In the branch _test_a_ build goes smoothly, without issues.

commit 629314c4 (2020-04-20): Corrections in cmake build

- Name pattern for libusb detection
- Include of GNUInstallDirs module
- Removed old ci-leftover in Makefile

(Fixes #936)
commit b74bb1de (2020-04-21): Fixed compilation errors on windows

- Include of GNUInstallDirs module
- Removed old ci-leftover in Makefile
- Name pattern for libusb detection

Extraction of downloaded libusb archive failed due to wrong command 
option formatting since commit a8c1f41f1b1c9dd1314f99a1d42de1008f5a4f74.
(Fixes #936)
commit 17f4e6a8 (2020-04-21): Fixed compilation errors on windows

- Include of GNUInstallDirs module
- Removed old ci-leftover in Makefile
- Fixed name pattern for libusb detection

Extraction of downloaded libusb archive failed due to wrong command 
option formatting since commit a8c1f41f1b1c9dd1314f99a1d42de1008f5a4f74.
(Fixes #936)
commit 9b19f922 (2020-04-21): Fixed compilation errors

- Moved include of GNUInstallDirs module in cmake-routine
- Removed old ci-leftover in Makefile
- cmake: Fixed name pattern for libusb module (Fixes #940)
- Wrong command option formatting for 7zip (Fixes #936)

Extraction of downloaded libusb archive on Windows failed since commit 
a8c1f41f1b1c9dd1314f99a1d42de1008f5a4f74.
closed 2020-04-21 by Nightwalker-87