stlink-org/stlink #937: Error building from develop branch, cpuid problems

Vascom, 2020-04-20:
I try build commit c7874f805c63c60285f2b190f6589da8ae5e47fa
And it failed with this errors:
```
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c: In function 'main':
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c:34:9: error: 'cpuid.revision' may be used uninitialized in this func
tion [-Werror=maybe-uninitialized]
   34 |         printf("cpuid:part = %#x, rev = %#x\n", cpuid.part, cpuid.revision);
      |         ^
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c:31:27: note: 'cpuid.revision' was declared here
   31 |         cortex_m3_cpuid_t cpuid;
      |                           ^
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c:34:9: error: 'cpuid.part' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   34 |         printf("cpuid:part = %#x, rev = %#x\n", cpuid.part, cpuid.revision);
      |         ^
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c:31:27: note: 'cpuid.part' was declared here
   31 |         cortex_m3_cpuid_t cpuid;
      |                           ^
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c:33:9: error: 'cpuid.variant' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   33 |         printf("cpuid:impl_id = %0#x, variant = %#x\n", cpuid.implementer_id, cpuid.variant);
      |         ^
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c:31:27: note: 'cpuid.variant' was declared here
   31 |         cortex_m3_cpuid_t cpuid;
      |                           ^
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c:33:9: error: 'cpuid.implementer_id' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   33 |         printf("cpuid:impl_id = %0#x, variant = %#x\n", cpuid.implementer_id, cpuid.variant);
      |         ^
/builddir/build/BUILD/stlink-org-stlink-c7874f8/tests/usb.c:31:27: note: 'cpuid.implementer_id' was declared here
   31 |         cortex_m3_cpuid_t cpuid;
      |                           ^
lto1: all warnings being treated as errors
```
How it can be fixed?

chenguokai, 2020-04-20:
On macOS, the code builds well. 

In `src/common.c`
```
int stlink_cpu_id(stlink_t *sl, cortex_m3_cpuid_t *cpuid) {
    uint32_t raw;

    if (stlink_read_debug32(sl, STLINK_REG_CM3_CPUID, &raw))
        return -1;

    cpuid->implementer_id = (raw >> 24) & 0x7f;
    cpuid->variant = (raw >> 20) & 0xf;
    cpuid->part = (raw >> 4) & 0xfff;
    cpuid->revision = raw & 0xf;
    return 0;
}
```

It seems that cpuid.part and cpuid.revision is indeed initialized here. Maybe a compiler misbehave?

Vascom, 2020-04-20:
```
    if (stlink_read_debug32(sl, STLINK_REG_CM3_CPUID, &raw))
        return -1;
```
In this case cpuid.* stay uninitialized.

Ant-ON, 2020-04-20:
On Ubuntu successful compiled too. Variable cpuid may use uninited in the case of error read register.

`--- a/tests/usb.c
+++ b/tests/usb.c
@@ -29,7 +29,11 @@ int main(int ac, char** av) {
         printf("-- core_id: %#x\n", sl->core_id);
 
         cortex_m3_cpuid_t cpuid;
-        stlink_cpu_id(sl, &cpuid);
+        if (stlink_cpu_id(sl, &cpuid))
+        {
+            printf("unknown chip\n");
+            memset(&cpuid, 0, sizeof(cortex_m3_cpuid_t))
+        }
         printf("cpuid:impl_id = %0#x, variant = %#x\n", cpuid.implementer_id, cpuid.variant);
         printf("cpuid:part = %#x, rev = %#x\n", cpuid.part, cpuid.revision);
`

Vascom, 2020-04-20:
In cmake/c_flags.cmake present
`add_cflag_if_supported("-Wundef")`
so it should be fixed.

chenguokai, 2020-04-20:
I think it can be fixed by adding a return value check on stlink_cpu_id and initialize those values to zero. I will give it a try.

Vascom, 2020-04-20:
I have `-Werror=maybe-uninitialized` because during configure:
```
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED - Success
```

chenguokai, 2020-04-20:
@Vascom May you test my branch https://github.com/chenguokai/stlink/tree/issue937 ?
If it works, I will raise a PR.

Vascom, 2020-04-20:
> @Vascom May you test my branch https://github.com/chenguokai/stlink/tree/issue937 ?
> If it works, I will raise a PR.

Yes, it works.

closed 2020-04-20 by Nightwalker-87