stlink-org/stlink #940: libusb not found while build for Fedora

Vascom, 2020-04-20:
Try build commit 4445e1c715c4f6358f8f4c7b2859417e6eff832d and have this error
```
-- Found _libusb: /usr/lib64/libusb-1.0.so  
-- Configuring incomplete, errors occurred!
See also "/builddir/build/BUILD/stlink-org-stlink-4445e1c/build/CMakeFiles/CMakeOutput.log".
See also "/builddir/build/BUILD/stlink-org-stlink-4445e1c/build/CMakeFiles/CMakeError.log".
CMake Error at cmake/modules/Find_libusb.cmake:152 (message):
  libusb library not found on your system! Install libusb 1.0.x from your
  package repository.
Call Stack (most recent call first):
  CMakeLists.txt:68 (find_package)
```
Before commit fb9f7770912140dc262d3dedc9c5e570295483f1 build was OK.

Nightwalker-87, 2020-04-20:
The second commit did not change anything related to libusb detection, it solely addressed packaging. Please test again.

Vascom, 2020-04-20:
Tested. The same error.

Nightwalker-87, 2020-04-20:
As far as I can see from fb9f7770912140dc262d3dedc9c5e570295483f1, there are no changes related to the part of your compilation that match. It must be a commit where there is a change in `Find_libusb.cmake`. Please test older commits going backwards. I don't have the possibility to test on fedora, so you need to find the exact commit where it first appeared. Otherwise I can hardly help any further.

Vascom, 2020-04-20:
Error came from this commit 629314c44f801f82d478fe49e7b409cdb7364077

Nightwalker-87, 2020-04-20:
ok, travis build reveals it now, I think.
Please give me some time and don't report any new issues until this is fixed.

Vascom, 2020-04-20:
OK.
Next reports will be after this fixed :)

Nightwalker-87, 2020-04-21:
@Vascom: Please try the temporary `test_a` branch to verify if everything is ok with it on Fedora now. Don't stay with it tough and move back to `develop` after testing this, as it is going to be deleted again as soon as develop is proved to be working again.

Vascom, 2020-04-21:
Tested. Problem solved.

commit 9b19f922 (2020-04-21): Fixed compilation errors

- Moved include of GNUInstallDirs module in cmake-routine
- Removed old ci-leftover in Makefile
- cmake: Fixed name pattern for libusb module (Fixes #940)
- Wrong command option formatting for 7zip (Fixes #936)

Extraction of downloaded libusb archive on Windows failed since commit 
a8c1f41f1b1c9dd1314f99a1d42de1008f5a4f74.
closed 2020-04-21 by Nightwalker-87