stlink-org/stlink #944: stlink.pc not installed

Vascom, 2020-04-21:
Time for next bugreport :)

Now stlink.pc file not installed at all.

It should be installed to /usr/lib64/pkgconfig/
lib64 - current libdir.

Nightwalker-87, 2020-04-21:
Possible, as package configuration is rewritten currently.
The whole procedure is refactored, so don't bother about it too much yet.
I just did not open a separate ticket for it as I am working on it myself...
Please stay with c7874f805c63c60285f2b190f6589da8ae5e47fa if you need a working old configuration.

Nightwalker-87, 2020-04-23:
@Vascom: Can you test the current `pkg` branch now and report back?

Vascom, 2020-04-23:
What commit need to test?

Vascom, 2020-04-23:
stlink.pc not installed at all.

Nightwalker-87, 2020-04-23:
Don't expect it to work on the `develop` branch, changes were only made on the latest `pkg` branch.
Should it not work there, then I have no clue, sry.

Nightwalker-87, 2020-04-23:
Something is not right at your end, at least. The version string reads 1.6.1 which is not present yet. I don't know where your build routine got this from. It should still be 1.6.0. The former has not been defined anywhere in the code yet.

Vascom, 2020-04-23:
Do you mean `stlink-1.6.1-0.1.20200420git78dd836`?
It is version in my spec file. It is not affect build.

Nightwalker-87, 2020-04-23:
I know it does not affect the build, but I knew it was not from the provided settings. However I can't figure out the problem right now. So I suggest you stay on the old version for know or look into the details yourself. I never had any error on this so far on my system. (I doubt though these settings are even used on debian-based systems. If so, then my build system has no problem with it.)

Vascom, 2020-04-23:
To be as in 1.6.0 you should
1. Copy https://raw.githubusercontent.com/stlink-org/stlink/v1.6.0/usr/lib/pkgconfig/CMakeLists.txt to cmake/pkgconfig/ and replace existing CMakeLists.txt.
2. Copy https://raw.githubusercontent.com/stlink-org/stlink/v1.6.0/usr/lib/pkgconfig/pkg-config.pc.cmake to cmake/pkgconfig/ and remove pkgconfig.pc.cmake.
3. Add `add_subdirectory(cmake/pkgconfig)` to main CMakeLists.txt.

With this changing all work.

Vascom, 2020-04-23:
Looks good and work.

But may be at x86_64 this
`libdir=${prefix}/lib/${deb_host_multiarch}`
must be
`libdir=${prefix}/lib64/${deb_host_multiarch}`

I am not sure.

Nightwalker-87, 2020-04-23:
No, because the tools still build a 32-bit library, also for 64-bit systems that's why they should install to `libdir=${prefix}/lib/${deb_host_multiarch}`.

Vascom, 2020-04-23:
I built for 64-bit. And I look at others .pc files in my system - it has lib64 in .pc files.
So it is important.

Nightwalker-87, 2020-04-23:
I'm not really willing to currently, because the library is still coded 32-bit, even if you build for 64-bit, as the `st-link ` code is not written for 64-bit. This is likely the reason why `lib` is not changed to `lib64`.

commit 3e3f782e (2020-04-23): Updated pkgconfig

(Closes #944)
closed 2020-04-26 by Nightwalker-87