stlink-org/stlink #946: [Refactoring] Cleanup of cmake build process

Nightwalker-87, 2020-04-25:
I've been actively working on a major cleanup of our cmake build routine.
Related changes will go into the `pkg` branch.
We should see some testing and verification as soon as a PR is opened.

Nightwalker-87, 2020-04-25:
@chenguokai: It looks as if it is necessary to figure out the correct destination for the GUI to install to. As far as I have found out, the file `gui.ui` and the `stlink-gui` executable are not installed to the same output directory what should be the case IMHO (otherwise one would have to switch to the directory containing the ui-file in order for the executable to launch, which is not helpful at all). Also we have a `stlink-gui-local` and a `stlink-gui` executable. We need to define what should go where upon install.

Nightwalker-87, 2020-04-25:
Here is my proposal for the updated cmake GUI build: https://pastebin.com/WmH2Uek6

chenguokai, 2020-04-25:
I am not so familiar with cmake and gtk. Sorry that I cannot provide any comments on the proposal.
One thing that I am sure is, there is no need to tweak those paths specially for macOS. Once I thought the GUI version would be compiled into an .app bundle on macOS that is expected to resident in /Applications. Fortunately it is not, so if Linux and BSDs find one path good, so will macOS do.

Nightwalker-87, 2020-04-25:
That's ok, I just have the impression that something was wrong there before I even started to clean things up. Maybe someone else can provide some feedback on this.

Apart from a few minor open questions I seem to be trough with this now. With respect to current coding activities around the `stlink-library`, `st-info` and `st-flash` I have not completely transitioned the file structure yet. Thus the rest will take place in a second step later on.

Nightwalker-87, 2020-04-25:
I have made up my mind after some local testing:
Up on `sudo make install` the executable `stlink-gui` will be installed to `/usr/local/bin` together with it's `stlink-gui.ui` file and the other binaries of the toolset (`st-flash`, `st-info` and `st-util`). This ensures that both files for the gui launch are always together and solves the problem mentioned before as the GUI now becomes directly accessible without the requirement for any additional steps or settings.

closed 2020-04-26 by Nightwalker-87