stlink-org/stlink #950: Written consent request

techtoys, 2020-04-30:
Dear Sir

My company is developing a board (commercial) with stlink embedded. I notice the clause to prohibit others from using the name of the project or its contributors to promote derived products without written consent.
May I learn if there is any charge or qualification that allow me to use stlink on my product?
Look forward to hearing from you soon.

John

slyshykO, 2020-04-30:
It is my personal opinion, but you can do whatever you want that allowed you by project license.
https://github.com/stlink-org/stlink#license

chenguokai, 2020-04-30:
There could be one misunderstanding here.

Below is my understanding.

ST-Link, which refers to the hardware developed by ST, charges as a product.

stlink, which refers to this program, is licensed under BSD license, that is, as long as you get one copy of this software, you are free (with RMS’s word, libre) to use, modify and redistribute. The clause restricts promotions like “This board provides user with a good experience with stlink developed by SOMENAMES”

Using this tool to develop is not a promotion in my view.

Nightwalker-87, 2020-04-30:
OK, let me, as one of the project maintainers, have a word on this as well:

> May I learn if there is any charge or qualification that allow me to use stlink on my product?

You may use this software toolset as long as you fully comply to the BSD-3 license conditions. Thus there is no charge on you or special qualification that has to be fulfilled to do so.

> I notice the clause to prohibit others from using the name of the project or its contributors to promote derived products without written consent.

As it appears impossible to gain consent from every single contributor of this software (which would be the only sufficient condition for the above according to the license), we must assume that you are _not allowed to promote your projects in relation to the name of this software_ which is `stlink` or any of its contributors. As @chenguokai  mentioned before, this is in contrast to `ST-Link` which refers to the programmers (hardware products) distributed by STMicroelectronics.

The background to this is that the `stlink` project has no legal relation to your company and thus the latter should not actively advertise work which is community-owned property.

techtoys, 2020-04-30:
@Nightwalker-87 we must assume that you are not allowed to **promote your projects** in relation to the name of this software which is `stlink` 

The keyword is **promote your projects**. Suppose I have another open-source project on GitHub with a board integrates by following the ST-Link hardware schematic from any of the Nucleo board combining with stlink firmware. Later on my GitHub I state the firmware source by directly linking to stlink-org; is the action regarded as **promote my project**?

Nightwalker-87, 2020-04-30:
> _"we must assume that you are not allowed to **promote your projects** in relation to the name of this software which is `stlink`"_
>
> The keyword is **promote your projects**. Suppose I have another [...]

I would not consider a direct reference to the `stlink` project or it's parent `stlink-org` (e.g. within _your_ project documentation) as _promotion_. To me, promotion would be linked to advertising in the sense of what @chenguokai stated as an example.

What I would not like to see is a reference related to any form of (customer) support for _your_ hardware products, and potential issues with these, as we only provide support for the `stlink` toolset itself (software-related). _If you start to sell commercial products, you will be responsible to provide support for these by yourself._

All stuff that carries the "ST"-prefix is not our business, for these names you need to consult the company STMicroelectronics to ensure that you comply with their legal provisions regarding product naming, brandings and licensing on a commercial basis.

techtoys, 2020-05-01:
Thank you. Everything is clear.

closed 2020-05-01 by techtoys