stlink-org/stlink #952: LIBUSB_ERROR_TIMEOUT after firmware upgrade (STM32F769NI-DISCO)

tdjastrzebski, 2020-05-02:
After upgrading board firmware V2.J29.M18 -> V2.J36.M26 STLink no longer works.
I get errors like:

```
2020-05-02T09:36:50 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002014
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_WRITEDEBUG_32BIT
2020-05-02T09:36:53 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002018
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_WRITEDEBUG_32BIT
2020-05-02T09:36:56 DEBUG common.c: *** stlink_write_debug32 0 to 0xe000201c
```

I have tried upgrade on two different boards with consistent result - upgrade breaks something.  
The same time openOCD works fine.  
Interestingly, if I use openOCD firs and then STLink without resetting the board, it works.

- Board type: STM32F769NI-DISCO
- Operating system: Windows 10
- Stlink tools version: 1.6.0 (released)
- Environment: VSCode + cortex-debug

Nightwalker-87, 2020-06-12:
@tdjastrzebski: Can you try with Firmware V2.J37.M26 as well with v1.6.0?

tdjastrzebski, 2020-06-20:
@Nightwalker-87 It seems that update to V2.J37.M26 solved the problem. Thanks.

Nightwalker-87, 2020-06-20:
@tdjastrzebski: Oh, that's good news for us. Thx for the feedback!

closed 2020-06-20 by Nightwalker-87