stlink-org/stlink #956: [Your device name]: [Title]

busier, 2020-05-11:
Thank you for giving feedback to the stlink project.

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: [enter here] (e.g Stlink /v1, /v2, /v2-clone, /v2-1)
- Programmer firmware version: [enter here] (e.g STSW-LINK007 2.27.15)
- Operating system and version: [enter here] (e.g Linux, Mac OS X, Windows)
- **Stlink tools version** and/or git commit hash: [enter here] (e.g v1.1.0/git-c722056)
- Stlink commandline tool name: [enter here] (e.g `st-info`, `st-flash`, `st-util`)
- Target chip (and board if applicable): [enter here] (e.g STM32F402VG)

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:

`short description of the expected value`

**NOTICE: This bug report may be closed without further notice, if not enough information is provided!**

Thank you for your support.

The stlink project maintainers

busier, 2020-05-11:
Hi,
Unfortunately, STlink Tools not working after [firmware upgrade](https://www.st.com/en/development-tools/stsw-link007.html#get-software) for board ST-LINK/V2.

```
st-flash write foo.bin 0x8000000 
st-flash 1.6.0-257-g38a5c14
2020-05-11T22:25:30 WARN common.c: unknown chip id! 0x5fa0004
Failed to connect to target
```
[](url)

```
$ st-util --verbose=99
st-util
2020-05-11T22:23:46 DEBUG common.c: *** looking up stlink version
2020-05-11T22:23:46 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-05-11T22:23:46 DEBUG common.c: stlink pid     = 0x3748
2020-05-11T22:23:46 DEBUG common.c: stlink version = 0x2
2020-05-11T22:23:46 DEBUG common.c: jtag version   = 0x25
2020-05-11T22:23:46 DEBUG common.c: swim version   = 0x7
2020-05-11T22:23:46 DEBUG common.c: stlink current mode: mass
2020-05-11T22:23:46 DEBUG common.c: *** set_swdclk ***
2020-05-11T22:23:46 DEBUG common.c: stlink current mode: mass
2020-05-11T22:23:46 DEBUG common.c: *** stlink_enter_swd_mode ***
2020-05-11T22:23:46 DEBUG common.c: *** stlink_jtag_reset ***
2020-05-11T22:23:46 DEBUG common.c: *** stlink_reset ***
2020-05-11T22:23:46 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-05-11T22:23:46 DEBUG common.c: Loading device parameters....
2020-05-11T22:23:46 DEBUG common.c: *** stlink_core_id ***
2020-05-11T22:23:46 DEBUG common.c: core_id = 0x2ba01477
2020-05-11T22:23:46 DEBUG common.c: *** stlink_read_debug32 5fa0004 is 0xe0042000
2020-05-11T22:23:46 WARN common.c: unknown chip id! 0x5fa0004
2020-05-11T22:23:46 DEBUG common.c: *** stlink_reset ***
2020-05-11T22:23:46 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-05-11T22:23:46 DEBUG gdb-server.c: Chip ID is 0x00000004, Core ID is 0x2ba01477.
2020-05-11T22:23:46 INFO gdb-server.c: Listening at *:4242...
^C2020-05-11T22:24:38 DEBUG common.c: *** stlink_run ***
2020-05-11T22:24:38 DEBUG common.c: *** stlink_exit_debug_mode ***
2020-05-11T22:24:38 DEBUG common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2020-05-11T22:24:38 DEBUG common.c: *** stlink_close ***
```

grzegorz-kraszewski, 2020-05-18:
I have encountered the same problem after updating ST-Link firmware to 2.37.26 in Nucleo-64 STM32F030R8 board.

chenguokai, 2020-05-18:
So may we add a comment like "DO NOT UPDATE STLINK FIRMWARE TO THE LATEST VERSION IF YOU WANT TO USE THIS SOFTWARE"?

Nightwalker-87, 2020-05-18:
To me it would be more interesting to know why this actually fails...
Maybe a patch is possible for the tools.
The proposed note should only be necessary if this is not possible.

grzegorz-kraszewski, 2020-05-18:
> DO NOT UPDATE STLINK FIRMWARE

For me it is too late... I have downgraded the firmware to the oldest version available on st.com, but it hasn't helped. As a workaround, I've managed to flash my code using bootloader, USART and `stm32flash` tool, so I know the target device is working.

busier, 2020-05-18:
> So may we add a comment like "DO NOT UPDATE STLINK FIRMWARE TO THE LATEST VERSION IF YOU WANT TO USE THIS SOFTWARE"?

I think: YES.
I was downgrade my st-link to stsw-link007_V2-36-26.

chenguokai, 2020-05-18:
Maybe this link can provide some help

https://community.st.com/s/question/0D50X00009XkhxW/downgrade-stlinkv2-firmware-revision

grzegorz-kraszewski, 2020-05-20:
> Could anyone figure out which version is the last known good one?

Unfortunately these old firmwares have only Windows updater. And, as one may guess, this updater does not work under wine.

chenguokai, 2020-05-20:
> Unfortunately these old firmwares have only Windows updater. And, as one may guess, this updater does not work under wine.

If I remember, I upgraded my stlink v2 clone with a java edition of the updater on macOS. The device functions well after the upgrade. I will check tomorrow if I still have that firmware on my computer.

Nightwalker-87, 2020-06-03:
I'll upgrade my STLINKv2/clone to the latest firmware and report if I run into any issues.

Nightwalker-87, 2020-06-03:
I have tested the original approach with the official firmware versions V2.J29.S7 and V2.J34.S7 and Release v1.6.1:

```
$USER@$DEVICE:/$LOCAL_PATH/Blinker/Release$ st-flash write Blinker.bin 0x8000000
st-flash 1.6.1
2020-06-03T14:19:16 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file Blinker.bin md5 checksum: 16277ff2c027d497361dbae4255081a1, stlink checksum: 0x0004644a
2020-06-03T14:19:16 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
2020-06-03T14:19:16 INFO common.c: Flash page at addr: 0x08000000 erased
2020-06-03T14:19:16 INFO common.c: Flash page at addr: 0x08000400 erased
2020-06-03T14:19:16 INFO common.c: Flash page at addr: 0x08000800 erased
2020-06-03T14:19:16 INFO common.c: Flash page at addr: 0x08000c00 erased
2020-06-03T14:19:16 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2020-06-03T14:19:16 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-06-03T14:19:16 INFO flash_loader.c: Successfully loaded flash loader in sram
  4/4 pages written
2020-06-03T14:19:16 INFO common.c: Starting verification of write complete
2020-06-03T14:19:16 INFO common.c: Flash written and verified! jolly good!

$USER@$DEVICE:/$LOCAL_PATH/git/stlink$ st-util --verbose=99
st-util
2020-06-03T13:13:44 DEBUG common.c: *** looking up stlink version
2020-06-03T13:13:44 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-06-03T13:13:44 DEBUG common.c: stlink pid     = 0x3748
2020-06-03T13:13:44 DEBUG common.c: stlink version = 0x2
2020-06-03T13:13:44 DEBUG common.c: jtag version   = 0x22
2020-06-03T13:13:44 DEBUG common.c: swim version   = 0x7
2020-06-03T13:13:44 DEBUG common.c: *** looking up stlink version
2020-06-03T13:13:44 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-06-03T13:13:44 DEBUG common.c: stlink pid     = 0x3748
2020-06-03T13:13:44 DEBUG common.c: stlink version = 0x2
2020-06-03T13:13:44 DEBUG common.c: jtag version   = 0x22
2020-06-03T13:13:44 DEBUG common.c: swim version   = 0x7
2020-06-03T13:13:44 DEBUG common.c: stlink current mode: mass
2020-06-03T13:13:44 DEBUG usb.c: JTAG/SWD freq set to 0
2020-06-03T13:13:44 DEBUG common.c: *** set_swdclk ***
2020-06-03T13:13:44 DEBUG common.c: stlink current mode: mass
2020-06-03T13:13:44 DEBUG common.c: *** stlink_enter_swd_mode ***
2020-06-03T13:13:44 DEBUG common.c: *** stlink_jtag_reset ***
2020-06-03T13:13:44 DEBUG common.c: *** stlink_reset ***
2020-06-03T13:13:44 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-06-03T13:13:44 DEBUG common.c: Loading device parameters....
2020-06-03T13:13:44 DEBUG common.c: *** stlink_core_id ***
2020-06-03T13:13:44 DEBUG common.c: core_id = 0x2ba01477
2020-06-03T13:13:44 DEBUG common.c: *** stlink_read_debug32 20036410 is 0xe0042000
2020-06-03T13:13:44 DEBUG common.c: *** stlink_read_debug32 ffff0040 is 0x1ffff7e0
2020-06-03T13:13:44 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2020-06-03T13:13:44 DEBUG common.c: *** stlink_reset ***
2020-06-03T13:13:44 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-06-03T13:13:44 DEBUG gdb-server.c: Chip ID is 0x00000410, Core ID is 0x2ba01477.
2020-06-03T13:13:44 INFO gdb-server.c: Listening at *:4242...
2020-06-03T14:07:48 DEBUG common.c: *** stlink_force_debug_mode ***
2020-06-03T14:07:48 DEBUG common.c: *** stlink_reset ***
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 3 to 0xe0002000
2020-06-03T14:07:48 DEBUG common.c: *** stlink_read_debug32 261 is 0xe0002000
2020-06-03T14:07:48 INFO gdb-server.c: Found 6 hw breakpoint registers
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002008
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe000200c
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002010
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002014
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002018
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe000201c
2020-06-03T14:07:48 DEBUG gdb-server.c: init watchpoints
2020-06-03T14:07:48 DEBUG common.c: *** stlink_read_debug32 0 is 0xe000edfc
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 1000000 to 0xe000edfc
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0001028
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0001038
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0001048
2020-06-03T14:07:48 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0001058
[... 17 lines cut ...]
2020-06-03T14:07:48 DEBUG gdb-server.c: send: 
2020-06-03T14:07:48 DEBUG gdb-server.c: recv: qL1200000000000000000
2020-06-03T14:07:48 DEBUG gdb-server.c: send: 
2020-06-03T14:07:48 DEBUG gdb-server.c: recv: Hc-1
2020-06-03T14:07:48 DEBUG gdb-server.c: send: 
2020-06-03T14:07:48 DEBUG gdb-server.c: recv: qC
2020-06-03T14:07:48 DEBUG gdb-server.c: send: 
2020-06-03T14:07:48 DEBUG gdb-server.c: recv: qAttached
2020-06-03T14:07:48 DEBUG gdb-server.c: query: Attached;
2020-06-03T14:07:48 DEBUG gdb-server.c: send: 
2020-06-03T14:07:48 DEBUG gdb-server.c: recv: g
2020-06-03T14:07:48 DEBUG common.c: *** stlink_read_all_regs ***
[... cut ...]
2020-06-03T14:07:48 DEBUG gdb-server.c: recv: qXfer:memory-map:read::27c,1000
2020-06-03T14:07:48 DEBUG gdb-server.c: query: Xfer;memory-map:read::27c,1000
2020-06-03T14:07:48 DEBUG gdb-server.c: Xfer: type:memory-map;op:read;annex:;addr:636;length:4096
2020-06-03T14:07:48 DEBUG gdb-server.c: send: l
```
I'll do a second test with the previous release for investigation. However, we can already confirm that there is no general problem with current STLINKv2 firmware versions, even if used with cloned programming devices.

Nightwalker-87, 2020-06-03:
>     * Programmer firmware version: stsw-link007_V2-37-26

@busier: This is the version of the upgrade tool, not the firmware version of the programmer. The latter is determined by the upgrade tool and the latest available version is displayed afterwards. Please retry with your hardware and only use official tools and versions from ST.

Nightwalker-87, 2020-06-03:
> **UPD. Jun 3 2020**. 2@**nightwalker-87**.
> I answer here, because the thread closed 4 comments.
> Last update is Programmer firmware version: V2J37S7.
> I did downgrade firmware V2J36S7 for to work st-flash.
> Your versions FW V2.J29.S7 and V2.J34.S7 are too old.

The thread is open again. These are the latest versions which are offered for a STLINK-v2 & and STLINK-v2 clone programmer here via ST. I personally can not test on any other hardware, sry.
If there is no further verification and contribution then we can't do anything about it I'm afraid.

Nightwalker-87, 2020-06-03:
Another test with Release v1.6.1 and Firmware V2.J36.S7 (tool-updated to 2.5.3):

```
$USER@$DEVICE:/$LOCAL_PATH:~/Schreibtisch/Blinker/Release$ st-flash write Blinker.bin 0x8000000
st-flash 1.6.1
2020-06-03T17:14:50 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file Blinker.bin md5 checksum: 16277ff2c027d497361dbae4255081a1, stlink checksum: 0x0004644a
2020-06-03T17:14:50 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
2020-06-03T17:14:50 INFO common.c: Flash page at addr: 0x08000000 erased
2020-06-03T17:14:50 INFO common.c: Flash page at addr: 0x08000400 erased
2020-06-03T17:14:50 INFO common.c: Flash page at addr: 0x08000800 erased
2020-06-03T17:14:50 INFO common.c: Flash page at addr: 0x08000c00 erased
2020-06-03T17:14:50 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2020-06-03T17:14:50 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-06-03T17:14:50 INFO flash_loader.c: Successfully loaded flash loader in sram
  4/4 pages written
2020-06-03T17:14:50 INFO common.c: Starting verification of write complete
2020-06-03T17:14:50 INFO common.c: Flash written and verified! jolly good!
```
We are getting closer - there is ANOTHER update available in the meanwhile which deploys V2J37S7 - so you are correct. I'll test this one as well then... :-/

Nightwalker-87, 2020-06-03:
Good news:
The latest firmware V2.J37.S7 (fw-tool-updated to 2.5.4) even works with Release v1.6.0.

```
$USER@$DEVICE:/$LOCAL_PATH:/$LOCAL_PATH/git/stlink$ /$LOCAL_PATH/git/stlink/build/Release/st-info --version
v1.6.0
$USER@$DEVICE:/$LOCAL_PATH:/$LOCAL_PATH/git/stlink$ /$LOCAL_PATH/git/stlink/build/Release/st-flash erase
st-flash 1.6.0
2020-06-03T17:53:24 INFO common.c: Loading device parameters....
2020-06-03T17:53:24 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2020-06-03T17:53:24 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
Mass erasing
$USER@$DEVICE:/$LOCAL_PATH:/$LOCAL_PATH/git/stlink$ /$LOCAL_PATH/git/stlink/build/Release/st-flash write /$LOCAL_PATH/Blinker/Release/Blinker.bin 0x8000000
st-flash 1.6.0
2020-06-03T17:54:22 INFO common.c: Loading device parameters....
2020-06-03T17:54:22 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2020-06-03T17:54:22 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2020-06-03T17:54:22 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000c00 erased
2020-06-03T17:54:22 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2020-06-03T17:54:22 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-06-03T17:54:22 INFO flash_loader.c: Successfully loaded flash loader in sram
  4/4 pages written
2020-06-03T17:54:22 INFO common.c: Starting verification of write complete
2020-06-03T17:54:22 INFO common.c: Flash written and verified! jolly good!
```

Nightwalker-87, 2020-06-03:
> I have encountered the same problem after updating ST-Link firmware to 2.37.26 in Nucleo-64 STM32F030R8 board.

@grzegorz-kraszewski: Your board uses a different programmer hardware: STLINK-v2-1.
Please open a separate issue for that if still relevant. This ticket is solely related to STLINK-v2. Thx.

@chenguokai: I think we should close this now as resolved, as the original problem obviously does not relate to the `stlink` toolset.

busier, 2020-06-03:
don't changes:
`~/stlink/build/Release/bin$ ./st-flash write ~/Downloads/monitor_lmi.bin 0x8000000
st-flash 1.6.0-364-g60f8461
2020-06-03T19:18:40 WARN common.c: unknown chip id! 0x5fa0004
Failed to connect to target
`

Nightwalker-87, 2020-06-03:
@busier: As shown, this is not a problem of the `stlink` toolset itself. Firmware is provided by ST Microelectronics as a commercial, proprietary product. Therefore you need to refer to the official ST-Link support forum. We can not provide firmware-support, because it is not part of this project nor do we have developer access to it.

closed 2020-06-03 by Nightwalker-87

chenguokai, 2020-06-04:
My local test with a ST Link v2 clone seems fine with FW version `V2J37S7` too.

Could there be some issues specific to the on-board ST Links?

Nightwalker-87, 2020-06-04:
We don't know yet, but that's why I suggested @grzegorz-kraszewski to open a separate ticket, if the problem persists on an onboard V2-1 module.

busier, 2020-06-04:
For Ubuntu 20.04 and Ubuntu 18.04.
[STM32CubeProgrammer with STM32F103C8T6 (bluepill)](https://www.dropbox.com/s/zn3ixgvs07i2oy9/%D0%A1%D0%BD%D0%B8%D0%BC%D0%BE%D0%BA%20%D1%8D%D0%BA%D1%80%D0%B0%D0%BD%D0%B0%20%D0%BE%D1%82%202020-06-04%2016-35-16.png?dl=0).
[STM32CubeProgrammer with STM32F407VET6](https://www.dropbox.com/s/e57hckjwrcjgk0j/%D0%A1%D0%BD%D0%B8%D0%BC%D0%BE%D0%BA%20%D1%8D%D0%BA%D1%80%D0%B0%D0%BD%D0%B0%20%D0%BE%D1%82%202020-06-04%2016-36-52.png?dl=0).
But st-flash out haven't any changes:
```
st-flash 1.6.0-364-g60f8461
2020-06-03T19:18:40 WARN common.c: unknown chip id! 0x5fa0004
Failed to connect to target
```

Nightwalker-87, 2020-06-04:
@busier: This ticket is closed and will not be reopened. Please do not recycle it.