stlink-org/stlink #966: udev rules installing to wrong directory

spetroce, 2020-05-27:
Thank you for giving feedback to the stlink project.

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: NUCLEO-H745ZI-Q with STLINK-V3
- Programmer firmware version: link007_V2-36-26
- Operating system and version: Fedora 31
- **Stlink tools version** and/or git commit hash: fd856ddc40f9e9f55f70b76f4429d16081ee5b02
- Stlink commandline tool name: st-flash
- Target chip (and board if applicable): STM32-H745

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

-- Installing: /usr/local/etc/stlink/udev/rules.d/49-stlinkv1.rules
-- Installing: /usr/local/etc/stlink/udev/rules.d/49-stlinkv2-1.rules
-- Installing: /usr/local/etc/stlink/udev/rules.d/49-stlinkv2.rules
-- Installing: /usr/local/etc/stlink/udev/rules.d/49-stlinkv3.rules

Expected/description:

During the installation, the etc directory is being prefixed with the default system wide installation directory. Should be the following:

-- Installing: /etc/stlink/udev/rules.d/49-stlinkv1.rules
-- Installing: /etc/stlink/udev/rules.d/49-stlinkv2-1.rules
-- Installing: /etc/stlink/udev/rules.d/49-stlinkv2.rules
-- Installing: /etc/stlink/udev/rules.d/49-stlinkv3.rules

**NOTICE: This bug report may be closed without further notice, if not enough information is provided!**

Thank you for your support.

The stlink project maintainers

Nightwalker-87, 2020-05-27:
Yes, that should indeed go to the mentioned directory path.

Nightwalker-87, 2020-05-27:
Fixed. Thx for reporting.

closed 2020-05-27 by Nightwalker-87

Nightwalker-87, 2020-05-27:
Related to #557.

commit 3e49becc (2020-05-27): Corrected udev & modprobe install directories

(Fixes #966)