stlink-org/stlink #979: Nucleo32l011

Krasnov-Max, 2020-06-07:
Thank you for giving feedback to the stlink project.

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: v2-1
- Programmer firmware version: [enter here] (e.g STSW-LINK007 2.27.15)
- Operating system and version: Linux
- **Stlink tools version** and/or git commit hash: v1.6.1
- Stlink commandline tool name: st-flash
- Target chip (and board if applicable): NUCLEO STM32L011K4

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:
 Executing task: /usr/local/bin/st-flash write ./build/nrf905.bin 0x08000000 <

st-flash 1.6.1
2020-06-07T15:38:43 INFO common.c: L011: 8 KiB SRAM, 16 KiB flash in at least 128 byte pages.
file ./build/nrf905.bin md5 checksum: 20e241dbd9511476721d9a864f43da, stlink checksum: 0x000b7fc7
2020-06-07T15:38:43 INFO common.c: Attempting to write 8288 (0x2060) bytes to stm32 address: 134217728 (0x8000000)
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000000 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000080 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000100 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000180 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000200 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000280 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000300 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000380 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000400 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000480 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000500 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000580 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000600 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000680 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000700 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000780 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000800 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000880 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000900 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000980 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000a00 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000a80 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000b00 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000b80 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000c00 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000c80 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000d00 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000d80 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000e00 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000e80 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000f00 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08000f80 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001000 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001080 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001100 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001180 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001200 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001280 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001300 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001380 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001400 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001480 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001500 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001580 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001600 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001680 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001700 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001780 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001800 erased
2020-06-07T15:38:43 INFO common.c: Flash page at addr: 0x08001880 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001900 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001980 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001a00 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001a80 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001b00 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001b80 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001c00 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001c80 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001d00 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001d80 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001e00 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001e80 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001f00 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08001f80 erased
2020-06-07T15:38:44 INFO common.c: Flash page at addr: 0x08002000 erased
2020-06-07T15:38:44 INFO common.c: Finished erasing 65 pages of 128 (0x80) bytes
2020-06-07T15:38:44 INFO common.c: Starting Half page flash write for STM32L core id
2020-06-07T15:38:44 INFO flash_loader.c: Successfully loaded flash loader in sram
2020-06-07T15:38:44 ERROR flash_loader.c: flash loader run error
2020-06-07T15:38:44 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2020-06-07T15:38:44 WARN common.c: 
write_half_pages failed == -1
Процесс терминала завершен с кодом выхода: 255

Krasnov-Max, 2020-06-07:
this problem present fo version 1.6.0 and 1.6.1. 
Version 1.5.0 OK

chenguokai, 2020-06-07:
It seems that no flash loader changes have been made from 1.5.0 to 1.6.0. Could there be some argument changes?

Krasnov-Max, 2020-06-07:
No argument not change.

RobertTDowling, 2020-06-12:
I have a similar failure with 1.6.1 on Nucleo-F303RE.  No failure with 1.6.0

The first time after enumerating my nucleo board, 1.6.1 will flash my firmware*, but I can't flash it a second time without reenumerating my nucleo.  With 1.6.0 and I can flash my firmware over and over without renumeration.

(* actually, it is more complicated than that, because I also flash a separate data payload up high, and that always fails with 1.6.1, but the main use case, flashing down low, has the "first time works" behavior very consistently)

I have not tried 1.6.2, but I'm just using the stock stlink installed in Fedora 32.  (It just updated to 1.6.1 today and it started failing.  Rolling back to 1.6.0 fixed my issue)

Here is output from 1.6.1.  Success first time; failure second time

08:43 % make burn
st-flash write build/f303-4mxtal-fungen.bin 0x8000000
st-flash 1.6.1
2020-06-12T08:43:36 INFO common.c: F303 high density: 64 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
file build/f303-4mxtal-fungen.bin md5 checksum: dd789144773d2bfbd515f06fa72be9c8, stlink checksum: 0x002be489
2020-06-12T08:43:36 INFO common.c: Attempting to write 27704 (0x6c38) bytes to stm32 address: 134217728 (0x8000000)
2020-06-12T08:43:36 INFO common.c: Flash page at addr: 0x08000000 erased
2020-06-12T08:43:36 INFO common.c: Flash page at addr: 0x08000800 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08001000 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08001800 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08002000 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08002800 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08003000 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08003800 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08004000 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08004800 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08005000 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08005800 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08006000 erased
2020-06-12T08:43:37 INFO common.c: Flash page at addr: 0x08006800 erased
2020-06-12T08:43:37 INFO common.c: Finished erasing 14 pages of 2048 (0x800) bytes
2020-06-12T08:43:37 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-06-12T08:43:37 INFO flash_loader.c: Successfully loaded flash loader in sram
 14/14 pages written
2020-06-12T08:43:38 INFO common.c: Starting verification of write complete
2020-06-12T08:43:38 INFO common.c: Flash written and verified! jolly good!

08:43 % make burn
st-flash write build/f303-4mxtal-fungen.bin 0x8000000
st-flash 1.6.1
2020-06-12T08:44:39 INFO common.c: F303 high density: 64 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
file build/f303-4mxtal-fungen.bin md5 checksum: dd789144773d2bfbd515f06fa72be9c8, stlink checksum: 0x002be489
2020-06-12T08:44:39 INFO common.c: Attempting to write 27704 (0x6c38) bytes to stm32 address: 134217728 (0x8000000)
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08000000 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08000800 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08001000 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08001800 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08002000 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08002800 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08003000 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08003800 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08004000 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08004800 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08005000 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08005800 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08006000 erased
2020-06-12T08:44:39 INFO common.c: Flash page at addr: 0x08006800 erased
2020-06-12T08:44:39 INFO common.c: Finished erasing 14 pages of 2048 (0x800) bytes
2020-06-12T08:44:39 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-06-12T08:44:39 INFO flash_loader.c: Successfully loaded flash loader in sram
2020-06-12T08:44:39 ERROR flash_loader.c: flash loader run error
2020-06-12T08:44:39 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
make: *** [Makefile:214: burn] Error 255

Nightwalker-87, 2021-03-24:
@Ant-ON This should be #1113 as well...

Ant-ON, 2021-03-24:
@Nightwalker-87 yes, this is most likely closes by #1113

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87