stlink-org/stlink #988: [STM32 device name]: [Title]

imyaFamiliev, 2020-06-18:
Thank you for giving feedback to the stlink project.

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: [enter here] (e.g Stlink /v1, /v2, /v2-clone, /v2-1)
- Programmer firmware version: [enter here] (e.g STSW-LINK007 2.27.15)
- Operating system and version: [enter here] (e.g Linux, Mac OS X, Windows)
- **Stlink tools version** and/or git commit hash: [enter here] (e.g v1.1.0/git-c722056)
- Stlink commandline tool name: [enter here] (e.g `st-info`, `st-flash`, `st-util`)
- Target chip (and board if applicable): [enter here] (e.g STM32F402VG)

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:

`short description of the expected value`

**NOTICE: This bug report may be closed without further notice, if not enough information is provided!**

Thank you for your support.

The stlink project maintainers

slyshykO, 2020-06-18:
@imyaFamiliev, Can you provide some information about the issue you complain?

slyshykO, 2020-06-18:
@imyaFamiliev, please provede size of **./build/iss.bin**

Nightwalker-87, 2020-06-18:
@imyaFamiliev: Please use the Template form you just deleted and fill it out. Otherwise I may simply discard this ticket - I'm fed up with this - really. We have so clear instructions on how to open issues properly and most people seem able to understand these.

imyaFamiliev, 2020-06-18:
iss.bin 131384 bytes

imyaFamiliev, 2020-06-18:
ok, message will be fixed.

Nightwalker-87, 2020-06-18:
Please follow #906.

Nightwalker-87, 2020-06-18:
Does this already occur with Release v1.6.0?

imyaFamiliev, 2020-06-18:
@Nightwalker-87 don't see windows release build of v1.6.0
I will try to compile and check.

Nightwalker-87, 2020-06-18:
oh, yeah - that's right. Please follow `/doc/compiling.md` for windows and make sure to cleanly remove the newer version beforehand. Release v1.6.1 was the first release after a long period to distribute windows builds.

imyaFamiliev, 2020-06-18:
1.6.0 flashes ok

	C:\VSARM\/stlink/bin/st-flash write  ./build/iss.bin 0x08000000 <
    st-flash 1.6.0
    2020-06-18T15:26:57 INFO common.c: Loading device parameters....
    2020-06-18T15:26:57 INFO common.c: Device connected is: F1 High-density device, id 0x10036414
    2020-06-18T15:26:57 INFO common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
    2020-06-18T15:26:57 INFO common.c: Attempting to write 131384 (0x20138) bytes to stm32 address: 134217728 (0x8000000)
    Flash page at addr: 0x08020000 erased
    2020-06-18T15:26:58 INFO common.c: Finished erasing 65 pages of 2048 (0x800) bytes
    2020-06-18T15:26:58 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
    2020-06-18T15:26:58 INFO flash_loader.c: Successfully loaded flash loader in sram
     65/65 pages written
    2020-06-18T15:27:04 INFO common.c: Starting verification of write complete
    2020-06-18T15:27:05 INFO common.c: Flash written and verified! jolly good!

slyshykO, 2020-06-18:
st-flash from stlink-1.6.1-i686-w64-mingw32.zip work well.
the problem occurs on st-flash from stlink-1.6.1-x86_64-w64-mingw32.zip

slyshykO, 2020-06-18:
@imyaFamiliev  which toolchain you use to build st-link? It is 64 or 32 bit one?

imyaFamiliev, 2020-06-18:
for 1.6.0 i use x86_64-5.3.0-win32-sjlj-rt_v4-rev0 as used in mingw64-build.bat
1.6.1 version  I downloaded as is stlink-1.6.1-x86_64-w64-mingw32.zip
check stlink-1.6.1-i686-w64-mingw32.zip - work fine

imyaFamiliev, 2020-06-18:
i have Intel(R)_Core(TM)_i5-8400_CPU

Nightwalker-87, 2020-06-18:
> st-flash from stlink-1.6.1-i686-w64-mingw32.zip work well.
> the problem occurs on st-flash from stlink-1.6.1-x86_64-w64-mingw32.zip

This could be somehow similar to the casting issues we faced with clang on macOS earlier, just in another context ...

chenguokai, 2020-06-18:
> This could be somehow similar to the casting issues we faced with clang on macOS earlier, just in another context ...
It seems to be caused by a size check combined with a cast somewhere

PR #989 (2020-06-18): st-flash: Fix for mmap() size_t overflow
fix #988
commit 1541dfcb (2020-06-18): fix #988
closed 2020-06-18 by Nightwalker-87