stlink-org/stlink #990: [All devices]: Bad user experiance due to too high clock frequency in st-info

oloftangrot, 2020-06-18:
- Programmer/board type: Every board 
- Programmer firmware version: Every firmware 
- Operating system and version: All operating systems 
- **Stlink tools version** and/or git commit hash: [HEAD] 
- Stlink commandline tool name: st-info
- Target chip (and board if applicable): Verified on a STM32F108C

The st-info tool defaults to 1,8 MHz clock speed without telling the user about it or allowing the user to select a more adequate speed. Depending on cable lengths between the programmer and the target, and other environmental conditions this is not a stable speed for a tool that does not require high band width but rather should provide confidence in the tools.

I have made a local branch that fixes the problem by:
- Defaults to a 100 kHz.
- Always displays the used clock frequency
- Allows the user to select frequency
- Does not terminate without a clear error message if the user selects a frequency that is not supported.
- Provides the user with information about valid clock frequencies.
- Terminates on a unknown options and shows the option that caused termination.

Nightwalker-87, 2020-06-21:
@oloftangrot: If you desire such change to our toolset, you should put up a PR with these changes for review.

oloftangrot, 2020-06-22:
Don't ask me about my desires. It is an issue report. I forgot to mention selecting frequency by the abusing absolute temperature in Kelvin or prefix mili was not an option. I guess that would be on the issue report for st-flash tough.

Nightwalker-87, 2020-06-22:
The point is if you have a fix - why don't you submit it as a proposal instead of opening an issue telling there is a solution?
I don't really understand the rest of your reply, sry.

Nightwalker-87, 2020-06-27:
@chenguokai @slyshykO: What shall we do with this? Should the documentation be improved?
I know that sometimes it does make sense to lower the clock speed (e.g. if the connection wire is of poor quality or very long).
However this is no general circumstance to be preset by a programming tool...

chenguokai, 2020-06-27:
I would suggest adding a documentation for this.

BTW if I got it right, the default clock has become adjustable in v1.6.1.

Nightwalker-87, 2020-06-27:
Well there actually IS some documentation already, since your contribution in #985 .
The point of @oloftangrot seems to be that he expected it to be present in `st-info`. 
I don't really understand why that should be necessary for this specific tool.

oloftangrot, 2020-06-29:
Added diff for my fix.

diff --git a/src/st-info/info.c b/src/st-info/info.c
index 1bc48a0..12bab28 100644
--- a/src/st-info/info.c
+++ b/src/st-info/info.c
@@ -5,16 +5,21 @@
 
 #include <stlink.h>
 
+const char * knownFreqs ="5k 15k 25k 50k 100k 125k 240k 480k 950k 1200k 1.2M 1800k 1.8M 4000k 4M";
+unsigned int myFrequencyInkHz = 100u; // Default clock speed.
+
 static void usage(void) {
     puts("st-info --version");
     puts("st-info --probe");
     puts("st-info --serial");
     puts("st-info --hla-serial");
-    puts("st-info --flash  [--connect-under-reset]");
-    puts("st-info --pagesize  [--connect-under-reset]");
-    puts("st-info --sram  [--connect-under-reset]");
-    puts("st-info --chipid  [--connect-under-reset]");
-    puts("st-info --descr  [--connect-under-reset]");
+    puts("st-info --flash [--freq nn] [--connect-under-reset]");
+    puts("st-info --pagesize [--freq nn] [--connect-under-reset]");
+    puts("st-info --sram [--freq nn] [--connect-under-reset]");
+    puts("st-info --chipid [--freq nn] [--connect-under-reset]");
+    puts("st-info --descr [--freq nn] [--connect-under-reset]");
+		fprintf(stdout, "Valid clock frequency selections: %s\n", knownFreqs);
+    
 }
 
 /* Print normal or OpenOCD hla_serial with newline */
@@ -80,15 +85,24 @@ static stlink_t *stlink_open_first(bool under_reset) {
 
     if (sl == NULL) {
         if (under_reset) {
-            sl = stlink_open_usb(0, 2, NULL, 0);
+            sl = stlink_open_usb(0, 2, NULL, myFrequencyInkHz);
         } else {
-            sl = stlink_open_usb(0, 1, NULL, 0);
+            sl = stlink_open_usb(0, 1, NULL, myFrequencyInkHz);
         }
     }
 
     return(sl);
 }
 
+static bool starts_with(const char * str, const char * prefix) {
+    size_t n = strlen(prefix);
+
+    if (strlen(str) < n) { 
+			return(false); 
+		}
+    return (0 == strncmp(str, prefix, n));
+}
+
 static int print_data(int ac, char **av) {
     stlink_t* sl = NULL;
     bool under_reset = false;
@@ -102,19 +116,89 @@ static int print_data(int ac, char **av) {
         return(0);
     }
 
-    if (ac == 3) {
-        if (strcmp(av[2], "--connect-under-reset") == 0) {
+		int extra = 2;
+	  bool searchForOpts = false;
+    if (ac >= 3) searchForOpts = true;
+    for (;searchForOpts;) {
+				bool unknownOpt = true;
+        if (strcmp(av[extra], "--connect-under-reset") == 0) {
             under_reset = true;
-        } else {
-            usage();
-            return(-1);
+						unknownOpt = false;
+        } 
+				else if (strcmp(av[extra], "--freq") == 0 || starts_with(av[extra], "--freq=")) {
+            const char* freq;
+
+            if (strcmp(av[extra], "--freq") == 0) {
+//                ac--;
+//                av++;
+								extra++;
+
+                if (extra > ac) { // No more arguments on the command line.
+										fprintf(stderr, "No valid frequency number given as argument!\n");
+                    return(-1);
+                }
+                freq = av[extra];
+            } else {
+                freq = av[extra] + strlen("--freq="); // Advance pointer throug char array.
+//								printf("String %s\n", freq);
+            }
+            // Check for digits followed by a valid postfix.
+            if (strcmp(freq, "5k") == 0) {
+                myFrequencyInkHz = 5u;
+								unknownOpt = false;
+            } else if (strcmp(freq, "15k") == 0) {
+                myFrequencyInkHz = 15u;
+								unknownOpt = false;
+            } else if (strcmp(freq, "25k") == 0) {
+                myFrequencyInkHz = 25u;
+								unknownOpt = false;
+            } else if (strcmp(freq, "50k") == 0) {
+                myFrequencyInkHz = 50u;
+								unknownOpt = false;
+            } else if (strcmp(freq, "100k") == 0) {
+                myFrequencyInkHz = 100u;
+								unknownOpt = false;
+            } else if (strcmp(freq, "125k") == 0) {
+                myFrequencyInkHz = 125u;
+								unknownOpt = false;
+            } else if (strcmp(freq, "240k") == 0) {
+                myFrequencyInkHz = 240u;
+								unknownOpt = false;
+            } else if (strcmp(freq, "480k") == 0) {
+                myFrequencyInkHz = 480u;
+								unknownOpt = false;
+            } else if (strcmp(freq, "950k") == 0) {
+                myFrequencyInkHz = 950u;
+								unknownOpt = false;
+            } else if ((strcmp(freq, "1200k") == 0) || (strcmp(freq, "1.2M") == 0)) {
+                myFrequencyInkHz = 1200u;
+								unknownOpt = false;
+            } else if ((strcmp(freq, "1800k") == 0 ) || (strcmp(freq, "1.8M") == 0 )) {
+                myFrequencyInkHz = 1800u;
+								unknownOpt = false;
+            } else if ((strcmp(freq, "4000k") == 0) || (strcmp(freq, "4M") == 0 )) {
+                myFrequencyInkHz = 4000u;
+								unknownOpt = false;
+            } else {
+								fprintf(stderr, "No valid frequency found!\nValid frequencies: %s\n", knownFreqs);
+                return(-1);
+            }
         }
+				if ( unknownOpt ) {
+					fprintf(stderr, "Unknown option! %s\nValid options:\n", av[extra]);
+					usage();
+					return(-1);
+				}
+				extra++;
+				if (extra >= ac) searchForOpts = false;
     }
-
+    fprintf(stdout, "Using clock %u kHz.\n", myFrequencyInkHz);
     sl = stlink_open_first(under_reset);
 
-    if (sl == NULL) { return(-1); }
-
+    if (sl == NULL) {
+			 fprintf(stderr, "Could not open programmer interface!");
+			 return(-1);
+		}
     sl->verbose = 0;
 
     if (stlink_current_mode(sl) == STLINK_DEV_DFU_MODE) { stlink_exit_dfu_mode(sl); }

Nightwalker-87, 2021-03-11:
@Ant-ON: What is your opinion on this?

Personally, I still don't see a general use case for this. Such long cable lengths that would possible require to reduce the clock speed down from the MHz range are not preferable in terms of primary line constants and/or external ingress. Also `st-info` is not a tool greatly interacting with the chipset - it only reads out MCU info.

Ant-ON, 2021-03-12:
@Nightwalker-87 I think adding a frequency setting is a useful feature. But changes from "fix.diff.txt" need some work. At least the code style need change. It does not match the one used in the project.

Nightwalker-87, 2021-03-12:
Ok, so who is willing to contribute here?

Nightwalker-87, 2021-03-20:
@oloftangrot Please provide a PR instead of a diff.

oloftangrot, 2021-03-22:
I might have flushed that branch long time ago and don't really have time to check on it. And I would not know how to make a pull request even if I found it. So I guess it will be least work for all parts if you just apply my diff and commit it directy self.

Nightwalker-87, 2021-03-22:
... and leave this ticket to automatic maintenance ;-)

PR #1114 (2021-03-26): [feature] Extended set of cmd line arguments for st-info and st-util
Added option `--freq` to st-info and st-util.
Added option `--hot-plug` to st-info. It allows to get information about the target without stop the program execution.
Changed the frequency parsing algorithm. Added the ability to set an arbitrary frequency since ST-Link/v3 does not have a predefined list of frequencies.

(Closes #990, Closes #1091)
closed 2021-03-26 by Nightwalker-87